evcc-io/evcc #16894: Kia template does not work anymore

thieso5000, 2024-10-27:
### Describe the bug

I have tried to update the container in my synology to Version v0.130.11 since the EVCC Dashboard told me there is a new version. I have stopped the container, reset it and restarted it under the assumption it will automatically grab the update. Obviously it did since the log says [main ] INFO 2024/10/27 08:12:04 evcc 0.130.11 
But after restart container Dashboard still shows bottom left "new version v0.130.11" is available. But in the log ist says it is updated to the new version? [main ] INFO 2024/10/27 08:12:04 evcc 0.130.11

However, since this moment the Kia connection and ID4 connection of my both cars does not work anymore. I have nothing change int the YAML file, but still it won't connect. Email and Passsword are correct. APP from Kia and VW works fine. 
ID4 is also not working but therer is no error in the File. Only Kia error shows up. 

Error File: 
[main ] INFO 2024/10/27 08:12:06 listening at :7070
[main ] ERROR 2024/10/27 08:12:14 creating vehicle Kia failed: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
[lp-1 ] DEBUG 2024/10/27 08:12:19 charge total import: 50.452kWh
[site ] WARN 2024/10/27 08:12:19 vehicle 'Kia Nero' does not support automatic detection

[lp-1 ] INFO 2024/10/27 08:34:09 vehicle updated: Kia Nero -> ID4
[site ] DEBUG 2024/10/27 08:34:19 ----
[lp-1 ] DEBUG 2024/10/27 08:34:19 charge power: 0W
[lp-1 ] DEBUG 2024/10/27 08:34:19 charge currents: [0 0 0]A

I am not sure if this is a coincidence or a bug. Yesterday it was still working. 
Thanks for helping out. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: Kia
    type: template
    template: kia
    title: Kia Nero
    user: xxx@aol.com # user
    password: xxxxxx # password
    vin:
    capacity: 68
    # onIdentify: # set defaults when vehicle is identified
    mode: pv # enable PV-charging when vehicle is identified
    
  - name: ID4
    type: template
    template: vw
    title: ID4 # Wird in der Benutzeroberfläche angezeigt (optional)
    user: xxx@mac.com # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxxxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    capacity: 77 # Akkukapazität in kWh (optional)
    timeout: 10s # optional
```

### Log details

```text
[main ] INFO 2024/10/27 08:12:06 listening at :7070
[main ] ERROR 2024/10/27 08:12:14 creating vehicle Kia failed: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
[lp-1 ] DEBUG 2024/10/27 08:12:19 charge total import: 50.452kWh
[site ] WARN 2024/10/27 08:12:19 vehicle 'Kia Nero' does not support automatic detection

[lp-1 ] INFO 2024/10/27 08:34:09 vehicle updated: Kia Nero -> ID4
[site ] DEBUG 2024/10/27 08:34:19 ----
[lp-1 ] DEBUG 2024/10/27 08:34:19 charge power: 0W
[lp-1 ] DEBUG 2024/10/27 08:34:19 charge currents: [0 0 0]A
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.130.11

VolkerK62, 2024-10-27:
no problem with Hyundai
and latest version is 0.131.1

0x3d13f, 2024-10-27:
[image]

0x3d13f, 2024-10-27:
https://cloud.eu.connect.kia.com/account-maintenance-de

Bitte beachten Sie, dass es am Sonntag, den 27. Oktober zwischen 7:00-9:00 Uhr (MEZ) zu vorübergehenden Anmeldeproblemen mit dem Kia Account kommen kann.

thieso5000, 2024-10-27:
Also das Update habe ich jetzt endlich hinbekommen. Aktuelle Version ist jetzt v0.131.1. 
Verbindung zu Kia klappt aber immer noch nicht.  Der Login von meinem ID4 klappt ja auch nicht (allerdings kommt da im Log keine Fehlermeldung). 
Über die APP auf dem Handy kann ich mich aber bei beiden einloggen. Daher vermute ich, dass die Wartungsarbeiten bei Kia nicht der Grund sein können.

0x3d13f, 2024-10-27:
Aber am Kia Template liegt es dann ja wohl auch eher nicht, wenn der ID4 auch nicht geht. Zumindest fiel es zeitlich mit den Wartungsarbeiten bei Kia zusammen. Vielleicht dauern diese ja noch an.
Kannst Du in Deiner Umgebung `evcc vehicle --log trace` ausführen?

0x3d13f, 2024-10-27:
Hier kommt der Fehler bei `evcc vehicle --log trace` jetzt auch. Vermutlich haben die bei den genannten Wartungsarbeiten wieder mal eine "Verbesserung" eingepflegt:
```
root@evcc:~# evcc vehicle --log trace
[main  ] INFO 2024/10/27 10:22:43 evcc 0.131.1
[main  ] INFO 2024/10/27 10:22:43 using config file: /root/evcc.yaml
[db    ] INFO 2024/10/27 10:22:43 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/10/27 10:22:43 connecting evcc-1318381628 at tcp://xxx:1883
[mqtt  ] DEBUG 2024/10/27 10:22:43 tcp://xxx:1883 connected
[kia   ] TRACE 2024/10/27 10:22:43 POST https://prd.eu-ccapi.kia.com:8080/api/v1/spa/notifications/register
[kia   ] TRACE 2024/10/27 10:22:43 {"pushRegId":"20861D4AA7CE85B208A83B2D60FCFD95F204384F3C7F3E396DAD38DBB7964239","pushType":"APNS","uuid":"76aa9d10-8bad-43e8-9011-ebe44bc1407f"}
--
{"retCode":"S","resCode":"0000","resMsg":{"deviceId":"b334e362-5065-4d2a-8923-c978010ed613"},"msgId":"055e38d0-9445-11ef-8aef-09455beb226a"}
[kia   ] TRACE 2024/10/27 10:22:43 GET https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/authorize?response_type=code&state=test&client_id=fdc85c00-0a2f-4c64-bcb4-2cfb1500730a&redirect_uri=https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/redirect
[kia   ] TRACE 2024/10/27 10:22:43 <a href="https://prd.eu-ccapi.kia.com:8080/web/v1/user/authorize?cache=reset">Found</a>.
[kia   ] TRACE 2024/10/27 10:22:43 GET https://prd.eu-ccapi.kia.com:8080/web/v1/user/authorize?cache=reset
[kia   ] TRACE 2024/10/27 10:22:43 <!doctype html><html lang="en"><head><meta charset="utf-8"><meta name="viewport" content="width=device-width,initial-scale=1,shrink-to-fit=no,maximum-scale=1,user-scalable=0"><meta name="theme-color" content="#ffffff"><link href="https://fonts.googleapis.com/css?family=Nanum+Gothic:400,700,800&amp;subset=korean" rel="stylesheet"><link href="https://fonts.googleapis.com/icon?family=Material+Icons" rel="stylesheet"><link rel="manifest" href="/web/v1/user/manifest.json"><link rel="icon" href="/web/v1/user/favicon_kia.ico" type="image/x-icon"><!--[if lt IE 10]><script src="/web/v1/user/fetch.js"></script><![endif]--><title></title><link href="/web/v1/user/static/css/main.1edc2e54.chunk.css" rel="stylesheet"></head><body><noscript>You need to enable JavaScript to run this app.</noscript><div id="root"></div><script>!function(u){function e(e){for(var o,r,t=e[0],n=e[1],a=e[2],c=0,l=[];c<t.length;c++)r=t[c],f[r]&&l.push(f[r][0]),f[r]=0;for(o in n)Object.prototype.hasOwnProperty.call(n,o)&&(u[o]=n[o]);for(p&&p(e);l.length;)l.shift()();return d.push.apply(d,a||[]),i()}function i(){for(var e,o=0;o<d.length;o++){for(var r=d[o],t=!0,n=1;n<r.length;n++){var a=r[n];0!==f[a]&&(t=!1)}t&&(d.splice(o--,1),e=s(s.s=r[0]))}return e}var r={},f={34:0},d=[];function s(e){if(r[e])return r[e].exports;var o=r[e]={i:e,l:!1,exports:{}};return u[e].call(o.exports,o,o.exports,s),o.l=!0,o.exports}s.e=function(a){var e=[],r=f[a];if(0!==r)if(r)e.push(r[2]);else{var o=new Promise(function(e,o){r=f[a]=[e,o]});e.push(r[2]=o);var t,c=document.createElement("script");c.charset="utf-8",c.timeout=120,s.nc&&c.setAttribute("nonce",s.nc),c.src=s.p+"static/js/"+({0:"Locale0",1:"Locale1",2:"Locale10",3:"Locale11",4:"Locale12",5:"Locale13",6:"Locale14",7:"Locale15",8:"Locale16",9:"Locale17",10:"Locale18",11:"Locale19",12:"Locale2",13:"Locale20",14:"Locale21",15:"Locale22",16:"Locale23",17:"Locale24",18:"Locale25",19:"Locale26",20:"Locale27",21:"Locale28",22:"Locale29",23:"Locale3",24:"Locale4",25:"Locale5",26:"Locale6",27:"Locale7",28:"Locale8",29:"Locale9",30:"agreement_cert",31:"agreement_genesis",33:"resident_registration"}[a]||a)+"."+{0:"8326571e",1:"b63f1662",2:"8976dc4e",3:"96c6e71b",4:"954b38b3",5:"46f01880",6:"e7c9e0d0",7:"2a3b9f3d",8:"e9dbe038",9:"06a8213c",10:"8d9c6412",11:"76a432f8",12:"92546437",13:"f51024b6",14:"3c5545c3",15:"4a860957",16:"396d6264",17:"4bc9d505",18:"2257f432",19:"32a85ecb",20:"9d3034a0",21:"e7cde999",22:"9308e09e",23:"25d10409",24:"ed8bea17",25:"093694e9",26:"1ff17ab2",27:"9d0ffa03",28:"98f01860",29:"73681047",30:"4d6cdb91",31:"f028c685",33:"d94aa440"}[a]+".chunk.js",t=function(e){c.onerror=c.onload=null,clearTimeout(l);var o=f[a];if(0!==o){if(o){var r=e&&("load"===e.type?"missing":e.type),t=e&&e.target&&e.target.src,n=new Error("Loading chunk "+a+" failed.\n("+r+": "+t+")");n.type=r,n.request=t,o[1](n)}f[a]=void 0}};var l=setTimeout(function(){t({type:"timeout",target:c})},12e4);c.onerror=c.onload=t,document.head.appendChild(c)}return Promise.all(e)},s.m=u,s.c=r,s.d=function(e,o,r){s.o(e,o)||Object.defineProperty(e,o,{enumerable:!0,get:r})},s.r=function(e){"undefined"!=typeof Symbol&&Symbol.toStringTag&&Object.defineProperty(e,Symbol.toStringTag,{value:"Module"}),Object.defineProperty(e,"__esModule",{value:!0})},s.t=function(o,e){if(1&e&&(o=s(o)),8&e)return o;if(4&e&&"object"==typeof o&&o&&o.__esModule)return o;var r=Object.create(null);if(s.r(r),Object.defineProperty(r,"default",{enumerable:!0,value:o}),2&e&&"string"!=typeof o)for(var t in o)s.d(r,t,function(e){return o[e]}.bind(null,t));return r},s.n=function(e){var o=e&&e.__esModule?function(){return e.default}:function(){return e};return s.d(o,"a",o),o},s.o=function(e,o){return Object.prototype.hasOwnProperty.call(e,o)},s.p="/web/v1/user/",s.oe=function(e){throw console.error(e),e};var o=window.webpackJsonp=window.webpackJsonp||[],t=o.push.bind(o);o.push=e,o=o.slice();for(var n=0;n<o.length;n++)e(o[n]);var p=t;i()}([])</script><script src="/web/v1/user/static/js/35.f02a2a81.chunk.js"></script><scr [... cut ...]
{"errId":"34dbab7f-212b-4e93-a269-ded3f8f882a5","errCode":"4003","errMsg":"Invalid values"}
[main  ] ERROR 2024/10/27 10:22:48 creating vehicle EV6 failed: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
EV6
---
.Soc:        vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
Capacity:    77.0kWh
Identifiers: [xxxxx]
Features:    [Offline Retryable]

```