evcc-io/evcc #16907: Netzladen SBS 5.0 funktioniert nur mit dem "falschen" template sma-sbs-15-25-modbus

Deepy1973, 2024-10-27:
### Describe the bug

Ich verwende evcc als Add-on unter Home Assistant in Kombination mit zwei SMA Wechselrichter (Sunny Tripower 10.0 und Sunny Tripower X12), einer 11kw BYD Batterie an SMA Batteriewechselrichter Sunny boy Storage 5.0 und dem SHM 2.0 mit der aktuellen Software 2.15.6. R

Trotz aller Bemühungen hab ich das System nicht dazu überreden können, die Batterie aus dem Netz zu laden :-(
Als Standardstromtarif habe ich 29ct und als günstig (von 0:00Uhr bis 5:00Uhr) 20ct eingetragen.

Nach langer Fehlersuche hab ich einfach mal das sma-sbs-15-25-modbus template für meinen Sunny Boy Storage 5.0 genommen. Und siehe da die Batterie lädt bis 2,6KW. 

Zurück auf das "richtige" template sma-sbs-modbus und laden aus dem Netz wird umgehend gestoppt. Die Batterie wird quasi "nur" gesperrt und es wird max. Hausverbrauch aus dem Netz gezogen. Ansonsten, z.B. beim Schnellladen lädt und entlädt sie bis 4,6kw. Läßt sich sperren für Schnelladen und nimmt bis zu 4,6kw vom PV Dach auf.

Kann es sein, das das sma-sbs-modbus template noch dieselben letzten Modifikationen des sma-sbs-15-25-modbus templates braucht?

### Steps to reproduce

1. Preis so eingestellt, das die Batterie aus dem Netz geladen werden soll.
UI bestätigt günstiges Netzladen
2. sma-sbs-modbus als template eintragen und Neustarten
3. Batterie lädt nicht aus dem Netz
4. sma-sbs-15-25-modbus als template eintragen und Neustarten
5. Batterie lädt umgehend mit bis zu 2,6kw aus dem Netz
6. sma-sbs-modbus als template eintragen und Neustarten
7. Batterie lädt nicht mehr aus dem Netz
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: xxxx

log: debug
levels:
  cache: error

# unique installation id
plant: xxxx

interval: 10s # control cycle interval

sponsortoken: xxxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.1.152  
  name: grid1
- type: template
  template: sma-inverter-speedwire 
  usage: pv  
  host: 192.168.1.91  
  password: xxxx  
  name: pv2
- type: template
  template: sma-inverter-speedwire 
  usage: pv  
  host: 192.168.1.42  
  password: xxxx  
  name: pv3
- type: template
  template: sma-sbs-modbus 
  usage: battery
[... 202 lines cut ...]
[sma   ] TRACE 2024/10/27 14:31:00 recv 192.168.1.91: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2024/10/27 14:31:00 failed to get values for 192.168.1.42:9522: no packet received in timeout
[sma   ] TRACE 2024/10/27 14:31:00 requestValues for 192.168.1.42:9522: 0x5180 0x214800 0x2148FF
[sma   ] TRACE 2024/10/27 14:31:00 send 192.168.1.42: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 14:31:00 recv 192.168.1.152: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 14:31:00 recv 192.168.1.42: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 14:31:00 requestValues for 192.168.1.42:9522: 0x5180 0x416400 0x4164FF
[sma   ] TRACE 2024/10/27 14:31:00 send 192.168.1.42: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 14:31:00 recv 192.168.1.42: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 14:31:00 requestValues for 192.168.1.42:9522: 0x5200 0x237700 0x2377FF
[sma   ] TRACE 2024/10/27 14:31:00 send 192.168.1.42: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 14:31:00 recv 192.168.1.42 invalid: invalid DeviceData - length 28
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.1.31.1

premultiply, 2024-10-27:
Ich sehe im Log kein modbus.

Habe es eben bei meinem SBS 3.7 nochmal getestet: Funktioniert einwandfrei.

Mit den Standardeinstellungen des SBS kann es sein, dass die Anpassung der Kurve im SBS SEHR lange dauert. Mehrere Minuten! Entweder aussitzen oder Einstellungen anpassen. Erfordert Grid Guard Code.

closed 2024-10-27 by premultiply

Deepy1973, 2024-10-27:
Noch der log dazu:

[site  ] DEBUG 2024/10/27 16:36:49 ----
[sma   ] TRACE 2024/10/27 16:36:49 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:36:49 requestValues for 192.168.1.91:9522: 0x5100 0x491E00 0x495DFF
[sma   ] TRACE 2024/10/27 16:36:49 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:36:49 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:36:49 requestValues for 192.168.1.91:9522: 0x5180 0x214800 0x2148FF
[sma   ] TRACE 2024/10/27 16:36:49 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:36:49 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:36:49 requestValues for 192.168.1.91:9522: 0x5180 0x416400 0x4164FF
[sma   ] TRACE 2024/10/27 16:36:49 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:36:49 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:36:49 requestValues for 192.168.1.91:9522: 0x5200 0x237700 0x2377FF
[sma   ] TRACE 2024/10/27 16:36:49 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[site  ] DEBUG 2024/10/27 16:36:49 pv 2 power: 104W
[site  ] DEBUG 2024/10/27 16:36:49 grid power: 3563W
[site  ] DEBUG 2024/10/27 16:36:49 grid powers: [3342 35 186]W
[site  ] DEBUG 2024/10/27 16:36:49 grid currents: [14.7 0.784 1.53]A
[battery4] TRACE 2024/10/27 16:36:49 modbus: send 00 58 00 00 00 06 03 04 78 37 00 02
[site  ] DEBUG 2024/10/27 16:36:49 pv 1 power: 111W
[site  ] DEBUG 2024/10/27 16:36:49 pv power: 215W
[battery4] TRACE 2024/10/27 16:36:49 modbus: recv 00 58 00 00 00 07 03 04 04 ff ff f5 be
[battery4] TRACE 2024/10/27 16:36:49 modbus: send 00 59 00 00 00 06 03 03 77 31 00 04
[sma   ] TRACE 2024/10/27 16:36:49 recv 192.168.1.152: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:36:49 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:36:49 requestValues for 192.168.1.91:9522: 0x5380 0x251E00 0x251EFF
[sma   ] TRACE 2024/10/27 16:36:49 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[battery4] TRACE 2024/10/27 16:36:49 modbus: recv 00 59 00 00 00 0b 03 03 08 00 00 00 00 00 4a 91 69
[battery4] TRACE 2024/10/27 16:36:49 modbus: send 00 5a 00 00 00 06 03 03 78 7d 00 02
[battery4] TRACE 2024/10/27 16:36:49 modbus: recv 00 5a 00 00 00 07 03 03 04 00 00 00 19
[site  ] DEBUG 2024/10/27 16:36:49 battery soc: 25%
[site  ] DEBUG 2024/10/27 16:36:49 battery power: -2626W
[site  ] DEBUG 2024/10/27 16:36:49 battery has priority at soc 25% (< 80%)
[site  ] DEBUG 2024/10/27 16:36:49 site power: 3663W

Deepy1973, 2024-10-27:
> Ich sehe im Log kein modbus.
>
> Habe es eben bei meinem SBS 3.7 nochmal getestet: Funktioniert einwandfrei.
>
> Mit den Standardeinstellungen des SBS kann es sein, dass die Anpassung der Kurve im SBS [...]

hab mehrere Stunden gewartet. Da passiert nix ausser sperren der Batterieentladung.

Deepy1973, 2024-10-27:
Auf sbs-modbus template gewechselt und wieder kein Netzladen der Batterie. Log dazu:

[site  ] DEBUG 2024/10/27 16:45:43 ----
[sma   ] TRACE 2024/10/27 16:45:43 recv 192.168.1.152: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[site  ] DEBUG 2024/10/27 16:45:43 grid power: 1112W
[battery4] TRACE 2024/10/27 16:45:43 modbus: send 00 3d 00 00 00 06 03 04 78 37 00 02
[site  ] DEBUG 2024/10/27 16:45:43 grid powers: [766 42 303]W
[site  ] DEBUG 2024/10/27 16:45:43 pv 2 power: 69W
[site  ] DEBUG 2024/10/27 16:45:43 pv 1 power: 83W
[site  ] DEBUG 2024/10/27 16:45:43 pv power: 152W
[site  ] DEBUG 2024/10/27 16:45:43 grid currents: [3.48 0.796 2.28]A
[battery4] TRACE 2024/10/27 16:45:43 modbus: recv 00 3d 00 00 00 07 03 04 04 ff ff ff e8
[battery4] TRACE 2024/10/27 16:45:43 modbus: send 00 3e 00 00 00 06 03 03 77 31 00 04
[battery4] TRACE 2024/10/27 16:45:43 modbus: recv 00 3e 00 00 00 0b 03 03 08 00 00 00 00 00 4a 91 71
[battery4] TRACE 2024/10/27 16:45:43 modbus: send 00 3f 00 00 00 06 03 03 78 7d 00 02
[battery4] TRACE 2024/10/27 16:45:43 modbus: recv 00 3f 00 00 00 07 03 03 04 00 00 00 1b
[site  ] DEBUG 2024/10/27 16:45:43 battery soc: 27%
[site  ] DEBUG 2024/10/27 16:45:43 battery power: -24W
[site  ] DEBUG 2024/10/27 16:45:43 battery has priority at soc 27% (< 80%)
[site  ] DEBUG 2024/10/27 16:45:43 site power: 1212W
[sma   ] TRACE 2024/10/27 16:45:43 recv 192.168.1.152: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 login for 192.168.1.91:9522
[sma   ] TRACE 2024/10/27 16:45:44 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[semp  ] TRACE 2024/10/27 16:45:44 send: {XMLName:{Space: Local:} Xmlns:urn:schemas-upnp-org:device-1-0 SpecVersion:{Major:1 Minor:0} Device:{DeviceType:urn:schemas-simple-energy-management-protocol:device:Gateway:1 FriendlyName:evcc Manufacturer:github.com/evcc-io/evcc ModelName:EVCC SEMP Server 0.131.1 UDN:uuid:270fd86f-947a-11ef-9890-2ccf678051d5 PresentationURL:http://192.168.1.57:7070 ServiceDefinition:{Xmlns:urn:schemas-simple-energy-management-protocol:service-1-0 Server:http://192.168.1.57:7070 BasePath:/semp Transport:HTTP/Pull ExchangeFormat:XML WsVersion:1.1.0} ServiceList:[]}}
[sma   ] TRACE 2024/10/27 16:45:44 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 requestValues for 192.168.1.91:9522: 0x5100 0x263F00 0x263FFF
[sma   ] TRACE 2024/10/27 16:45:44 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 requestValues for 192.168.1.91:9522: 0x5100 0x295A00 0x295AFF
[sma   ] TRACE 2024/10/27 16:45:44 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 recv 192.168.1.152: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 requestValues for 192.168.1.91:9522: 0x5100 0x411E00 0x4120FF
[sma   ] TRACE 2024/10/27 16:45:44 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 requestValues for 192.168.1.91:9522: 0x5100 0x464000 0x4642FF
[sma   ] TRACE 2024/10/27 16:45:44 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 requestValues for 192.168.1.91:9522: 0x5100 0x464800 0x4655FF
[sma   ] TRACE 2024/10/27 16:45:44 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 requestValues for 192.168.1.91:9522: 0x5100 0x465700 0x4657FF
[sma   ] TRACE 2024/10/27 16:45:44 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 requestValues for 192.168.1.91:9522: 0x5100 0x491E00 0x495DFF
[sma   ] TRACE 2024/10/27 16:45:44 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 requestValues for 192.168.1.91:9522: 0x5180 0x214800 0x2148FF
[sma   ] TRACE 2024/10/27 16:45:44 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 requestValues for 192.168.1.91:9522: 0x5180 0x416400 0x4164FF
[sma   ] TRACE 2024/10/27 16:45:44 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 recv 192.168.1.152: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:44 requestValues for 192.168.1.91:9522: 0x5200 0x237700 0x2377FF
[sma   ] TRACE 2024/10/27 16:45:44 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[... cut ...]
[sma   ] TRACE 2024/10/27 16:45:54 requestValues for 192.168.1.91:9522: 0x5800 0x821E00 0x8220FF
[sma   ] TRACE 2024/10/27 16:45:54 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:54 recv 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:54 logout for 192.168.1.91:9522
[sma   ] TRACE 2024/10/27 16:45:54 send 192.168.1.91: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/27 16:45:54 recv 192.168.1.152: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]

IkarusPV, 2024-10-28:
Bei mir funktioniert das Netzladen der Hausbatterie auch nicht. Habe ebenfalls einen SHM 2.0 mit FW 2.15.6.R, BYD HVS 10.1 und einem SMA Hybridwechselrichter STP80SE und entsprechendem Template "sma-hybrid" mit Modbus für Wechselrichter und Batterie.

premultiply, 2024-10-29:
> SMA Hybridwechselrichter STP80SE

Völlig anderer Fall.

premultiply, 2024-10-29:
> Da passiert nix ausser sperren der Batterieentladung.

Dann muss irgendwas in den SBS-Einstellungen selbst noch falsch sein, denn die Funktion und Register sind im Prinzip für beide Fälle die gleichen.

premultiply, 2024-10-29:
> Betriebsart Wirkleistungsvorgabe steht auf aus

"Externe Wirkleistungsvorgabe" wäre korrekt.
und

"Externe Wirkleistungsvorgabe 2, Niedrige Priorität" auf "Aus".

Deepy1973, 2024-10-29:
Lass mich raten ändern kann ich das nur mit einem Grid Guard Code, richtig?

Wie bekomme ich den?

Deepy1973, 2024-10-29:
> > Betriebsart Wirkleistungsvorgabe steht auf aus
> 
> "Externe Wirkleistungsvorgabe" wäre korrekt. und
> 
> "Externe Wirkleistungsvorgabe 2, Niedrige Priorität" auf "Aus".

Beides so eingetragen. Keine Änderung :-(

premultiply, 2024-10-29:
Dann weiss ich es auch nicht.
Muss dann wohl noch was anderes sein.
Funktioniert hier jedenfalls einwandfrei.

Auch mit `evcc meter <Gerätename> -b charge|normal|hold` ausprobiert?

Deepy1973, 2024-10-29:
> Dann weiss ich es auch nicht. Muss dann wohl noch was anderes sein. Funktioniert hier jedenfalls einwandfrei.
> 
> Auch mit `evcc meter <Gerätename> -b charge|normal|hold` ausprobiert?

wo kann/muss ich das eingeben? Hab evcc als Raddon unter HA am laufen

Deepy1973, 2024-10-29:
> > > Betriebsart Wirkleistungsvorgabe steht auf aus
> > 
> > 
> > "Externe Wirkleistungsvorgabe" wäre korrekt. und
> > "Externe Wirkleistungsvorgabe 2, Niedrige Priorität" auf "Aus".
> 
> Beides so eingetragen. Keine Änderung :-(

Und wieso lädt er dann über den SHM 2.0 auf dem Netz ? Über Zeitfenster mit voller Energie 4.6kw.  Bei EVCC Netzladen über Preisfenster gestartet. Direkt wieder runter bis auf 24W.

Deepy1973, 2024-11-01:
was ich jetzt noch beobachten konnte:
Ist der Speicher im Standby weil er über Nacht leer gesaugt wurde, wacht er mit dem template Sam-sbs-modbus noch nicht einmal mehr auf.

Wechsel ich auf das "falsche" template Sma-sbs-15-25-modbus dann wacht er auf, stellt auf Batterie entladen und Sekunden später lädt er mit 2.6kw auf dem Netz.

Welche Einstellungen könnten hier im Batterie-Wechselrichter falsch gesetzt sein, um so ein Stranges verhalten zu zeigen?

Deepy1973, 2024-11-01:
Hier mal meine Batterieparameter im Wechselrichter SBS 5.0:

Nennkapazität der Batterie | 11.100 Wh |   |   |  
-- | -- | -- | -- | --
Batterietyp | Lithium-Ionen (Li-Ion) |   |   |  
Nennkapazität der Batterie [A] | 11.100 Wh |   |   |  
Nennkapazität der Batterie [B] | ------- |   |   |  
Nennkapazität der Batterie [C] | ------- |   |   |  
Batterie-Installationsmodus | Aus |   |   |  

Can-Test Modus	0		
(0 … 255)
Rücksetzen der DC Eingangs-Konfiguration	-------			
Anzahl der Batterien	3		
(1 … 3)
[Batteriesteller]
Maximale Ladeleistung des Batteriestellers	5.000 W			
Minimale Ladeleistung des Batteriestellers	0 W			
Maximale Entladeleistung des Batteriestellers	5.300 W			
Minimale Entladeleistung des Batteriestellers	0 W			
[Nutzungsbereiche]
Untere Grenze des Tiefentladeschutzbereichs vor Abschaltung [A]	2 %		
(0 % … 5 %)
Untere Grenze des Tiefentladeschutzbereichs vor Abschaltung [B]	2 %		
(0 % … 5 %)
Untere Grenze des Tiefentladeschutzbereichs vor Abschaltung [C]	2 %		
(0 % … 5 %)
Minimale Breite des Tiefentladeschutzbereichs [A]	3 %		
(1 % … 5 %)
Minimale Breite des Tiefentladeschutzbereichs [B]	1 %		
(1 % … 5 %)
Minimale Breite des Tiefentladeschutzbereichs [C]	1 %		
(1 % … 5 %)
Minimale Breite des Ersatzstrombereichs	0 %		
(0 % … 100 %)
Breite des Bereichs zur Erhaltung des Batterieladezustands [A]	3 %		
(1 % … 10 %)
Breite des Bereichs zur Erhaltung des Batterieladezustands [B]	5 %		
(1 % … 10 %)
Breite des Bereichs zur Erhaltung des Batterieladezustands [C]	5 %		
(1 % … 10 %)
[Externe Vorgabe]
Untere Entladegrenze bei Wirkleistungsvorgabe	0 %		
(0 % … 100 %)
Obere Entladegrenze bei Wirkleistungsvorgabe	100 %		
(0 % … 100 %)
[Leistungsprofil]
Breite des Nutzungsbereiches für die zeitgesteuerte Leistungsvorgabe	50,00 %		
(0,00 % … 100,00 %)
Breite des Nutzungsbereiches für den zeitgesteuerten Leistungssollwert	50,00 %		
(0,00 % … 100,00 %)

Deepy1973, 2024-11-01:
Meine Einstellungen zur Anlagen- und Gerätesteuerung
Sanftanlaufsrate P nach Netzfehler | 9 %/min |   |   |  
-- | -- | -- | -- | --
Sanftanlauf P nach Netzfehler | Ein |   |   |  
Sanftanlaufsrate P | 1.200 %/min |   |   |  
Anstiegsrate bei Einstrahlungsänderung | 10.000 %/min |   |   |  
Sanftanlauf P | Ein |   |   |  
Anstiegsrate bei Einstrahlungsänderung | Aus |   |   |  
Sanftanlaufsrate Q | 1.200 %/min |   |   |  
Sanftanlauf Q | Aus |   |   |  

[Wirkleistungsverfahren]
Netztrennung bei 0%-Wirkleistungsvorgabe	Nein			
Betriebsart Wirkleistungsvorgabe	Externe Wirkleistungsvorgabe			
[Manuelle Vorgabe
Wirkleistungsbegrenzung in W	4.600 W			
Wirkleistungsbegrenzung in %	100 %			
[Externe Vorgabe]
Timeout für ausbleibende Wirkleistungsvorgabe	30,00 min		
(0 s … 1 h)
Externe Wirkleistungsvorgabe, Rückfallwert der Wirkleistungsvorgabe	-4.600 W			
[Dynamik]
Externe Wirkleistungsvorgabe, Einstellzeit Sollwertfilter	3,00 s			
Externe Wirkleistungsvorgabe, Begrenzung der Änderungsrate	Ein			
Externe Wirkleistungsvorgabe, Anstiegsrate	0,50 %/s			
Externe Wirkleistungsvorgabe, Absenkungsrate	0,50 %/s			
Externe Wirkleistungsvorgabe, Sollwertfilter	Aus			
[Externe Vorgabe 2]
Externe Wirkleistungsvorgabe 2, Niedrige Priorität	Aus			
[Dynamik]
Externe Wirkleistungsvorgabe 2, Sollwertfilter	Aus			
Externe Wirkleistungsvorgabe 2, Einstellzeit Sollwertfilter	3,00 s			
Externe Wirkleistungsvorgabe 2, Begrenzung der Änderungsrate	Ein			
Externe Wirkleistungsvorgabe 2, Anstiegsrate	0,50 %/s			
Externe Wirkleistungsvorgabe 2, Absenkungsrate	0,50 %/s

Deepy1973, 2024-11-01:
Anlagensteuerung und Rückfallverhalten:

Quelle der Referenzmessung zur Blind-/Wirkleistungsregelung | Wechselrichter |
[Q, externe Vorgabe]
Externe Blindleistungsvorgabe, Rückfallverhalten	Rückfallwerte übernehmen			
Externe Blindleistungsvorgabe, Timeout	10 min		
( 1 s … 30 min)
Externe Blindleistungsvorgabe, Rückfallwert	0,00 %		
(-100,00 % … 100,00 %)

[cos φ, externe Vorgabe]
Externe cos φ-Vorgabe, Rückfallverhalten	Rückfallwerte übernehmen			
Externe cos φ-Vorgabe, Timeout	10 min		
( 1 s … 30 min)
Externe cos φ-Vorgabe, Rückfallwert des cos φ bei Wirkleistungsabgabe	1,0000		
(0,8000 … 1,0000)
Externe cos φ-Vorgabe, Rückfallwert der Erregungsart bei Wirkleistungsabgabe	Übererregt			
Externe cos φ-Vorgabe, Rückfallwert des cos φ bei Wirkleistungsaufnahme	1,0000		
(0,8000 … 1,0000)
Externe cos φ-Vorgabe, Rückfallwert der Erregungsart bei Wirkleistungsaufnahme	Übererregt
			
[Externe Wirkleistungsvorgabe]
Externe Wirkleistungsvorgabe, Rückfallverhalten	Werte beibehalten			
Externe Wirkleistungsvorgabe, Timeout	1 min		
( 1 s … 30 min)
Externe Wirkleistungsvorgabe, Rückfallwert der minimalen Wirkleistung	-4.600,00 W		
(-5.000,00 W … 0,00 W)
Externe Wirkleistungsvorgabe, Rückfallwert der maximalen Wirkleistung	4.600,00 W		
(0,00 W … 5.000,00 W)

[Externe Bezugsspannungsvorgabe]
Q(U), Rückfallbezugsspannung	1,00 p.u.			
Q(U), Rückfallverhalten bei ausbleibender Bezugsspannungsvorgabe	Rückfallwerte übernehmen			
Q(U), Timeout für ausbleibende Bezugsspannungsvorgabe	10,00 min		
(1 s … 30,00 min)

[Externe Wirkleistungsvorgabe 2]
Externe Wirkleistungsvorgabe 2, Rückfallverhalten	Werte beibehalten			
Externe Wirkleistungsvorgabe 2, Rückfallwert der minimalen Wirkleistung	-4.600,00 W		
(-5.000,00 W … 0,00 W)
Externe Wirkleistungsvorgabe 2, Rückfallwert der maximalen Wirkleistung	4.600,00 W		
(0,00 W … 5.000,00 W)
Externe Wirkleistungsvorgabe 2, Timeout	1 min		
( 1 s … 30 min)

[Netzanschlusspunkt]
Schieflastbegrenzung	Ein			
Maximale Schieflast	4.600 W			
Betriebsart der Wirkleistungsbegrenzung am Netzanschlusspunkt	Aus			
Eingestellte Wirkleistungsgrenze am Netzanschlusspunkt	4.600 W		
(0 W … 500.000 W)
Eingestellte Wirkleistungsgrenze am Netzanschlusspunkt	100 %		
(0 % … 100 %)

[Leistungsprofil
Aktivieren der Funktion Leistungsprofil	inaktiv

Deepy1973, 2024-11-02:
> > Da passiert nix ausser sperren der Batterieentladung.
> 
> Dann muss irgendwas in den SBS-Einstellungen selbst noch falsch sein, denn die Funktion und Register sind im Prinzip für beide Fälle die gleichen.

Vielleicht magst Du deine Einstellungen auch posten, dann können wir eventuelle sehen "Wo der Hase im Pfeffer sitzt"

Deepy1973, 2024-11-02:
Was ich noch beobachtet habe mit dem richtigen template für meinen SBS 5.0 steht im Webportal des WR:

Bei **nicht** Netzladen
[Betrieb]
Quelle der maximalen Wirkleistungsvorgabe	Wirkleistungsvorgabe			
Quelle der minimalen Wirkleistungsvorgabe	Wirkleistungsvorgabe

Bei **Netzladen**
[Betrieb]
Quelle der maximalen Wirkleistungsvorgabe	W**irkleistungsvorgabe	2**		
Quelle der minimalen Wirkleistungsvorgabe	Wirkleistungsvorgabe
Aber nix passiert.

Wechsle ich auf das template Sma-sbs-15-25-modbus bleibt auch beim **Netzladen** 
[Betrieb]
Quelle der maximalen Wirkleistungsvorgabe	**Wirkleistungsvorgabe**	
eingetragen. 
Aber er lädt aus dem Netz. Wenn auch nur mit 2.6kw.

Wenn ich die Einträge im WR im Webportal betrachte, sehe ich keine Unterschiede.

irqnet, 2024-11-06:
Ich glaube bei mir ist es exakt das gleiche "Problem".

SBS 5.0 - Batterie im Standby da über Nacht leer und nix passiert, obwohl ich unterm Schwellwert fürs Netzladen liege.

Deepy1973, 2024-11-06:
> Ich glaube bei mir ist es exakt das gleiche "Problem".
> 
> SBS 5.0 - Batterie im Standby da über Nacht leer und nix passiert, obwohl ich unterm Schwellwert fürs Netzladen liege.
> 
> Mit dem Template "sma-sbs-15-25-modbus" funktioniert es auch bei mir sofort (mit 2,6 kw)

Liegt es vielleicht am FW stand des BWR? Bei mir ist es die 4.4.11.R beim SMA SBS 5.0. Welchen FW Stand hast du?

irqnet, 2024-11-06:
> Liegt es vielleicht am FW stand des BWR? Bei mir ist es die 4.4.11.R beim SMA SBS 5.0. Welchen FW Stand hast du?

Bin auch auf 4.4.11.R, habe das Update vor ein paar Wochen gemacht. Bin mir aber sicher das es mit dem "richtigen" Template schon mal funktioniert hat.

Was mir gerade bei dem Test auffällt, nachdem die Batterie auf ca. 4% geladen war mit dem "sma-sbs-15-25-modbus" Template, habe ich die Konfiguration wieder zurück gedreht und nach dem Neustart von evcc lädt er jetzt mit 4,6 kw.

Mir scheint es also dass das Template den WR/Batterie nicht aus dem Standby bekommt wenn sie auf 0% ist.

Deepy1973, 2024-11-06:
> Bin auch auf 4.4.11.R, habe das Update vor ein paar Wochen gemacht. Bin mir aber sicher das es mit dem "richtigen" Template schon mal funktioniert hat.
>
> Was mir gerade bei dem Test auffällt, nachdem [...]

Das bei mir def. nicht so. Lädt er mit dem „falschen“ Template mit 2.6kw und ich dreh zurück auf das Template SMA-sbs-modbus dann stoppt er nach Neustart von evcc wieder das Netzladen. 

Magst du mal deine Parameter-Einstellungen des BWR posten. Müssten dann ja zu meinen anders sein…

irqnet, 2024-11-06:
[image]

[image]

Deepy1973, 2024-11-06:
@irqnet danke fürs teilen. Dachte aber mehr an diese Details 
[WR.Einstellungen.pdf](https://github.com/user-attachments/files/17647824/WR.Einstellungen.pdf)

irqnet, 2024-11-25:
> @irqnet danke fürs teilen. Dachte aber mehr an diese Details 
> [WR.Einstellungen.pdf](https://github.com/user-attachments/files/17647824/WR.Einstellungen.pdf)
> 
> 

Ich denke die Einstellungen sind es nicht.
Hatte diese Nacht wieder das Problem das nicht geladen wurde.

Ich denke das ist ein Problem des template wenn die Batterie auf 0% ist und dann nicht aufgeweckt wird.

Deepy1973, 2024-11-25:
> Ich denke die Einstellungen sind es nicht. Hatte diese Nacht wieder das Problem das nicht geladen wurde.
>
> Ich denke das ist ein Problem des template wenn die Batterie auf 0% ist und dann nicht [...]

Netzladen soll jetzt mit dem aktuellen Nightly Release funktionieren. Schon mal ausprobiert?

irqnet, 2024-11-25:
Funktioniert tatsächlich, hatte ich noch nicht gesehen. Gerade das Update gemacht :)