evcc-io/evcc #16911: Fahrzeugstatus ERROR - MG4

lmfgpeace, 2024-10-27:
### Describe the bug

Fahrzeugstatus kann beim MG4 nicht mehr abgerufen werden. Error siehe Log.

### Steps to reproduce

1. update from 0.13x.xx to v0.131.2
2. now you have to problem :)
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 5s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3MjkyNzQ0MDAsImlhdCI6MTcyODQ5NjgwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.gnfLjIPCnFt3TmCUhoXCmA3nHK5RkQcOrLIejCQfc4w
# trial token, valid until 2024-10-18
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3Mjk4NjQ4MDAsImlhdCI6MTcyOTA4NzIwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.3iF50HEbpf7hF7nob075jJxseDt4c2rFeSS-ZHpxzuE

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri:
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
[... 246 lines cut ...]
  #   tags: <tags>
```

### Log details

```text
[main ] ERROR 2024/10/27 17:20:47 vehicle status: Get "https://gateway-mg-eu.soimt.com/api.app/v1/vehicle/charging/mgmtData?vin=d95ee66d4a7b24d57c9fa15b6540f0d736acb349ecaab527145943bf18243076": net/http: request canceled (Client.Timeout exceeded while awaiting headers)

[lp-1 ] ERROR 2024/10/27 17:28:07 vehicle soc: Get "https://gateway-mg-eu.soimt.com/api.app/v1/vehicle/charging/mgmtData?vin=d95ee66d4a7b24d57c9fa15b6540f0d736acb349ecaab527145943bf18243076": net/http: request canceled (Client.Timeout exceeded while awaiting headers)

[lp-1 ] ERROR 2024/10/27 17:21:02 vehicle odometer: Get "https://gateway-mg-eu.soimt.com/api.app/v1/vehicle/charging/mgmtData?vin=d95ee66d4a7b24d57c9fa15b6540f0d736acb349ecaab527145943bf18243076": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[lp-1 ] ERROR 2024/10/27 17:21:02 vehicle soc: Get "https://gateway-mg-eu.soimt.com/api.app/v1/vehicle/charging/mgmtData?vin=d95ee66d4a7b24d57c9fa15b6540f0d736acb349ecaab527145943bf18243076": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.2

lmfgpeace, 2024-10-27:
[image]

CarG70, 2024-10-27:
Ich kann Dir zwar nicht helfen aber Du hast oben gerade Deine Zugangsdaten zu MG verbrannt. Ändere die mal schnell...

lmfgpeace, 2024-10-27:
> Ich kann Dir zwar nicht helfen aber Du hast oben gerade Deine Zugangsdaten zu MG verbrannt. Ändere die mal schnell...

ja geil überhaupt nicht nachgedacht wie dumm - danke!

d3ni3, 2024-10-30:
Sicher das es kein Netzwerk- / Verfügbarkeitsproblem ist?
Die SAIC-Server sind nicht gerade für ihre Zuverlässigkeit bekannt. Wenn ich heute Abend wieder zu Hause bin, schaue ich mal wie das bei mir aussieht.

lmfgpeace, 2024-10-30:
> Sicher das es kein Netzwerk- / Verfügbarkeitsproblem ist?
> 
> Die SAIC-Server sind nicht gerade für ihre Zuverlässigkeit bekannt. Wenn ich heute Abend wieder zu Hause bin, schaue ich mal wie das bei mir aussieht.

Ziemlich sicher. Unmittelbar vor dem Update war alles schick.

d3ni3, 2024-10-30:
Okay, dann müsste ich das ja nachstellen können.

d3ni3, 2024-10-30:
Habe den MG soeben angeschlossen, bei mir funktioniert der Abruf.
Installiert habe ich die  v0.131.3.

closed 2024-10-31 by andig

lmfgpeace, 2024-11-17:
> Habe den MG soeben angeschlossen, bei mir funktioniert der Abruf. Installiert habe ich die v0.131.3.

@d3ni3 
Funktioniert bei dir die Übermittlung des vehicleSoC beim Laden des MG auch nur so sporadisch? Mal wird der vehicleSoC direkt angezeigt, mal gar nicht.

Das nervt irgendwie ein bisschen.

Haste schonmal hier dran rumgespielt?:

cache:  # Zeitintervall nach dem Daten erneut vom Fahrzeug abgefragt werden (optional)

oder
poll: # Definiert, wie die Fahrzeug APIs benutzt werden um aktuelle Informationen des Fahrzeugs abzurufen.

d3ni3, 2024-11-18:
Ich achte da nicht immer so genau drauf. Gestern, beim Laden, wurde mir der SoC angezeigt. In der Konfigurationssicht werden aber irgendwie immer seltsame Werte angezeigt, auf dem Dashboard passt es dann aber. Was ich aber sagen kann ist, das der Webservice von SAIC öfter mal nicht erreichbar ist oder einfach keine Werte liefert. Die App zeigt dann auch keine Werte an. Dem MG-Forum nach kommt das relativ häufig vor.