evcc-io/evcc #16912: evcc doesnt start after update to 0.131.2

TomF79, 2024-10-27:
### Describe the bug

evcc doesnt start after update to 0.131.2

No problems with version 0.131.1

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
meters:
  - name: my_grid
    type: template
    template: fronius-gen24
    usage: grid
    host: 192.168.50.40 # IP-Adresse oder Hostname
    port: 502 # Port (optional)
  - name: my_pv
    type: template
    template: fronius-gen24
    usage: pv
    host: 192.168.50.40 # IP-Adresse oder Hostname
    port: 502 # Port (optional)
  - name: my_battery
    type: template
    template: fronius-gen24
    usage: battery
    host: 192.168.50.40 # IP-Adresse oder Hostname
    port: 502 # Port (optional)
    capacity: 12.8 # Akkukapazität in kWh (optional)
```

### Log details

```text
[main ] INFO 2024/10/27 17:42:47 evcc 0.131.2
[main ] INFO 2024/10/27 17:42:47 using config file: /config/evcc.yaml
[db ] INFO 2024/10/27 17:42:47 using sqlite database: /data/evcc.db
[main ] INFO 2024/10/27 17:42:47 listening at :7070
[main ] FATAL 2024/10/27 17:42:57 cannot create meter 'my_grid': cannot create meter type 'template': cannot create meter type 'custom': power: read tcp 192.168.50.3:41882->192.168.50.40:502: i/o timeout
[main ] FATAL 2024/10/27 17:42:57 will attempt restart in: 15m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.2

dm82m, 2024-10-27:
Same here:

```
[main  ] FATAL 2024/10/27 19:08:14 cannot create meter 'grid': cannot create meter type 'template': cannot create meter type 'custom': power: read tcp 192.168.178.4:54862->192.168.178.5:502: i/o timeout
[main  ] FATAL 2024/10/27 19:08:14 will attempt restart in: 15m0s
```

andig, 2024-10-27:
Oh yeah, missed the grid meters modbus id 200. Fix to follow. Sorry for the inconvenience.

andig, 2024-10-28:
Fixed in master, nightly in 30mins.

TomF79, 2024-10-28:
> Fixed in master, nightly in 30mins.

Bitte auch für uns HA-User ein neues Release.

TomF79, 2024-10-28:
> Fixed in master, nightly in 30mins.

@andig 

Bitte auch für uns HA-User ein neues Release.

dm82m, 2024-10-28:
Same issue with nightly:

```
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/config/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/10/28 08:33:57 evcc 0.131.2 (6505a046)
[main  ] INFO 2024/10/28 08:33:57 using config file: /config/evcc.yaml
[main  ] INFO 2024/10/28 08:33:57 listening at :7070
[main  ] FATAL 2024/10/28 08:33:58 cannot create meter 'grid': cannot create meter type 'template': cannot create meter type 'custom': power: sunspec model not found: [{203 0 W} {213 0 W}]
[main  ] FATAL 2024/10/28 08:33:58 will attempt restart in: 15m0s
```

benesolar, 2024-10-28:
@dm82m 
are you using 3phase inverter and grid meter?

speedst3r, 2024-10-28:
Similar issue with Fronius Symo (3-phase inverter) using `sunspec-inverter` template type. 0.131.1 works without issue.

```
[main  ] INFO 2024/10/28 08:02:49 evcc 0.131.2
[main  ] INFO 2024/10/28 08:02:49 using config file: /etc/evcc.yaml
[main  ] INFO 2024/10/28 08:02:49 listening at :7070
[main  ] FATAL 2024/10/28 08:02:57 cannot create meter 'pv2': cannot create meter type 'template': cannot create meter type 'custom': power: sunspec model not found: [{101 0 W} {111 0 W} {103 0 W} {113 0 W}]
[main  ] FATAL 2024/10/28 08:02:57 will attempt restart in: 15m0s
```

Config:
```yaml
meters:
- type: template
  template: sunspec-inverter
  usage: grid
  host: 192.168.50.15
  name: grid1
  modbus: tcpip
  id: 240
- type: template
  template: sunspec-inverter
  usage: pv
  host: 192.168.50.15
  name: pv2
  modbus: tcpip
  id: 1
```

dm82m, 2024-10-28:
> @dm82m are you using 3phase inverter and grid meter?

Think so, yes

dm82m, 2024-10-28:
I restored the old 0.113.1 db and reverted the addon to 0.113.1 to be able to charge now. It’s sunny here.

kad49007, 2024-10-28:
Same here with Fronius Inverter, also lated nightly (30mins ago) not working. Went back to 0.131.1 at the moment

andig, 2024-10-28:
@speedst3r opened https://github.com/evcc-io/evcc/issues/16924 for the Symo

kjs-DE, 2024-10-28:
After 0.132.2 update evcc doesn't talk with my EM2GO Wallbox any more. Naturally it isn't able to recognize that a vehicle is connected as connection to the wallbox is always refused. Downgraded to 0.131.1 and everything works fine again.

andig, 2024-10-28:
@kjs-DE please open new issue. This is not about chargers.

timpetersson81, 2024-10-28:
Is version 0.1.31.3 released when the bug I fixed?

Syonis, 2024-10-28:
I've worked around this by temporarily choose SunSpec (I use HomeAssistant):
```
-   name: EVU
    type: template
    template: fronius-solarapi-v1 # fronius-gen24 mit port geht seit Version 0.131.2 am 28.10.2024 nicht mehr...
    host: 192.168.5.10
    #port: 502
    usage: grid
```

andig, 2024-10-28:
@dm82m the new message looks very similar to https://github.com/evcc-io/evcc/issues/16924. I assume there's a problem with the sunspec model caching we've added to improve startup times. Any chance you could give info@evcc.io remote access to the inverter (VPN, Tailscale, Wireguard, SSH)?

grappa24, 2024-10-28:
same here with Fronius GEN24:
cannot create meter 'grid1': cannot create meter type 'template': cannot create meter type 'custom': power: read tcp 192.168.178.56:56730->192.168.178.129:502: i/o timeout

lucode, 2024-10-28:
Same here with Victron inverter and OCPP charger. No specific error message, just won't start and tries again.

dm82m, 2024-10-28:
> @dm82m the new message looks very similar to #16924. I assume there's a problem with the sunspec model caching we've added to improve startup times. Any chance you could give [info@evcc.io](mailto:info@evcc.io) remote access to the inverter (VPN, Tailscale, Wireguard, SSH)?

You have tailscale access now @andig

jpwenzel, 2024-10-28:
Hate to chime in here — observing a similar issue with a Sungrow inverter (modbus) and OCPP wallbox.
Downgraded to 0.131.1 and the same setup works.

sad91188, 2024-10-28:
Eastron and Growatt its same, Gridmeter timeout
Backup to 0.129 its running

andig, 2024-10-28:
Hopefully fixed in https://github.com/evcc-io/evcc/commit/4458043e92a11834ffc6793cec46c3d74281cc43. Please try nightly in 30min.

dm82m, 2024-10-28:
Works in current nightly.

if anyone on home assistants wants to use the nightly, you can stop the original addon and use my nightly till 0.113.3 is released. Ensure that you do the same configuration: path to config, path to db, port and so on.

https://github.com/dm82m/hassio-addons

andig, 2024-10-28:
Will release asap, this is a bad breakage. Sorry.

closed 2024-10-28 by andig

timpetersson81, 2024-10-28:
I still have the issue in 0.31.3

dm82m, 2024-10-28:
> I still have the issue in 0.31.3

0.131.3 works for me. We need logs and potentially also a new issue for it.

timpetersson81, 2024-10-28:
> > I still have the issue in 0.31.3
> 
> 0.131.3 works for me. We need logs and potentially also a new issue for it.

I postet it here, but has Bin closed

lucode, 2024-10-29:
Same here with 0.131.3 , not working yet:
`cannot create charger 'Carport': cannot create charger type 'template': cannot create charger type 'ocpp': timeout`

andig, 2024-10-29:
This issue is not about chargers, please don't hijack it.