evcc-io/evcc #16924: Fronius Symo (3-phase inverter): `sunspec-inverter` broken

andig, 2024-10-28:
Similar issue with Fronius Symo (3-phase inverter) using `sunspec-inverter` template type. 0.131.1 works without issue.

```
[main  ] INFO 2024/10/28 08:02:49 evcc 0.131.2
[main  ] INFO 2024/10/28 08:02:49 using config file: /etc/evcc.yaml
[main  ] INFO 2024/10/28 08:02:49 listening at :7070
[main  ] FATAL 2024/10/28 08:02:57 cannot create meter 'pv2': cannot create meter type 'template': cannot create meter type 'custom': power: sunspec model not found: [{101 0 W} {111 0 W} {103 0 W} {113 0 W}]
[main  ] FATAL 2024/10/28 08:02:57 will attempt restart in: 15m0s
```

Config:
```yaml
meters:
- type: template
  template: sunspec-inverter
  usage: grid
  host: 192.168.50.15
  name: grid1
  modbus: tcpip
  id: 240
- type: template
  template: sunspec-inverter
  usage: pv
  host: 192.168.50.15
  name: pv2
  modbus: tcpip
  id: 1
```

_Originally posted by @speedst3r in https://github.com/evcc-io/evcc/issues/16912#issuecomment-2440811773_

unsignedpixel, 2024-10-28:
I ran into this problem *after* updating to 0.131.2, reverting to 131.1 fixed it for me (Fronius Gen24 Symo).

andig, 2024-10-28:
The Symo is 3p, right?

speedst3r, 2024-10-28:
> The Symo is 3p, right?

Correct, specifically this model is a Symo 5.0-3-M.

Macrisu, 2024-10-28:
Symo 3P is not working (two installations).
Fronius gen24 is offline for renovation.

speedst3r, 2024-10-28:
```
[main  ] INFO 2024/10/28 14:48:38 evcc 0.131.2
--------- Model 1 common ---------
Mn            Fronius    string
Md       Symo 5.0-3-M    string
Opt          3.31.1-7    string
Vr           0.3.30.1    string
SN           28093020    string
DA               1.00    uint16
Pad            0x0067    pad
--------- Model 103 inverter ---------
A                    0.00    uint16
AphA                 0.00    uint16
AphB                 0.00    uint16
AphC                 0.00    uint16
A_SF                   -2    sunssf
PPVphAB              0.00    uint16
PPVphBC              0.00    uint16
PPVphCA              0.00    uint16
PhVphA               0.00    uint16
PhVphB               0.00    uint16
PhVphC               0.00    uint16
V_SF                   -2    sunssf
W                    0.00    int16
W_SF                   -2    sunssf
Hz                   0.00    uint16
Hz_SF                  -2    sunssf
VA                    n/a    int16
VA_SF                 n/a    sunssf
VAr                   n/a    int16
VAr_SF                n/a    sunssf
PF                    n/a    int16
PF_SF                 n/a    sunssf
WH            75438296.00    acc32
WH_SF                   0    sunssf
DCA                   n/a    uint16
DCA_SF                n/a    sunssf
DCV                   n/a    uint16
DCV_SF                n/a    sunssf
DCW                  0.00    int16
DCW_SF                 -2    sunssf
TmpCab                n/a    int16
TmpSnk                n/a    int16
TmpTrns               n/a    int16
TmpOt                 n/a    int16
Tmp_SF                n/a    sunssf
St                      3    enum16
StVnd                   3    enum16
Evt1           0x00000000    bitfield32
Evt2           0x00000000    bitfield32
EvtVnd1        0x00000000    bitfield32
EvtVnd2        0x00000000    bitfield32
EvtVnd3        0x00000000    bitfield32
EvtVnd4        0x00000000    bitfield32
--------- Model 120 nameplate ---------
DERTyp                          4    enum16
WRtg                      5000.00    uint16
WRtg_SF                         1    sunssf
VARtg                     5000.00    uint16
VARtg_SF                        1    sunssf
VArRtgQ1                  5000.00    int16
VArRtgQ2                      n/a    int16
VArRtgQ3                      n/a    int16
VArRtgQ4                 -5000.00    int16
VArRtg_SF                       1    sunssf
ARtg                         7.20    uint16
ARtg_SF                        -2    sunssf
PFRtgQ1                     -0.80    int16
PFRtgQ2                       n/a    int16
PFRtgQ3                       n/a    int16
[... 94 lines cut ...]
N                       2    count
TmsPer                n/a    uint16
-- Block 1 --
ID                 1.00    uint16
IDStr          String 1    string
DCA                 n/a    uint16
DCV                 n/a    uint16
DCW                 n/a    uint16
DCWH        46306100.00    acc32
Tms        783467323.00    uint32
Tmp                 n/a    int16
DCSt                  3    enum16
DCEvt               n/a    bitfield32
-- Block 2 --
ID                 2.00    uint16
IDStr          String 2    string
DCA                0.00    uint16
DCV                0.00    uint16
DCW                0.00    uint16
DCWH        29132100.00    acc32
Tms        783467323.00    uint32
Tmp                 n/a    int16
DCSt                  3    enum16
DCEvt               n/a    bitfield32
```

andig, 2024-10-28:
Thank you. Seems the error is wrong:

> power: sunspec model not found: [{101 0 W} {111 0 W} {103 0 W} {113 0 W}]

Because 103 exists just fine. Now we'll need to find out why it is not found. Could you kindly share inverter access with info@evcc.io (VPN, Tailscale, Wireguard, SSH)?

Macrisu, 2024-10-28:
I dont have access now, but I get:

`cannot create meter type 'template': cannot create meter type 'custom': power: sunspec model not found: [{201 0 W} {211 0 W} {203 0 W} {213 0 W}] `

andig, 2024-10-28:
@Macrisu that's different. Do you also have the problem with `pv`?

Macrisu, 2024-10-28:
> @Macrisu that's different. Do you also have the problem with `pv`?

Not working since latest  version.
[image]

Macrisu, 2024-10-28:
**Symo 1 settings:**

```
meters:
- name: grid_v28
  type: template
  template: sunspec-inverter
  id: 240
  host: 192.168.28.16
  port: 502
  usage: grid
  modbus: tcpip
  
- name: pv_v28
  type: template
  template: sunspec-inverter
  id: 1
  host: 192.168.28.16
  port: 502
  usage: pv
  modbus: tcpip 
```

error:

` cannot create meter type 'template': cannot create meter type 'custom': power: sunspec model not found: [{201 0 W} {211 0 W} {203 0 W} {213 0 W}]`

**Symo 2 settings:**

```
meters:
  
- name: grid
  type: template
  template: sunspec-inverter
  id: 240
  host: 192.168.1.197
  port: 502
  usage: grid
  modbus: tcpip
  
- name: pv
  type: template
  template: sunspec-inverter
  id: 1
  host: 192.168.1.197
  port: 502
  usage: pv
  modbus: tcpip  
```

`[main  ] FATAL 2024/10/27 20:57:32 cannot create meter 'grid': cannot create meter type 'template': cannot create meter type 'custom': power: sunspec model not found: [{201 0 W} {211 0 W} {203 0 W} {213 0 W}]`

<br class="Apple-interchange-newline">

Macrisu, 2024-10-28:
Here it is for Symo 1:

```
evcc sunspec 192.168.28.16:502 --id 1
[main  ] INFO 2024/10/28 16:45:28 evcc 0.131.2
--------- Model 1 common ---------
Mn            Fronius    string
Md       Symo 8.2-3-M    string
Opt          3.31.1-5    string
Vr           0.3.30.1    string
SN           31329509    string
DA               1.00    uint16
Pad            0x0071    pad
--------- Model 113 inverter ---------
A                    0.75    float32
AphA                 0.25    float32
AphB                 0.25    float32
AphC                 0.25    float32
PPVphAB             399.4    float32
PPVphBC         398.80002    float32
PPVphCA         397.80002    float32
PhVphA              230.6    float32
PhVphB              229.6    float32
PhVphC          230.40001    float32
W                     177    float32
Hz                     50    float32
VA                    177    float32
VAr                     0    float32
PF                    100    float32
WH              2.632e+07    float32
DCA                   n/a    float32
DCV                   n/a    float32
DCW                193.84    float32
TmpCab                n/a    float32
TmpSnk                n/a    float32
TmpTrns               n/a    float32
TmpOt                 n/a    float32
St                      4    enum16
StVnd                   4    enum16
Evt1           0x00000000    bitfield32
Evt2           0x00000000    bitfield32
EvtVnd1        0x00000000    bitfield32
EvtVnd2        0x00000000    bitfield32
EvtVnd3        0x00000000    bitfield32
EvtVnd4        0x00000000    bitfield32
--------- Model 120 nameplate ---------
DERTyp                          4    enum16
WRtg                      8200.00    uint16
WRtg_SF                         1    sunssf
VARtg                     8200.00    uint16
VARtg_SF                        1    sunssf
VArRtgQ1                  8200.00    int16
VArRtgQ2                      n/a    int16
VArRtgQ3                      n/a    int16
VArRtgQ4                 -8200.00    int16
VArRtg_SF                       1    sunssf
ARtg                        11.40    uint16
ARtg_SF                        -2    sunssf
PFRtgQ1                     -0.80    int16
PFRtgQ2                       n/a    int16
PFRtgQ3                       n/a    int16
PFRtgQ4                      0.80    int16
PFRtg_SF                       -3    sunssf
WHRtg                         n/a    uint16
WHRtg_SF                      n/a    sunssf
AhrRtg                        n/a    uint16
AhrRtg_SF                     n/a    sunssf
MaxChaRte                     n/a    uint16
MaxChaRte_SF                  n/a    sunssf
MaxDisChaRte                  n/a    uint16
[... 85 lines cut ...]
N                       2    count
TmsPer                n/a    uint16
-- Block 1 --
ID                 1.00    uint16
IDStr          String 1    string
DCA                0.33    uint16
DCV              519.40    uint16
DCW              171.40    uint16
DCWH        20941500.00    acc32
Tms        783449239.00    uint32
Tmp                 n/a    int16
DCSt                  4    enum16
DCEvt               n/a    bitfield32
-- Block 2 --
ID                 2.00    uint16
IDStr          String 2    string
DCA                0.13    uint16
DCV              172.60    uint16
DCW               22.44    uint16
DCWH         5378570.00    acc32
Tms        783449239.00    uint32
Tmp                 n/a    int16
DCSt                  4    enum16
DCEvt               n/a    bitfield32
```

andig, 2024-10-28:
Hopefully fixed in https://github.com/evcc-io/evcc/commit/4458043e92a11834ffc6793cec46c3d74281cc43. Please try nightly in 30min.

neupertd, 2024-10-28:
Hallo,
mit der Nightly klappt es wieder.

@andig Falls du eine Möglichkeit zum debuggen brauchst kannst Du dich bei mir melden.

closed 2024-10-28 by andig

sllitzenberger, 2025-02-08:
Hallo, 
ich bekomme mit der sunspec und dem fronius-template den nachfolgenden Fehler:

starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/02/08 20:14:21 evcc 0.133.0
[main  ] INFO 2025/02/08 20:14:21 using config file: /config/evcc.yaml
[db    ] INFO 2025/02/08 20:14:21 using sqlite database: /data/evcc.db
[main  ] INFO 2025/02/08 20:14:22 listening at :7070
[main  ] FATAL 2025/02/08 20:14:22 cannot create meter 'battery': cannot create meter type 'template': cannot create meter type 'custom': power: sunspec model not found: [{101 0 W} {111 0 W} {103 0 W} {113 0 W}]
[main  ] FATAL 2025/02/08 20:14:22 will attempt restart in: 15m0s

das ganze läuft auf einer homeassistant-instanz. wenn ich über experimental im config-mode die geräte hinzufüge funktioniert nur die API-Schnittstelle. ich habe mehrere modbus-adressen getestet. 240, 200, 241.

slxreaper, 2025-02-08:
Ich habe heute den glauben Fehler beim fronius symo. Ich hab auch die verbindung über einen modbus-proxy versucht. Jedes Mal der gleiche Fehler.  
Über die Experimental-Einstellungen in der Konfiguration konnte ich nur eine Verbindung mit der api-Schnittstelle herstellen. Über die Yaml-Datei funktioniert dies jedoch wiederum nicht.

andig, 2025-02-08:
@sixreaper issue has been fixed. Please open new issue with full context.