evcc-io/evcc #16931: evcc läuft nicht mehr auf altem iPad Air 1 seit Version 0.132.2

jekruege, 2024-10-28:
### Describe the bug

evcc läuft nicht mehr im alten Safari Browser auf dem iPad Air 1 seit der neuesten Version 0.132.2. Es kommt nur ein leeres Browserfenster.
[image]

### Steps to reproduce

1. neues Version installiert
2.
3.
...

### Configuration details

```yaml
not relevant
```

### Log details

```text
not relevant
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.131.2

naltatis, 2024-10-28:
@jekruege Magst du mal Details (bspw. iOS Version) posten?

jekruege, 2024-10-28:
Danke schon mal!
IOS 12.5.7

maduuto, 2024-10-28:
Das hatte ich mit Chrome auch auf Smartphone. Musste den Cache löschen dann ging es wieder. Vielleicht hilft das dir?

HeikoH78, 2024-10-29:
Hab gerade bei mir den Cache gelöscht, hat leider nicht geholfen.
Antwort auch auf: #16943

naltatis, 2024-10-29:
> IOS 12.5.7

@jekruege Danke. Dann schau ich mir den an.

> Safari ist ebenso wie Firefox betroffen.

@HeikoH78 Ist das Firefox auf dem iPad? Wenn nicht hätt ich auch da gerne einmal die Browser Version vom Firefox.

HeikoH78, 2024-10-29:
> @HeikoH78 Ist das Firefox auf dem iPad? Wenn nicht hätt ich auch da gerne einmal die Browser Version vom Firefox.

Ja, Firefox auf dem iPad. Auf allen anderen (neueren) Geräten funktioniert es einwandfrei.
Die Version von Firefox ist Daylight 36.0 (5732).

jekruege, 2024-10-30:
Aber es wird über dieselbe Rendering-Machine generiert wie beim Safari. Apple erlaubt erst seit iOS18.2 eine eigene Rendering Machine.

naltatis, 2024-11-02:
Zwischenstand der Analyse: bauen aktuell ein dediziertes JavaScript-Bundle um alte Geräte zu unterstützen. Das wird auch weiterhin korrekt erzeugt. Aus irgend einem Grund wird es aber auf dem iOS 12 Safari nicht genutzt. Durch welche Änderung das genau passiert ist habe ich noch nicht nachgestellt. Ich vermute eines der npm-Dependency Upgrades.

closed 2024-11-03 by andig

HeikoH78, 2024-11-04:
Danke euch!

jekruege, 2024-11-11:
Funktioniert in neuster Version! Besten Dank!

naltatis, 2024-11-11:
@jekruege sehr schön! Wir haben leider aktuell keinen einfachen und guten Weg, die Funktionsweise in alten Browsern automatisiert zu testen (wäre ggf. via BrowserStack möglich). iOS 12 bewegt sich da leider in so einem Mittelfeld. Der Browser kann viele moderne Funktionen und bekommt daher nicht das große Legacy-JavaScript, einige Dinge kann er dann aber doch noch nicht.
Ich werd bei neuen größeren Funktionen bewusster drauf achten. Wenn doch mal was durchrutscht - gerne wieder die Hand heben, dann machen wir das wieder heile 😄

unstressable, 2024-11-24:
Selbes Problem bei mir.
Seit Version 0.131.6 lädt es am iPhone und iPad nicht mehr.
Auch Version 0.137.7 läuft nicht.

HeikoH78, 2025-02-17:
Ich glaube das Problem tritt gerade wieder auf mit der 0.200.1.

naltatis, 2025-02-17:
Argh. @HeikoH78 mach gerne ein neues Issue dazu auf. Genaue iOS Version und ab wann (0.200.0, 0.200.1, früher?) das auftritt. Das erleichtert die Fehlersuche.

HeikoH78, 2025-02-17:
> Argh. [@HeikoH78](https://github.com/HeikoH78) mach gerne ein neues Issue dazu auf. Genaue iOS Version und ab wann (0.200.0, 0.200.1, früher?) das auftritt. Das erleichtert die Fehlersuche.

#18894