evcc-io/evcc #16941: evcc mit 0.131.3:  Kann keine Verbindung zum Server aufbauen

grappa24, 2024-10-28:
### Describe the bug

sorry, aber mit 0.131.3 bekomme ich erst gar keine Verbindung mehr zum meinem evcc Server

### Steps to reproduce

1. update to 0.131.3
2.
3.
...

### Configuration details

```yaml
chargers:
- type: template
  template: ocpp-fronius-wattpilot
  name: wallbox5
```

### Log details

```text
"Firefox kann keine Verbindung zu dem Server unter 192.168.178.56:7070 aufbauen."

Hier der Status:
Oct 28 23:32:57 raspberrypi evcc[1745]: [main  ] INFO 2024/10/28 23:32:57 evcc 0.131.3
Oct 28 23:32:57 raspberrypi evcc[1745]: [main  ] INFO 2024/10/28 23:32:57 using config file: /etc/evcc.yaml
Oct 28 23:32:57 raspberrypi evcc[1745]: [db    ] INFO 2024/10/28 23:32:57 using sqlite database: /var/lib/evcc/evcc.db
Oct 28 23:32:58 raspberrypi evcc[1745]: [mqtt  ] INFO 2024/10/28 23:32:58 connecting evcc at tcp://192.168.178.44:1883
Oct 28 23:32:58 raspberrypi evcc[1745]: [mqtt  ] DEBUG 2024/10/28 23:32:58 tcp://192.168.178.44:1883 connected
Oct 28 23:32:58 raspberrypi evcc[1745]: [main  ] INFO 2024/10/28 23:32:58 listening at :7070
Oct 28 23:32:58 raspberrypi evcc[1745]: [ocpp-1] DEBUG 2024/10/28 23:32:58 waiting for chargepoint: 5m0s
Oct 28 23:33:00 raspberrypi evcc[1745]: [ocpp  ] INFO 2024/10/28 23:33:00 charge point connected, registering: 91027406
Oct 28 23:37:58 raspberrypi evcc[1745]: [main  ] FATAL 2024/10/28 23:37:58 cannot create charger 'wallbox5': cannot create charger type 'template': cannot create charger typ
Oct 28 23:37:58 raspberrypi evcc[1745]: [main  ] FATAL 2024/10/28 23:37:58 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.3

TomF79, 2024-10-29:
Same here.

`Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/10/29 05:47:01 evcc 0.131.3
[main  ] INFO 2024/10/29 05:47:01 using config file: /config/evcc.yaml
[main  ] INFO 2024/10/29 05:47:01 listening at :7070
[ocpp  ] INFO 2024/10/29 05:47:10 charge point connected, registering: xxxxxxx `

matti1993, 2024-10-29:
bei mir ist es auch ab 0.131.2 schon, und hab eine ABB Tera AC als Wallbox. 
Evcc startet kann aber keine Verbindung Aufbauen.

grappa24, 2024-10-29:
Hier der Vollständigkeit halber noch meine evcc.yaml

`
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

sponsortoken: xxxx

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.3249 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh

  co2:
    type: grünstromindex
    zip: 60437 # For post codes with leading zeros, enclose in ""

# unique installation id
plant: xxxx

interval: 30s # control cycle interval

meters:
- type: template
  template: fronius-gen24 
  usage: grid  
  host: 192.168.178.129  
  port: 502  
  name: grid1
- type: template
  template: fronius-gen24 
  usage: pv  
  host: 192.168.178.129  
  port: 502  
  name: pv2
- type: template
  template: fronius-gen24 
  usage: battery  
  host: 192.168.178.129  
  port: 502  
  name: battery3

chargers:
- type: template
  template: ocpp-fronius-wattpilot 
  name: wallbox5

vehicles:
- type: template
  template: seat 
  title: Cupra Formentor  
  icon: car  
  user: yyyy
  password: xxxx
  vin: zzzz
  capacity: 10.4  
  phases: 1  
  cache: 15m  
  maxCurrent: 16  
  name: ev4

loadpoints:
- title: Wattpilot Garage
  charger: wallbox5
  vehicle: ev4
  mode: pv
  enable:
    threshold: -1400
    delay: 2m
  disable:
    threshold: 1400
    delay: 5m
  guardduration: 5m

site:
  title: pv-anlage 
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 100

`

musbach, 2024-10-29:
Bei mir funktioniert auch die 0.131.2 nicht:

<details><summary>Log</summary>
<p>

[site  ] DEBUG 2024/10/29 09:05:12 ----
[lp-1  ] DEBUG 2024/10/29 09:05:12 charge power: 0W
[lp-1  ] DEBUG 2024/10/29 09:05:12 charge currents: [0 0 0]A
[site  ] DEBUG 2024/10/29 09:05:12 grid power: 254W
[site  ] DEBUG 2024/10/29 09:05:12 pv 1 power: 183W
[site  ] DEBUG 2024/10/29 09:05:12 battery soc: 29%
[site  ] DEBUG 2024/10/29 09:05:12 battery power: -26W
[site  ] DEBUG 2024/10/29 09:05:12 pv 2 power: 66W
[site  ] DEBUG 2024/10/29 09:05:12 pv power: 249W
[site  ] DEBUG 2024/10/29 09:05:12 battery has priority at soc 29% (< 45%)
[site  ] DEBUG 2024/10/29 09:05:12 site power: 354W
[lp-1  ] DEBUG 2024/10/29 09:05:12 charge voltages: [236 237 237]V
[lp-1  ] DEBUG 2024/10/29 09:05:12 detected connected phases: 3p
[lp-1  ] DEBUG 2024/10/29 09:05:12 charge total import: 1147.521kWh
[lp-1  ] DEBUG 2024/10/29 09:05:12 charger status: B
Stopping evcc.service - evcc...
evcc.service: Deactivated successfully.
Stopped evcc.service - evcc.
evcc.service: Consumed 20min 15.596s CPU time.
Started evcc.service - evcc.
[main  ] INFO 2024/10/29 09:06:00 evcc 0.131.3
[main  ] INFO 2024/10/29 09:06:00 using config file: /etc/evcc.yaml
[main  ] INFO 2024/10/29 09:06:01 listening at :7070
[ocpp-1] DEBUG 2024/10/29 09:06:01 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/10/29 09:06:04 charge point connected, registering: EVB-P22205323
[main  ] FATAL 2024/10/29 09:11:01 cannot create charger 'wallbox7': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main  ] FATAL 2024/10/29 09:11:01 will attempt restart in: 15m0s

</p>
</details> 

Gehe wieder zurück auf die 0.131.1. Die läuft.

andig, 2024-10-29:
Nightly aktualisiert- Fehler behoben?

grappa24, 2024-10-29:
Obwohl der status wie folgt aussieht, können meine Browser keine Verbindung zum evcc Server aufbauen 

```
Oct 29 09:28:57 raspberrypi systemd[1]: Started evcc.
Oct 29 09:28:58 raspberrypi evcc[8588]: [main  ] INFO 2024/10/29 09:28:58 evcc 0.131.3
Oct 29 09:28:58 raspberrypi evcc[8588]: [main  ] INFO 2024/10/29 09:28:58 using config file: /etc/evcc.yaml
Oct 29 09:28:58 raspberrypi evcc[8588]: [db    ] INFO 2024/10/29 09:28:58 using sqlite database: /var/lib/evcc/evcc.db
Oct 29 09:28:58 raspberrypi evcc[8588]: [mqtt  ] INFO 2024/10/29 09:28:58 connecting evcc at tcp://192.168.178.44:1883
Oct 29 09:28:59 raspberrypi evcc[8588]: [mqtt  ] DEBUG 2024/10/29 09:28:59 tcp://192.168.178.44:1883 connected
Oct 29 09:28:59 raspberrypi evcc[8588]: [main  ] INFO 2024/10/29 09:28:59 listening at :7070
Oct 29 09:28:59 raspberrypi evcc[8588]: [ocpp-1] DEBUG 2024/10/29 09:28:59 waiting for chargepoint: 5m0s
Oct 29 09:29:10 raspberrypi evcc[8588]: [ocpp  ] INFO 2024/10/29 09:29:10 charge point connected, registering: 91027406

```

andig, 2024-10-29:
Was willst Du damit bzgl. https://github.com/evcc-io/evcc/issues/16941#issuecomment-2443545273 sagen?

musbach, 2024-10-29:
Super! Danke! Das hat geklappt!

<details><summary>Log</summary>
<p>

Stopping evcc.service - evcc...
evcc.service: Deactivated successfully.
Stopped evcc.service - evcc.
evcc.service: Consumed 46.435s CPU time.
Started evcc.service - evcc.
[main  ] INFO 2024/10/29 10:09:18 evcc 0.131.3 (7c8bd1a2)
[main  ] INFO 2024/10/29 10:09:18 using config file: /etc/evcc.yaml
[main  ] INFO 2024/10/29 10:09:18 listening at :7070
[ocpp-1] DEBUG 2024/10/29 10:09:18 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/10/29 10:09:21 charge point connected, registering: EVB-Pxxx
[ocpp-1] DEBUG 2024/10/29 10:09:51 failed configuring availability: timeout
[ocpp-1] WARN 2024/10/29 10:10:27 failed configuring MeterValuesSampledData: timeout
[lp-1  ] DEBUG 2024/10/29 10:10:32 charge total import: 1147.893kWh
[site  ] INFO 2024/10/29 10:10:32 site config:
[site  ] INFO 2024/10/29 10:10:32   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/10/29 10:10:32     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/10/29 10:10:32     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/10/29 10:10:32     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/10/29 10:10:32     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/10/29 10:10:32   vehicles:
[site  ] INFO 2024/10/29 10:10:32     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2024/10/29 10:10:32 loadpoint 1:
[lp-1  ] INFO 2024/10/29 10:10:32   mode:        pv
[lp-1  ] INFO 2024/10/29 10:10:32   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/10/29 10:10:32   meters:      charge ✓
[lp-1  ] INFO 2024/10/29 10:10:32     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2024/10/29 10:10:32 set buffer soc: 55
[site  ] DEBUG 2024/10/29 10:10:32 set buffer start soc: 0
[site  ] DEBUG 2024/10/29 10:10:32 set priority soc: 45
[lp-1  ] DEBUG 2024/10/29 10:10:32 phase timer inactive
[lp-1  ] DEBUG 2024/10/29 10:10:32 pv timer inactive
[lp-1  ] DEBUG 2024/10/29 10:10:32 max charge current: 6A
[site  ] WARN 2024/10/29 10:10:32 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/10/29 10:10:32 ----
[lp-1  ] DEBUG 2024/10/29 10:10:32 charge power: 1419W
[lp-1  ] DEBUG 2024/10/29 10:10:32 charge currents: [6.2 0 0]A
[site  ] DEBUG 2024/10/29 10:10:32 grid power: 1653W
[site  ] DEBUG 2024/10/29 10:10:32 battery soc: 31%
[site  ] DEBUG 2024/10/29 10:10:32 battery power: -157W
[site  ] DEBUG 2024/10/29 10:10:32 pv 1 power: 327W
[site  ] DEBUG 2024/10/29 10:10:33 pv 2 power: 111W
[site  ] DEBUG 2024/10/29 10:10:33 pv power: 438W
[site  ] DEBUG 2024/10/29 10:10:33 battery has priority at soc 31% (< 45%)
[site  ] DEBUG 2024/10/29 10:10:33 site power: 1753W
[lp-1  ] DEBUG 2024/10/29 10:10:33 charge voltages: [234 237 235]V
[lp-1  ] DEBUG 2024/10/29 10:10:33 detected connected phases: 3p
[lp-1  ] DEBUG 2024/10/29 10:10:33 charge total import: 1147.893kWh
[lp-1  ] DEBUG 2024/10/29 10:10:33 charger status: C
[lp-1  ] DEBUG 2024/10/29 10:10:33 charge start energy: 1147.893kWh
[lp-1  ] INFO 2024/10/29 10:10:33 car connected
[lp-1  ] DEBUG 2024/10/29 10:10:33 vehicle api refresh
[lp-1  ] DEBUG 2024/10/29 10:10:33 pv timer elapse
[lp-1  ] DEBUG 2024/10/29 10:10:33 pv timer inactive
[lp-1  ] DEBUG 2024/10/29 10:10:33 charge total import: 1147.893kWh
[lp-1  ] DEBUG 2024/10/29 10:10:33 charge start energy: 1147.893kWh
[lp-1  ] INFO 2024/10/29 10:10:33 start charging ->
[lp-1  ] DEBUG 2024/10/29 10:10:33 wake-up timer: stop
[main  ] DEBUG 2024/10/29 10:10:33 vehicle status: C (Tarraco)
[lp-1  ] INFO 2024/10/29 10:10:33 vehicle updated: unknown -> Tarraco
[lp-1  ] DEBUG 2024/10/29 10:10:33 vehicle status: C
[lp-1  ] DEBUG 2024/10/29 10:10:33 vehicle soc: 20%
[lp-1  ] DEBUG 2024/10/29 10:10:33 vehicle range: 7km
[lp-1  ] DEBUG 2024/10/29 10:10:33 pv charge current: min 6A > 0A (1753W @ 1p, battery: false)
[site  ] DEBUG 2024/10/29 10:10:37 ----
[lp-1  ] DEBUG 2024/10/29 10:10:37 charge power: 1419W
[lp-1  ] DEBUG 2024/10/29 10:10:37 charge currents: [6.2 0 0]A
[site  ] DEBUG 2024/10/29 10:10:37 grid power: 1647W
[site  ] DEBUG 2024/10/29 10:10:37 pv 1 power: 326W
[site  ] DEBUG 2024/10/29 10:10:37 battery soc: 31%
[site  ] DEBUG 2024/10/29 10:10:37 battery power: -157W
[site  ] DEBUG 2024/10/29 10:10:37 pv 2 power: 111W

</p>
</details>

matti1993, 2024-10-29:
Was hast du jetzt genau gemacht?

closed 2024-10-29 by andig

TomF79, 2024-10-29:
@andig 

Bitte auch um Aktualisierung des aktuellen Release für uns HA-User, vielen lieben Dank.

TomF79, 2024-10-29:
@grappa24 

Funktioniert es bei dir auch wieder mit der geänderten Nightly?

grappa24, 2024-10-29:
> Was willst Du damit bzgl. [#16941 (comment)](https://github.com/evcc-io/evcc/issues/16941#issuecomment-2443545273) sagen?

dass evcc bei mir mit 0.131.3 nicht erreichbar ist, wohl aber mit der 0.131.0

grappa24, 2024-10-29:
> @grappa24
> 
> Funktioniert es bei dir auch wieder mit der geänderten Nightly?

nein, ist nicht erreichbar

andig, 2024-10-29:
Da fällt mir leider wenig ein. Hast Du ein trace log?

grappa24, 2024-10-29:
> Da fällt mir leider wenig ein. Hast Du ein trace log?

wenn du mir kurz skizzierst, wie man das erstellt ...

grappa24, 2024-10-29:
Scheint wohl an meiner Wallbox-Definition zu liegen:

`Oct 29 11:26:58 raspberrypi evcc[10516]: [main  ] FATAL 2024/10/29 11:26:58 cannot create charger 'wallbox5': cannot create charger type 'template': cannot create charger type 'ocpp': timeout`

und hier meine Wallbox Defintion (ist ein Fronius Wattpilot):
```
chargers:
    type: template
    template: ocpp-fronius-wattpilot
    name: wallbox5

```

TomF79, 2024-10-29:
> Da fällt mir leider wenig ein. Hast Du ein trace log?

Hab diesselbe config mit Gen24 und Wattpilot kann aber nicht testen da keine Nightly möglich bei HA, bräuchte dazu ein neues Release.

TomF79, 2024-10-29:
> Da fällt mir leider wenig ein. Hast Du ein trace log?

@andig 

Hab diesselbe config mit Gen24 und Wattpilot kann aber nicht testen da keine Nightly möglich bei HA, bräuchte dazu ein neues Release.

Bis Version 0.131.1 funktionierte jedenfalls alles.

TomF79, 2024-10-29:
@andig 

Konnte die nightly mit @dm82m HA Addon testen und  hiermit bestätigen das es auch mit Gen24 und Wattpilot läuft:

```go
[main ] INFO 2024/10/29 13:00:34 evcc 0.131.3 (b0060ad4)
[main ] INFO 2024/10/29 13:00:34 using config file: /config/evcc.yaml
[db ] INFO 2024/10/29 13:00:34 using sqlite database: /data/evcc.db
[main ] INFO 2024/10/29 13:00:34 listening at :7070
[ocpp-1] DEBUG 2024/10/29 13:00:34 waiting for chargepoint: 5m0s
[ocpp ] INFO 2024/10/29 13:00:39 charge point connected, registering: xxxxxxxx
[lp-1 ] DEBUG 2024/10/29 13:00:58 charge total import: 6250.294kWh
[site ] INFO 2024/10/29 13:00:58 site config:
[site ] INFO 2024/10/29 13:00:58 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/10/29 13:00:58 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2024/10/29 13:00:58 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2024/10/29 13:00:58 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site ] INFO 2024/10/29 13:00:58 vehicles:
[site ] INFO 2024/10/29 13:00:58 vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✓
[site ] INFO 2024/10/29 13:00:58 vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1 ] INFO 2024/10/29 13:00:58 loadpoint 1:
[lp-1 ] INFO 2024/10/29 13:00:58 mode: off
[lp-1 ] INFO 2024/10/29 13:00:58 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2024/10/29 13:00:58 meters: charge ✓
[lp-1 ] INFO 2024/10/29 13:00:58 charge: power ✓ energy ✓ currents ✓
[lp-1 ] DEBUG 2024/10/29 13:00:58 phase timer inactive
[lp-1 ] DEBUG 2024/10/29 13:00:58 pv timer inactive
[site ] WARN 2024/10/29 13:00:58 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site ] DEBUG 2024/10/29 13:00:58 ----
[lp-1 ] DEBUG 2024/10/29 13:00:58 charge power: 0W
[lp-1 ] DEBUG 2024/10/29 13:00:58 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/29 13:00:58 grid power: -16W
[site ] DEBUG 2024/10/29 13:00:58 pv power: 1649W
[site ] DEBUG 2024/10/29 13:00:58 grid powers: [-296 -343 623]W
[site ] DEBUG 2024/10/29 13:00:58 battery soc: 25%
[site ] DEBUG 2024/10/29 13:00:58 battery power: -400W
[site ] DEBUG 2024/10/29 13:00:58 grid currents: [-1.37 -1.68 2.77]A
[site ] DEBUG 2024/10/29 13:00:58 site power: -406W
[lp-1 ] DEBUG 2024/10/29 13:00:58 charge total import: 6250.294kWh
[lp-1 ] DEBUG 2024/10/29 13:00:58 charger status: A
[lp-1 ] INFO 2024/10/29 13:00:58 car disconnected
[lp-1 ] DEBUG 2024/10/29 13:00:58 set battery boost: false
[lp-1 ] DEBUG 2024/10/29 13:00:58 set session soc limit: 0
[lp-1 ] DEBUG 2024/10/29 13:00:58 set session energy limit: 0
[site ] DEBUG 2024/10/29 13:01:18 ----
``` 

Bitte um Neues offizielles Release, danke.

etecprojekt, 2024-10-29:
@TomF79: Mehrfach getestet, auch mit angestecktem Auto ect?
Wir haben Probleme, dass es mal mit Glück klappt und mal nicht.

grappa24, 2024-10-30:
> @andig
>
> Konnte die nightly mit @dm82m HA Addon testen und hiermit bestätigen das es auch mit Gen24 und Wattpilot läuft:
>
> ```go
> [main ] INFO 2024/10/29 13:00:34 evcc 0.131.3 (b0060ad4)
> [main ] INFO [...]

@TomF79 läuft denn evcc jetzt bei dir, ich bekomme weiterhin bei der 0.131.3 keine Verbindung. Kenne mich mir der Entwicklung und den Update Cyclen auf git nicht so aus, ist "unser" Anliegen denn noch in Arbeit oder ist das geschlossen?

TomF79, 2024-10-30:
> @TomF79 läuft denn evcc jetzt bei dir, ich bekomme weiterhin bei der 0.131.3 keine Verbindung. Kenne mich mir der Entwicklung und den Update Cyclen auf git nicht so aus, ist "unser" Anliegen denn noch in Arbeit oder ist das geschlossen?

Nein, du musst leider warten bis @andig gewillt ist eine neue Version zu releasen, ab 0.131.4 wird es dann wieder funktionieren.

grappa24, 2024-10-30:
> Nein, du musst leider warten bis @andig gewillt ist eine neue Version zu releasen, ab 0.131.4 wird es dann wieder funktionieren.

danke, dann ist ja alles gut; mir war nur nicht klar, ob das überhaupt noch im Fokus ist.

TomF79, 2024-10-30:
> danke, dann ist ja alles gut; mir war nur nicht klar, ob das überhaupt noch im Fokus ist.

Du kannst einstweilen ein Downgrade auf 0.131.1 machen, da läuft noch alles.