evcc-io/evcc #16952: EVCC not starting

timpetersson81, 2024-10-29:
### Describe the bug

after update to 0.131.3 I have the same issues as with 0.131.2

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  port: 7070

log: debug

interval: 15s

meters:
  - name: my_grid
    type: template
    template: sungrow-hybrid
    usage: grid
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.1.63 # Hostname
    port: 502 # Port
    timeout: 10s # optional

  - name: my_pv
    type: template
    template: sungrow-hybrid
    usage: pv

    modbus: tcpip
    id: 1
    host: 192.168.1.63 # Hostname
    port: 502 # Port
    timeout: 10s # optional

  - name: my_battery
    type: template
    template: sungrow-hybrid
    usage: battery

    modbus: tcpip
    id: 1
    host: 192.168.1.63 # Hostname
    port: 502 # Port
    timeout: 10s # optional

chargers:
  - name: my_charger
    type: template
    template: pulsarplus-fw5

vehicles:
  - name: Xpeng
    type: template
    template: offline
    title: # Will be displayed in the user interface (optional)
    capacity: 100

site:
  title: Zuhause
  meters:
    grid: my_grid
    pv: my_pv
    battery: my_battery

loadpoints:
  - title: Carport
    charger: my_charger
    mode: pv
    meter: my_grid

tariffs:
  currency: DKK
  grid:
    type: template
    template: energinet
    region: dk2 # optional
#   zones:
#      - days: Mo-So
#        hours: 00-06
#        charges: 0.1431
#      - days: Mo-So
#        hours: 06-17
#        charges: 0.4293 # DKK/kWh
#      - days: Mo-So
#        hours: 17-21
#        charges: 1.28 # DKK/kWh
#      - days: Mo-So
#        hours: 21-00
#        charges: 0.4293 # DKK/kWh

sponsortoken:  eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidGltcGV0ZXJzc29uODEiLCJleHAiOjE4MjQzNjQ4MDAsImlhdCI6MTcyOTc1NjgwMCwic3JjIjoiZ2gifQ.K0GIZT6c3jCGxdJjTs-FZ96VYmuPKQ5OwrsoPHsgI9Y
```

### Log details

```text
Using config file: /config/evcc1.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc1.yaml'
[main  ] INFO 2024/10/29 13:39:17 evcc 0.131.3
[main  ] INFO 2024/10/29 13:39:17 using config file: /config/evcc1.yaml
[db    ] INFO 2024/10/29 13:39:17 using sqlite database: /data/evcc.db
[main  ] INFO 2024/10/29 13:39:17 listening at :7070
[ocpp-1] DEBUG 2024/10/29 13:39:17 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/10/29 13:40:10 charge point connected, registering: wallbox
Using config file: /config/evcc1.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc1.yaml'
[main  ] INFO 2024/10/29 13:42:22 evcc 0.131.3
[main  ] INFO 2024/10/29 13:42:22 using config file: /config/evcc1.yaml
[db    ] INFO 2024/10/29 13:42:22 using sqlite database: /data/evcc.db
[main  ] INFO 2024/10/29 13:42:22 listening at :7070
[ocpp-1] DEBUG 2024/10/29 13:42:22 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/10/29 13:43:16 charge point connected, registering: wallbox
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.3

timpetersson81, 2024-10-29:
i don't know what nightly build is, but I could not submit the ticket without checking the box

TomF79, 2024-10-29:
Sollte hiermit gelöst sein sobald es ein neues Release ( @andig ) gibt: https://github.com/evcc-io/evcc/issues/16941

closed 2024-10-29 by andig

timpetersson81, 2024-10-29:
why did you close it @andig ?

andig, 2024-10-29:
https://github.com/evcc-io/evcc/issues/16952#issuecomment-2444127032