evcc-io/evcc #16966: OCPP Wallbox add suffix for phase measurement of current power and current import

pvdoda, 2024-10-29:
### Describe the bug

Delta AC Max Wallbox connected via OCPP template.  The phase names of the meter values for  "Current.Import" and "Power.Active.Import" have a suffix "-N". The values with prefix should be prioritized. 

[    res, found, err := conn.phaseMeasurements(types.MeasurandPowerActiveImport, "")](https://github.com/evcc-io/evcc/blob/master/charger/ocpp/connector.go#L269)

[    res, found, err := conn.phaseMeasurements(types.MeasurandCurrentImport, "")](https://github.com/evcc-io/evcc/blob/master/charger/ocpp/connector.go#L350)

This feature is already implemented for "Voltage" .

https://github.com/evcc-io/evcc/blob/master/charger/ocpp/connector.go#L371

```
[
    2,
    "173023500351",
    "MeterValues",
    {
        "connectorId": 1,
        "transactionId": 1730229549,
        "meterValue": [
            {
                "timestamp": "2024-10-29T20:50:03Z",
                "sampledValue": [
                    {
                        "value": "222.30",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Voltage",
                        "phase": "L1-N",
                        "location": "Body",
                        "unit": "V"
                    },
                    {
                        "value": "15.52",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Current.Import",
                        "phase": "L1-N",
                        "location": "Body",
                        "unit": "A"
                    },
                    {
                        "value": "3.43",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Power.Active.Import",
                        "phase": "L1-N",
                        "location": "Body",
                        "unit": "kW"
                    },
                    {
                        "value": "2331816",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Energy.Active.Import.Register",
                        "location": "Body",
                        "unit": "Wh"
                    },
                    {
                        "value": "16.00",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Current.Offered",
                        "location": "Body",
                        "unit": "A"
                    },
                    {
                        "value": "231.60",
                        "context": "Sample.Periodic",
[... 199 lines cut ...]
[ocpp ] TRACE 2024/10/29 21:50:46 handling incoming CALL [173023504658, MeterValues] from JVH223002421WE
[ocpp ] TRACE 2024/10/29 21:50:46 sent CALL RESULT [173023504658] for JVH223002421WE
[ocpp ] TRACE 2024/10/29 21:50:46 sent JSON message to JVH223002421WE: [3,"173023504658",{}]
[ocpp ] TRACE 2024/10/29 21:50:47 received JSON message from JVH223002421WE: [2,"173023504759","Heartbeat",{}]
[ocpp ] TRACE 2024/10/29 21:50:47 handling incoming CALL [173023504759, Heartbeat] from JVH223002421WE
[ocpp ] TRACE 2024/10/29 21:50:47 sent CALL RESULT [173023504759] for JVH223002421WE
[ocpp ] TRACE 2024/10/29 21:50:47 sent JSON message to JVH223002421WE: [3,"173023504759",{"currentTime":"2024-10-29T20:50:47Z"}]
[ocpp ] TRACE 2024/10/29 21:50:53 received JSON message from JVH223002421WE: [2,"173023505360","MeterValues",{"connectorId":1,"transactionId":1730229549,"meterValue":[{"timestamp":"2024-10-29T20:50:53Z","sampledValue":[{"value":"222.60","context":"Sample.Periodic","format":"Raw","measurand":"Voltage","phase":"L1-N","location":"Body","unit":"V"},{"value":"15.61","context":"Sample.Periodic","format":"Raw","measurand":"Current.Import","phase":"L1-N","location":"Body","unit":"A"},{"value":"3.46","context":"Sample.Periodic","format":"Raw","measurand":"Power.Active.Import","phase":"L1-N","location":"Body","unit":"kW"},{"value":"2331864","context":"Sample.Periodic","format":"Raw","measurand":"Energy.Active.Import.Register","location":"Body","unit":"Wh"},{"value":"16.00","context":"Sample.Periodic","format":"Raw","measurand":"Current.Offered","location":"Body","unit":"A"},{"value":"231.50","context":"Sample.Periodic","format":"Raw","measurand":"Voltage","phase":"L2-N","location":"Body","unit":"V"},{"value":"230.50","context":"Sample.Periodic","format":"Raw","measurand":"Voltage","phase":"L3-N","location":"Body","unit":"V"},{"value":"0.00","context":"Sample.Periodic","format":"Raw","measurand":"Current.Import","phase":"L2-N","location":"Body","unit":"A"},{"value":"0.00","context":"Sample.Periodic","format":"Raw","measurand":"Current.Import","phase":"L3-N","location":"Body","unit":"A"},{"value":"0","context":"Sample.Periodic","format":"Raw","measurand":"Power.Offered","location":"Body","unit":"W"},{"value":"0.00","context":"Sample.Periodic","format":"Raw","measurand":"Power.Active.Import","phase":"L2-N","location":"Body","unit":"kW"},{"value":"0.00","context":"Sample.Periodic","format":"Raw","measurand":"Power.Active.Import","phase":"L3-N","location":"Body","unit":"kW"}]}]}]
[ocpp ] TRACE 2024/10/29 21:50:53 handling incoming CALL [173023505360, MeterValues] from JVH223002421WE
[ocpp ] TRACE 2024/10/29 21:50:53 sent CALL RESULT [173023505360] for JVH223002421WE
[ocpp ] TRACE 2024/10/29 21:50:53 sent JSON message to JVH223002421WE: [3,"173023505360",{}]
[lp-1 ] ERROR 2024/10/29 21:50:54 charge power: not available
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.3+1730189808

premultiply, 2024-10-29:
PR welcome

pvdoda, 2024-10-29:
wie PR ohne branch? Sorry, bin neu in git.

andig, 2024-10-30:
> The values with prefix should be prioritized.

Why? They shouldn't. Current flows through a conductor, not relative to some neutral phase?!