evcc-io/evcc #16972: Netzladen mit sma-hybrid funktioniert nicht

IkarusPV, 2024-10-30:
### Describe the bug

Ich verwende evcc mit einem SMA-Hybridwechselrichter STP 8.0 SE, einer BYD-Batterie HVS 10.2 und dem SMA Home Manager 2.0 mit der aktuellen Firmware 2.15.6.R.

Die Batterie wird über die Funktion Hausbatterie -> Netzladen nicht aus dem Stromnetz geladen.
UI bestätigt günstiges Netzladen bei Erfüllung der eingestellten Preisgrenzenbedingung.
Einzig erkennbarer Effekt ist das Sperren der Batterie für Laden und Entladen.

### Steps to reproduce

1. Preis so eingestellt, dass die Batterie aus dem Netz geladen werden soll. 
2. UI bestätigt günstiges Netzladen.
3. Batterie lädt nicht aus dem Netz.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  SMA-SHM20: trace
  SMA-STP80: trace
  BYD-Batterybox: trace
  
database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db

# unique installation id
plant: ***

interval: 30s # control cycle interval

sponsortoken: ***

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- name: SMA-SHM20
  type: template
  template: sma-home-manager 
  host: 192.168.188.8 
  usage: grid  
 
- name: SMA-STP80
  type: template
  template: sma-hybrid 
  id: 3  
  host: 192.168.188.6  
  port: 502  
  usage: pv  
  modbus: tcpip
  maxacpower: 8000
  watchdog: 30s  
 
- name: BYD-Batterybox
  type: template
  template: sma-hybrid 
  id: 3  
  host: 192.168.188.6  
  port: 502  
  usage: battery  
  modbus: tcpip  
[... 564 lines cut ...]
[sma   ] TRACE 2024/10/30 08:44:20 recv 192.168.188.8: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/30 08:44:20 recv 192.168.188.8: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/30 08:44:20 recv 192.168.188.8: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/30 08:44:20 recv 192.168.188.8: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/30 08:44:20 recv 192.168.188.8: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/30 08:44:21 recv 192.168.188.8: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/30 08:44:21 recv 192.168.188.8: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/30 08:44:21 recv 192.168.188.8: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/30 08:44:21 recv 192.168.188.8: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/30 08:44:21 recv 192.168.188.8: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/30 08:44:22 recv 192.168.188.8: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/30 08:44:22 recv 192.168.188.8: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.3 (4b51a301)

andig, 2024-10-31:
Magst Du das Log mal um BYD und den SHM bereinigen so dass man den wesentlichen Teil sieht?

IkarusPV, 2024-10-31:
@andig

Log um alles bereinigt, was nicht modbus-Kommunikation zwischen Wechselrichter und Batterie ist.
Passt das so? Soll ich Logeinträge für einen längeren Zeitraum bereitstellen?

[SMA-STP80] TRACE 2024/10/30 08:42:59 modbus: send 00 28 00 00 00 06 03 03 78 35 00 02
[SMA-STP80] TRACE 2024/10/30 08:42:59 modbus: recv 00 28 00 00 00 07 03 03 04 00 00 00 00
[SMA-STP80] TRACE 2024/10/30 08:42:59 modbus: send 00 29 00 00 00 06 03 03 78 f1 00 02
[SMA-STP80] TRACE 2024/10/30 08:42:59 modbus: recv 00 29 00 00 00 07 03 03 04 00 00 00 37
[BYD-Batterybox] TRACE 2024/10/30 08:42:59 modbus: send 00 2a 00 00 00 06 03 04 7a a3 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:42:59 modbus: recv 00 2a 00 00 00 07 03 04 04 00 00 00 00
[BYD-Batterybox] TRACE 2024/10/30 08:42:59 modbus: send 00 2b 00 00 00 06 03 04 7a a1 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:42:59 modbus: recv 00 2b 00 00 00 07 03 04 04 00 00 00 00
[SMA-STP80] TRACE 2024/10/30 08:42:59 modbus: send 00 2c 00 00 00 06 03 03 77 31 00 04
[SMA-STP80] TRACE 2024/10/30 08:42:59 modbus: recv 00 2c 00 00 00 0b 03 03 08 00 00 00 00 00 be f9 f9
[BYD-Batterybox] TRACE 2024/10/30 08:42:59 modbus: send 00 2d 00 00 00 06 03 03 7a a9 00 04
[BYD-Batterybox] TRACE 2024/10/30 08:43:00 modbus: recv 00 2d 00 00 00 0b 03 03 08 00 00 00 00 00 28 1b 2d
[BYD-Batterybox] TRACE 2024/10/30 08:43:00 modbus: send 00 2e 00 00 00 06 03 03 78 7d 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:43:00 modbus: recv 00 2e 00 00 00 07 03 03 04 00 00 00 00
[BYD-Batterybox] TRACE 2024/10/30 08:43:00 modbus: send 00 2f 00 00 00 0b 03 10 9c d5 00 02 04 00 00 00 00
[BYD-Batterybox] TRACE 2024/10/30 08:43:00 modbus: recv 00 2f 00 00 00 06 03 10 9c d5 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:43:00 modbus: send 00 30 00 00 00 0b 03 10 9c d7 00 02 04 00 00 03 22
[BYD-Batterybox] TRACE 2024/10/30 08:43:00 modbus: recv 00 30 00 00 00 06 03 10 9c d7 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:43:15 modbus: send 00 31 00 00 00 0b 03 10 9c d5 00 02 04 00 00 00 00
[BYD-Batterybox] TRACE 2024/10/30 08:43:15 modbus: recv 00 31 00 00 00 06 03 10 9c d5 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:43:15 modbus: send 00 32 00 00 00 0b 03 10 9c d7 00 02 04 00 00 03 22
[BYD-Batterybox] TRACE 2024/10/30 08:43:15 modbus: recv 00 32 00 00 00 06 03 10 9c d7 00 02
[SMA-STP80] TRACE 2024/10/30 08:43:29 modbus: send 00 33 00 00 00 06 03 03 78 35 00 02
[SMA-STP80] TRACE 2024/10/30 08:43:29 modbus: recv 00 33 00 00 00 07 03 03 04 00 00 00 00
[SMA-STP80] TRACE 2024/10/30 08:43:29 modbus: send 00 34 00 00 00 06 03 03 78 f1 00 02
[SMA-STP80] TRACE 2024/10/30 08:43:29 modbus: recv 00 34 00 00 00 07 03 03 04 00 00 00 3c
[BYD-Batterybox] TRACE 2024/10/30 08:43:29 modbus: send 00 35 00 00 00 06 03 04 7a a3 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:43:30 modbus: recv 00 35 00 00 00 07 03 04 04 00 00 00 00
[BYD-Batterybox] TRACE 2024/10/30 08:43:30 modbus: send 00 36 00 00 00 06 03 04 7a a1 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:43:30 modbus: recv 00 36 00 00 00 07 03 04 04 00 00 00 00
[SMA-STP80] TRACE 2024/10/30 08:43:30 modbus: send 00 37 00 00 00 06 03 03 77 31 00 04
[SMA-STP80] TRACE 2024/10/30 08:43:30 modbus: recv 00 37 00 00 00 0b 03 03 08 00 00 00 00 00 be f9 f9
[BYD-Batterybox] TRACE 2024/10/30 08:43:30 modbus: send 00 38 00 00 00 06 03 03 7a a9 00 04
[BYD-Batterybox] TRACE 2024/10/30 08:43:30 modbus: recv 00 38 00 00 00 0b 03 03 08 00 00 00 00 00 28 1b 2d
[BYD-Batterybox] TRACE 2024/10/30 08:43:30 modbus: send 00 39 00 00 00 06 03 03 78 7d 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:43:30 modbus: recv 00 39 00 00 00 07 03 03 04 00 00 00 00
[BYD-Batterybox] TRACE 2024/10/30 08:43:30 modbus: send 00 3a 00 00 00 0b 03 10 9c d5 00 02 04 00 00 00 00
[BYD-Batterybox] TRACE 2024/10/30 08:43:30 modbus: recv 00 3a 00 00 00 06 03 10 9c d5 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:43:30 modbus: send 00 3b 00 00 00 0b 03 10 9c d7 00 02 04 00 00 03 22
[BYD-Batterybox] TRACE 2024/10/30 08:43:30 modbus: recv 00 3b 00 00 00 06 03 10 9c d7 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:43:45 modbus: send 00 3c 00 00 00 0b 03 10 9c d5 00 02 04 00 00 00 00
[BYD-Batterybox] TRACE 2024/10/30 08:43:45 modbus: recv 00 3c 00 00 00 06 03 10 9c d5 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:43:45 modbus: send 00 3d 00 00 00 0b 03 10 9c d7 00 02 04 00 00 03 22
[BYD-Batterybox] TRACE 2024/10/30 08:43:45 modbus: recv 00 3d 00 00 00 06 03 10 9c d7 00 02
[SMA-STP80] TRACE 2024/10/30 08:43:59 modbus: send 00 3e 00 00 00 06 03 03 78 35 00 02
[SMA-STP80] TRACE 2024/10/30 08:43:59 modbus: recv 00 3e 00 00 00 07 03 03 04 00 00 00 00
[SMA-STP80] TRACE 2024/10/30 08:43:59 modbus: send 00 3f 00 00 00 06 03 03 78 f1 00 02
[SMA-STP80] TRACE 2024/10/30 08:43:59 modbus: recv 00 3f 00 00 00 07 03 03 04 00 00 00 47
[BYD-Batterybox] TRACE 2024/10/30 08:43:59 modbus: send 00 40 00 00 00 06 03 04 7a a3 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:43:59 modbus: recv 00 40 00 00 00 07 03 04 04 00 00 00 00
[BYD-Batterybox] TRACE 2024/10/30 08:43:59 modbus: send 00 41 00 00 00 06 03 04 7a a1 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:43:59 modbus: recv 00 41 00 00 00 07 03 04 04 00 00 00 00
[SMA-STP80] TRACE 2024/10/30 08:43:59 modbus: send 00 42 00 00 00 06 03 03 77 31 00 04
[SMA-STP80] TRACE 2024/10/30 08:43:59 modbus: recv 00 42 00 00 00 0b 03 03 08 00 00 00 00 00 be f9 f9
[BYD-Batterybox] TRACE 2024/10/30 08:43:59 modbus: send 00 43 00 00 00 06 03 03 7a a9 00 04
[BYD-Batterybox] TRACE 2024/10/30 08:43:59 modbus: recv 00 43 00 00 00 0b 03 03 08 00 00 00 00 00 28 1b 2d
[... cut ...]
[BYD-Batterybox] TRACE 2024/10/30 08:44:00 modbus: send 00 46 00 00 00 0b 03 10 9c d7 00 02 04 00 00 03 22
[BYD-Batterybox] TRACE 2024/10/30 08:44:00 modbus: recv 00 46 00 00 00 06 03 10 9c d7 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:44:15 modbus: send 00 47 00 00 00 0b 03 10 9c d5 00 02 04 00 00 00 00
[BYD-Batterybox] TRACE 2024/10/30 08:44:15 modbus: recv 00 47 00 00 00 06 03 10 9c d5 00 02
[BYD-Batterybox] TRACE 2024/10/30 08:44:15 modbus: send 00 48 00 00 00 0b 03 10 9c d7 00 02 04 00 00 03 22
[BYD-Batterybox] TRACE 2024/10/30 08:44:15 modbus: recv 00 48 00 00 00 06 03 10 9c d7 00 02

RichardIstSauer, 2024-11-01:
Beim SBS 2.5 funktioniert das Netzladen auch nicht mehr..

premultiply, 2024-11-01:
Offensichtlich hilft es angeblich laut PV-Forum den WR und Batterie komplett stromlos zu machen und dann nach ein paar Minuten neu hochzufahren.

Bitte mal ausprobieren.

michelix11, 2024-11-01:
Auch bei mir keine Besserung. Schade.

Deepy1973, 2024-11-01:
Zwar sbs 5.0 statt hybrid aber selbes Problem und ebenfalls keine Besserung

andig, 2024-11-01:
Nachdem wir das jetzt wiederholen: gibts ein Ticket bei SMA?

philipp750, 2024-11-01:
Gerade Update von 0.131.1 auf 0.131.4 durchgeführt und Netzladen funktioniert, sogar besser als vorher. Vorher wurde im 20s Takt mit maximaler Ladeleistung von 10kw und dann 1kw geladen. Das immer hin und her. Jetzt hält er die Leistung dauerhaft. 
STP10, SHM2.0 und BYD HVS12,8

andig, 2024-11-01:
Was auch immer Du da beobachtest- an evcc hat sich nichts geändert. Vielmehr geht es heir um ein Fehlverhalten des WR...

michelix11, 2024-11-01:
Aber irgendwie schon seltsam das es bei dem einen geht und bei dem anderen nicht.
Ich habe genau das gleiche  Setup wie phillip750 und bei mir funktioniert es leider nicht. Kann das auch irgendwas mit den Einstellungen im WR oder Sunnyportal zu tun haben?

andig, 2024-11-01:
Frag bitte SMA. An evcc hat sich nix geändert. Die Ursache muss also anderswo liegen...

premultiply, 2024-11-01:
> Kann das auch irgendwas mit den Einstellungen im WR oder Sunnyportal zu tun haben?

Ich würde die Wahrscheinlichkeit als äusserst hoch ansehen 😉

Deepy1973, 2024-11-02:
> > Kann das auch irgendwas mit den Einstellungen im WR oder Sunnyportal zu tun haben?
> 
> Ich würde die Wahrscheinlichkeit als äusserst hoch ansehen 😉

Dann teilt doch bitte eure Einstellungen mit uns. Denn Betriebsart Wirkleistungsvorgabe auf "Externe Wirkleistungsvorgabe" und
"Externe Wirkleistungsvorgabe 2, Niedrige Priorität" auf "Aus" allein löst den Zauber nicht aus 😉

wittmannchris, 2024-11-02:
Habt ihr schon alle die 2.15.5/6 FW auf dem SHM drauf ?  Mit der vorherigen geht es nicht. 

Der SBS 2.5 funktioniert mit Netzladen einwandfrei.

IkarusPV, 2024-11-02:
> Offensichtlich hilft es angeblich laut PV-Forum den WR und Batterie komplett stromlos zu machen und dann nach ein paar Minuten neu hochzufahren.
> 
> Bitte mal ausprobieren.

Auch hier leider keine Besserung.

michelix11, 2024-11-02:
Mein Setup ist quasi wie bei phillip750
STP10, SHM2.0 und BYD HVS12,8
Neustart von Batterie Wechselrichter und SHM2 auch schon alles durchgeführt.
Alles soweit mit der neusten Firmware.
Vielleicht wäre es möglich das er hier mal seine Parameter postet, wenn er denn will.
Da es ja bei einigen anscheinend funktioniert und bei anderen nicht, kann es ja fast nur irgendwo an den Einstellungen im Sunnyportal oder an WR liegen.....

Deepy1973, 2024-11-02:
> Gerade Update von 0.131.1 auf 0.131.4 durchgeführt und Netzladen funktioniert, sogar besser als vorher. Vorher wurde im 20s Takt mit maximaler Ladeleistung von 10kw und dann 1kw geladen. Das immer hin und her. Jetzt hält er die Leistung dauerhaft. STP10, SHM2.0 und BYD HVS12,8

Bitte mal die Einstellungen in Sunnyportal und WR posten. Hier muss ja der "Hase im Pfeffer" liegen

philipp750, 2024-11-02:
Parameter SHM:
[image]

Parameter Wechselrichter Batterie:
[image]

Falls weitere Parameter gewünscht sind, gerne schreiben. Ich bin mir nicht sicher, welche ihr genau braucht...

Deepy1973, 2024-11-02:
@philipp750 denke von nutzen wären hier mehr die Einstellungen im Bereich Wechselrichter.

michelix11, 2024-11-02:
Super und Danke schonmal. Ist bei mir aber alles auch genauso so eingestellt.
Im PV Forum habe ich gelesen das es wohl auch eine neue Firmware für den Wechselrichter gibt welche sich im Rollout befindet. Bei mir ist die 3.04.16.R installiert.
Welche Firmware hast du auf dem WR?

[https://www.photovoltaikforum.com/thread/234375-sunny-tripower-5-10-smart-energy-neue-firmware-3-4-21-r/](url)

IkarusPV, 2024-11-02:
Ich habe auch noch die Firmware 3.04.16.R auf dem WR.

philipp750, 2024-11-02:
Dieselbe Firmware habe ich auch

IkarusPV, 2024-11-02:
Grundsätzlich ist es bedauerlich, dass SMA 3rd-Party Entwicklung nicht wirklich zu unterstützen scheint. Sieht man ja auch am EV-Charger, aber das nur nebenbei.

Das Netzladen funktioniert ja - umständlich - mit SMA-Bordmitteln über das Portal mit der unflexiblen und "dummen" Zeitfenstersteuerung.

michelix11, 2024-11-02:
Aber schon irgendwie seltsam das es bei manchen funktioniert und bei anderen nicht. Obwohl augenscheinlich Einstellungen alle soweit gleich....
Was die Update Politik von SMA angeht sowie Unterstützung 3rd Party ist natürlich auch so eine Sache.
Da muss man nach einem Update immer in Angst sein das mehr verschlimmbessert wie verbessert wird.
Ich werde auf alles Fälle zu dem Thema mal ein Ticket bei SMA eröffnen

philipp750, 2024-11-02:
> @philipp750 denke von nutzen wären hier mehr die Einstellungen im Bereich Wechselrichter.

Meinst du die Parameter unter Anlagen- und Geräteteuerung? Welche genau?
[image]

deepy007, 2024-11-02:
Ja zum Beispiel. Bei meinem SBS 5.0 sieht es so aus und läuft **nicht**
[WR Einstellungen.pdf](https://github.com/user-attachments/files/17606639/WR.Einstellungen.pdf)

Wahrscheinlich nicht 1:1 identisch zu euren aber vielleicht hilft es trotzdem

wittmannchris, 2024-11-02:
Ich musste den BYD samt SBS2.5 auch tatsächlich mal für 5min komplett aus machen. 

Das Update auf die 2.15.6 (hatte die Feldtest 2.15.5 laufen) packte er wohl nicht....

RichardIstSauer, 2024-11-02:
> Ich musste den BYD samt SBS2.5 auch tatsächlich mal für 5min komplett aus machen.
> 
> Das Update auf die 2.15.6 (hatte die Feldtest 2.15.5 laufen) packte er wohl nicht....

Und danach läuft bei dir alles wieder?

wittmannchris, 2024-11-03:
> > Ich musste den BYD samt SBS2.5 auch tatsächlich mal für 5min komplett aus machen.
> > Das Update auf die 2.15.6 (hatte die Feldtest 2.15.5 laufen) packte er wohl nicht....
> 
> Und danach läuft bei dir alles wieder?

Ein aktueller Test zeigt : Ist der Speicher auf 0% und der BWR in Stand By , lässt er kein steuern zu.

Sobald 1% drinnen sind geht alles wie gewohnt....

philipp750, 2024-11-03:
@deepy007 
[SMA Gerät - Geräteparameter 4.pdf](https://github.com/user-attachments/files/17610142/SMA.Gerat.-.Gerateparameter.4.pdf)

Deepy1973, 2024-11-03:
> @deepy007 [SMA Gerät - Geräteparameter 4.pdf](https://github.com/user-attachments/files/17610142/SMA.Gerat.-.Gerateparameter.4.pdf)

Besten Dank! 
Hab mal soweit möglich (unterschiedliche WR) verglichen und angepasst. Keine Besserung!!
Auch nach Neustart der kompletten Anlage nicht

michelix11, 2024-11-03:
Ich habe jetzt soweit nochmal alles soweit verglichen mit philipp750. Die Einstellungen sind haargenau die gleichen. Mein Setup ist auch gleich, es funktioniert trotzdem nicht.
Gerade nochmal Neustart von WR, BYD Batterie und SHM2, bringt auch nichts.
Das einzigste was mir aufgefallen ist, das bei den Parametern im sunnyportal philipp750 Firmware-Version des Betriebssystems   1.02.0.R steht und bei mir steht dort 4.01.0.R .........

RichardIstSauer, 2024-11-03:
Ich habe gerade eben nochmal getestet, die Batterie aus dem Netz zu laden und plötzlich funktioniert es wieder. Das mit SMA ist doch echt ein Trauerspiel.

RichardIstSauer, 2024-11-04:
> Ein aktueller Test zeigt : Ist der Speicher auf 0% und der BWR in Stand By , lässt er kein steuern zu.
> 
> Sobald 1% drinnen sind geht alles wie gewohnt....

Kann's bestätigen, das Netzladen funktioniert erst ab 1% Batterieladung.

premultiply, 2024-11-04:
Dann ist es ein Konfigurationsproblem und die Lösung klar und wir machen hier mal zu.

closed 2024-11-04 by premultiply

IkarusPV, 2024-11-04:
> > Ein aktueller Test zeigt : Ist der Speicher auf 0% und der BWR in Stand By , lässt er kein steuern zu.
> > Sobald 1% drinnen sind geht alles wie gewohnt....
> 
> Kann's bestätigen, das Netzladen funktioniert erst ab 1% Batterieladung.

Kann ich so nicht bestätigen. Bei mir (mit sma-hybrid) startet das Netzladen unabhängig vom Ladestand nicht. Es hat keinen Einfluss, ob der Ladestand 0% oder höher ist.  Darüber hinaus handelt es sich beim SBS2.5 ja um einen anderen Wechselrichtertyp (Batteriewechselrichter), um den es hier eigentlich gar nicht geht.

michelix11, 2024-11-04:
Für mich ist nichts klar. Ich habe jetzt mal ein Ticket bei SMA eröffnet, mal gespannt wann da eine Antwort kommt. Trotzdem würde ich mich hier über weitere Lösungsansätze freuen.

cRichter86, 2024-11-06:
Hallo,
gibt es hier schon was neues? Ich habe auch die Kombination STP10, SHM2.0 und BYD. Bei mir funktioniert das Laden aus dem Netz auch nicht.

Nokius, 2024-11-11:
Gerne gebe ich meinen Senf auch noch dazu.

Bei einem Setup mit SBS 3.7, kann ich nun sehr gut nachvolziehen das evcc ein laden nicht starten kann wenn die HVM (in meinem Fall) bei 0% (evcc) ist.

Mein Workaround ist es via sunny portal das laden (Zeitgesteuert im Preisfeld) zu starten dann übernimmt evcc wieder.

etwas nervig, aber wohl ein feature das SMA will :shrug:

wittmannchris, 2024-11-11:
> Gerne gebe ich meinen Senf auch noch dazu.
>
> Bei einem Setup mit SBS 3.7, kann ich nun sehr gut nachvolziehen das evcc ein laden nicht starten kann wenn die HVM (in meinem Fall) bei 0% (evcc) ist. [...]

Habe gerade im PV Forum gelesen man soll Geduld haben. Und tatsächlich. Der SBS braucht ca. Ne Minute bis er „aufwacht“. Dann läuft es einwandfrei. Sowohl mit Modbus als auch mit EVCC

IkarusPV, 2024-11-25:
Netzladen (sma-hybrid) funktioniert jetzt mit dem aktuellen Nightly Release 0.131.7 (f4111139).