evcc-io/evcc #16977: Tariffs: improve flexibility

andig, 2024-10-30:
## Formulas

Currently, tariffs can be customized using `tax` and `charges`:

    price := (cost + charges) * (1 + tax)

where the base cost comes from an api, either for current cost or as cost forecast.

https://github.com/evcc-io/evcc/issues/16851 requested to add `margin` and `uplifts` to the cost calculation:

    price := (cost * (1 + margin) + charges) * (1 + tax) + uplifts

It has also been pointed out that this calculation is not flexible enough, as it does for example not take negative prices sufficiently into account.

Instead of https://github.com/evcc-io/evcc/pull/16855 we propose to revert https://github.com/evcc-io/evcc/pull/16855 to the original calculation and add an option for a custom, freely-definable calculation formula:

    price := (math.Abs(cost) + 0.02) * 1.19

Input to the formula is the api-provided cost, output is the final price.

Configuration of the formula could look like this:

```yaml
grid:
  type: elering
  tax: 0.19
  charges: 0.02
  formula: (math.Abs(cost) + charges) * (1 + tax)
```

## Zones

Further requirement (https://github.com/evcc-io/evcc/issues/16829) is to add zoned calculation components ("tariffs"). If we add zones, similar to the fixed tariff, to the calculation, we need to determine which price components need be definable per zone. These additional components should be accessible from the formula in the same way:

    price := (math.Abs(cost) + charges) * 1.19

where `cost` comes from the api and `charges` would be defined per zone.

VolkerK62, 2024-10-30:
So the following example would be possible?
`formula: (price + math.Abs(price) * 0.33 + charges) * (1 + tax) + 0.22`

andig, 2024-10-30:
Yes. But- as long as we do not add zone parameters- you would/could write this as:

    (price + math.Abs(price) * 0.33 + 0.11) * 1.19 + 0.22

and only depend on price!

timpetersson81, 2024-10-30:
Nice :)

naltatis, 2024-10-30:
Yes, i think having this flexibility is great. Some vendors in Germany also started to offer dynamic tariffs with a [fixed upper limit](https://www.ostrom.de/ostrom-price-cap). Would be easy to implement concepts like this with this new `formula`. 

Only drawback is, that writing a correct formula will be a challenge for non-technical users. Templates for specific tariffs would ease this.

amtssp, 2024-10-30:
Hi the zone question has been discussed before 
here: https://github.com/evcc-io/evcc/discussions/12331#discussioncomment-8595277 

There was even a commit (unfortunately I did not know how to test it so it was closed): https://github.com/evcc-io/evcc/pull/12871 
Maybe it can be reused?

RenatusRo, 2024-10-30:
I just want to remind that feedin (here in .be) for dynamic contracts is dynamic too:
for instance
0.87 * price - 0.004
total := (price * (1 + margin) + charges)
see https://github.com/evcc-io/evcc/discussions/16465#discussioncomment-10839844

closed 2024-11-06 by andig