evcc-io/evcc #16994: Vehicle selection popup menus not working in Tesla browser

g-mocken, 2024-10-30:
### Describe the bug

When I access the EVCC web interface from the browser (chromium-based, I think) in the Tesla infotainment, then everything works, except the vehicle selection.

### Steps to reproduce

Goto EVCC in LAN.
Tap on currently displayed vehicle name. Nothing happens.
Goto charging sessions, open a session and tap on vehicle name there. Nothing happens.

All other browsers on other devices display a popup with configured vehicles plus the guest vehicle.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info #debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

meters:
- type: template
  template: sma-energy-meter 
  usage: grid  
  host: smaXXX.fritz.box  
  name: grid1
- type: template
  template: sma-inverter 
  usage: pv  
  host: smaXXXXX.fritz.box  
  password: *****
  name: pv2

chargers:
- type: template
  template: phoenix-em-eth 
  id: 180  
  host: walli.fritz.box 
  port: 502  
  modbus: tcpip  
  name: wallbox3

loadpoints:
- title: Garage
  charger: wallbox3
  soc:
    estimate: true
  enable:
      delay: 5m
      threshold: -600  
  disable:
      delay: 10m
      threshold: 1000 
  guardduration: 5m
      
site:
- title: XXXX 
  meters:
    grid: grid1
    pv:
    - pv2
  residualPower: 0
  
vehicles:
- name: zoe
  title: Renault ZOE
  type: renault
  user: *****
  password: *****
  vin: *****
  capacity: 22
  onIdentify:
    mincurrent: 8
    

- name: modely
  type: template
  template: tesla
  title: Tesla Model Y
  icon: car 
  mode: PV
  minCurrent: 6 
  maxCurrent: 16 
  accessToken: *****
  refreshToken: *****
  vin: *****
```

### Log details

```text
(available on request, but I don't think HTML/JS issues will show up here)
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.4

naltatis, 2024-10-31:
Sadly the builtin Tesla browser has no (or very unreliable) support for native select elements. The evcc interfaces uses native select elements wherever possible because they provide a good user experience on the various devices we support and come with accessibility features.

We had a similar issue before where language select did not work (see https://github.com/evcc-io/evcc/issues/10459). There we solved this by improving auto-language detection. But this won't work for your use-case.

I dont see a good and easy workaround here. Introducing custom selects only for Tesla browser is a lot of work for a quite niche application.

For reference: which browser version (or software version) runs on your Tesla? Since this is a web standard, maybe Tesla will come around to fix this in the future.

closed 2024-10-31 by naltatis

g-mocken, 2024-10-31:
It's version 2024.32.7 for the whole car – don't know how to get a version for just the browser.

Yes, it's a bit sad, especially since the car selection is more a less the only part of EVCC that I would typically use in the car itself. But I guess I can create some kind of extended-URL bookmark that directly switches to the Tesla vehicle as a workaround.

naltatis, 2024-10-31:
>  But I guess I can create some kind of extended-URL bookmark that directly switches to the Tesla vehicle as a workaround.

@g-mocken I hope so. The call you want to trigger is `POST /api/loadpoints/<id>/vehicle/<name>`