evcc-io/evcc #16996: Testtoken (Sponsorship) does not work

AndyNew2, 2024-10-30:
### Describe the bug

The given testoken till 3.Nov. seems to be corrupted. Maybe the issue is, when the testperiod overlaps from previous month to next month? This was now the case. Since I updated to the latest token I always get this output:

starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/10/30 21:25:55 evcc 0.131.4
[main  ] INFO 2024/10/30 21:25:55 using config file: /config/evcc.yaml
[main  ] INFO 2024/10/30 21:25:55 listening at :7070
[main  ] FATAL 2024/10/30 21:25:55 sponsortoken: rpc error: code = Unknown desc = token has invalid claims: token is expired
[main  ] FATAL 2024/10/30 21:25:55 will attempt restart in: 15m0s

### Steps to reproduce

1. Have a config needing a sponsor token
2. Copy the current test token in the evcc.yaml
3. Start EVCC
...

### Configuration details

```yaml
I run EVCC in Home Assistant AddOn. Don't think this makes any difference, but who knows...
```

### Log details

```text
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/10/30 21:25:55 evcc 0.131.4
[main  ] INFO 2024/10/30 21:25:55 using config file: /config/evcc.yaml
[main  ] INFO 2024/10/30 21:25:55 listening at :7070
[main  ] FATAL 2024/10/30 21:25:55 sponsortoken: rpc error: code = Unknown desc = token has invalid claims: token is expired
[main  ] FATAL 2024/10/30 21:25:55 will attempt restart in: 15m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.1 and also 0.131.4

wenzul, 2024-10-30:
#16070

naltatis, 2024-10-31:
yes, looks like duplicate. Please comment, when the solution in the linked issue does not work for you.

closed 2024-10-31 by naltatis

AndyNew2, 2024-10-31:
Sure will try now. So I need to delete the EVCC.db. Hope I will find the files. Since I am in trial phase, the data is not relevant for me.

AndyNew2, 2024-10-31:
Hmm I am to stupid to find evcc.db. Seems complicated steps with docker to access that file when installed as HA AddOn. I deinstalled EVCC and install it new, in the hope that will recreated the evcc.db... Will report.

AndyNew2, 2024-10-31:
Confirmed. Reinstall fixed it. So evcc.db seems to be rebuild after reinstall, was a lot easier for me, even in the docu part everything is explained ;-) Shame on me.
And yes it fixed the broken token issue as well. Thanks for support.