evcc-io/evcc #16998: NRGkick Gen2 shows around 0.8W while not charging

wenzul, 2024-10-30:
### Describe the bug

During idle my NRGkick Gen2 consumes and shows around 0.8W and is printed as 1W. That's why my main screen often looks like this. Is that the intention? I would expect from evcc the ‘standby consumption’ to be suppressed.
[image]

Is there already a global setting to provide the charger with a small threshold value? Or does this have to be hard-coded in `func (nrg *NRGKickGen2) CurrentPower() (float64, error)`?

### Steps to reproduce

1. Query NRGkick Gen2 during not charging

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

sponsortoken: ...

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: pv1
    type: template
    template: sungrow-hybrid
    id: 1
    host: 192.168.1.145
    port: 502
    usage: pv
    modbus: tcpip
  - name: battery1
    type: template
    template: sungrow-hybrid
    id: 1
    host: 192.168.1.145
    port: 502
    usage: battery
    modbus: tcpip
    capacity: 16
  - name: my_grid
    type: template
    template: sungrow-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.1.145 # Hostname
    port: 502 # Port
    timeout: 10s # optional

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for documentation see https://docs.evcc.io/docs/devices/chargers
chargers:
  - type: template
    template: nrggen2
    id: 1
    host: 192.168.2.141
    port: 502
    modbus: tcpip
    phases1p3p: true
    name: wallbox1
# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: bmw
    type: template
    template: bmw
    title: ***
    icon: car
    user: ***
    password: ***
    vin: ***
    capacity: 37.9
    phases: 3
    cache: 10m
    mode: pv
    minCurrent: 6
    maxCurrent: 16
    region: EU
    welcomecharge: false
  - name: byd
    type: template
    template: offline
    title: ***
    icon: car
    capacity: 60.48
    phases: 3
    mode: pv
    minCurrent: 6
    maxCurrent: 16

# site describes the EVU connection, PV and home battery
site:
  title: Home # display name for UI
  meters:
    grid: my_grid # grid meter
    pv:
      - pv1 # list of pv inverters/ meters
    battery:
      - battery1 # list of battery meters
  residualPower: 0 # additional household usage margin
  maxGridSupplyWhileBatteryCharging: 0 # ignore battery charging if AC consumption is above this value

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Carport links # display name for UI
    charger: wallbox1 # charger
    vehicle: bmw
    mode: pv # default charge mode to apply when vehicle is disconnected; use "off" to disable by default if charger is publicly available

    # remaining settings are experts-only and best left at default values
    priority: 0 # relative priority for concurrent charging in PV mode with multiple loadpoints (higher values have higher priority)
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
        mode: charging
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)

# tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price (or price zones)
    type: fixed
    price: 0.369 # EUR/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0703 # EUR/kWh
  co2:
    # co2 tariff provides co2 intensity forecast and is for co2-optimized target charging if no variable grid tariff is specified
    # type: grünstromindex # GrünStromIndex (Germany only)
    # zip: <zip>

    # type: electricitymaps # https://app.electricitymaps.com/map
    # uri: <uri>
    # token: <token> # needs to be a token with forecast (not in the free tier)
    # zone: DE

    # type: ngeso # National Grid Electricity System Operator data (Great Britain only) https://carbonintensity.org.uk/
    # provides national data if both region and postcode are omitted - Choose ONE only!
    # region: 1 # optional, coarser than using a postcode - The region details are at https://carbon-intensity.github.io/api-definitions/#region-list
    # postcode: SW1 # optional - Outward postcode i.e. RG41 or SW1 or TF8. Do not include full postcode, outward postcode only

# mqtt message broker
mqtt:
  # broker: localhost:1883
  # topic: evcc # root topic for publishing, set empty to disable
  # user:
  # password:

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# eebus credentials
eebus:
  # uri: # :4712
  # interfaces: # limit eebus to specific network interfaces
  # - en0
  # certificate: # local signed certificate, required, can be generated via `evcc eebus-cert`
  #   public: # public key
  #   private: # private key

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
[nrggen2] TRACE 2024/10/30 00:57:02 modbus: send 00 c8 00 00 00 06 01 03 00 d2 00 02
[nrggen2] TRACE 2024/10/30 00:57:02 modbus: recv 00 c8 00 00 00 07 01 03 04 03 01 00 00
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.3

naltatis, 2024-10-31:
@andig das Standby Power Problem kommt mir bekannt vor. Wie gehen wir denn an anderen Stellen damit um? In der UI unterdrücken, würd ich nicht machen. Pauschal ist das ja schwierig. Gibt mit Schaltsteckdosen ja auch Ladepunkte, wo kleine Leistungen durchaus angezeigt werden sollen.

andig, 2024-10-31:
Dann ist das works as designed

closed 2024-10-31 by andig

wenzul, 2024-10-31:
Hallo zusammen,
danke für eure Antworten.
Dennoch finde ich das wegen <1W auf der UI wirklich nicht schön. In der NRGkick App selbst steht 0W. Standby Power ist <1W auch zu wenig denke ich. Es wird einfach ein ADC Rauschen sein...
[image]
Ist es so abwegig nur für <1W zu 0W abzurunden oder in der Grafik nicht zu erfassen?

Offtopic: Aktuell hat der NRGkick bei Umschaltung des Ladesstroms oder Anzahl der Phasen auch per API ein akustisches Signal. Ich habe eine Anfrage zu Dinitech geschickt, sie werden vermutlich den akustischen Ton vielleicht per API zum Wohl meiner Nachbarn steuerbar in einer der nächsten Releases rausgeben.

premultiply, 2024-10-31:
Prima, dann können sie bei der Gelegenheit auch das Metering fixen. ;)