evcc-io/evcc #17005: Eingabefenster für Konfiguration → Regelverhalten hat mehrere Probleme

SolarPowerEV, 2024-10-31:
### Describe the bug

Zum einen steht im Feld "Max Netzbezug während des Ladens" kein Wert mehr drin (zumindest wenn die Variable auf 0 steht), sodass man beim Speichern erstmal aufgefordert wird, dort "eine Nummer" (sic!) einzutragen. 
Hat man dann einen Wert eingetragen, kommt beim Speichern ein 404-Error. 

Das Problem tritt sowohl im Browser als auch am PC auf.

[image]

[image]

[image]

P.S.: den Haken bei "I have verified that the issue is reproducible with the latest nightly build *" habe ich nur gesetzt, weil mich Github sonst dieses Formular nicht abschicken lässt!!!

### Steps to reproduce

1. evcc im Browser öffnen
2. auf Konfiguration klicken
3. auf Regelverhalten klicken
4. Einen der ersten beide Werte verändern und auf Speichern klicken
→ evcc beschwert sich über den fehlenden Wert in "Max Netzbezug"

5. dort einen Wert eintragen
6. Speichern klicken

"Request failed with status code 404" wird in rot angezeigt.
...

### Configuration details

```yaml
an der Konfiguration habe ich nichts verändert zu vorher. Daher muss der Bug mit einer der V0.131.x eingeführt worden sein, denn vorher trat der Fehler nicht auf.
```

### Log details

```text
Request failed with status code 404.POST http://amadeus:7070/api/maxgridsupply/0
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.3

VolkerK62, 2024-10-31:
ist im aktuellen Nightly gelöst
https://github.com/evcc-io/evcc/pull/16984

andig, 2024-10-31:
>  I have verified that the issue is reproducible with the latest nightly build

Das hat sich wohl überschnitten.

closed 2024-10-31 by andig