evcc-io/evcc #17007: 0.131.4 Battery discharges also it is locked

Peter1984Git, 2024-10-31:
### Describe the bug

In 0.131.4 the battery discharges in Fast Mode also discharging isn´t allowed.

[image]

### Steps to reproduce

1. Set charging mode to fast.
2.
3.
...

### Configuration details

```yaml
nothing changed
```

### Log details

```text
none
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.4

VolkerK62, 2024-10-31:
ohne Config und Log wird das sofort geschlossen werden.
Man weiß ja noch nicht mal, um welche Hardware es sich handelt.

andig, 2024-10-31:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-10-31 by andig

Peter1984Git, 2024-11-01:
Hi and sorry, I was in hurry yesterday... here is a log file. 
I set charge mode to now at 13:11:26.
At time stamp 13:12:10 you can see that Battery mode is "hold" (why isn´t that set before charging starts?) but Battery Power is used.

[evcc-20241101-131430-debug.log](https://github.com/user-attachments/files/17599547/evcc-20241101-131430-debug.log)

andig, 2024-11-08:
> Hi and sorry, I was in hurry yesterday... here is a log file.

In Deinem Issue fehlt jegliche angefragte Information. Ich sollte daher eigentlich nicht antworten, aber: 

> Hardware: Fronius Gen24

Es gibt 2 Templates. Probier mal das Andere...