evcc-io/evcc #17008: Energieanzeige bei gesperrtem Akku

maatinh, 2024-10-31:
### Describe the bug

Wenn ich die Batterie auf einen Modus einstelle, indem sie nur lädt, aber nicht entlädt und das Haus nur vom Netz versorgt wird, ist die Anzeige nicht falsch, aber es wäre richtiger, wenn die Batterie unter der Sonne angezeigt würde und das Haus unter dem Grid.

[image]

[image]

### Steps to reproduce

Siehe Beschreibung 

### Configuration details

```yaml
Nicht relevant
```

### Log details

```text
Keine
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

131.3

naltatis, 2024-10-31:
Heißt du versetzt deinen Hybrid-Wechselrichter in einen Zustand bei dem er nur DC-DC PV-Strom in Hausspeicher lädt? So etwas können wir heute gar nicht abbilden bzw. von anderen Szenarien unterscheiden. Wir betrachten Produktion und Aufnahme unabhängig voneinander. Das zieht sich nicht nur durch die UI sondern auch durch die zugrundeliegenden Berechnungen (bspw. die 12ct Stromkosten für den Hausverbrauch). Daher würde ich jetzt hier erst einmal nichts machen.

Aus Interesse: Was ist dein Anwendungsfall hier? Lieber mehr in den Hausakku um dann zu teuren Zeiten (dyn Tarif) damit überberbrücken zu können?

closed 2024-10-31 by naltatis

maatinh, 2024-10-31:
Ok, kein Problem. Wollte es nur mal angesprochen haben.

Ja, der Use-Case ist in Verbindung mit einem dynamischen Stromtarif.
Das Preisdelta zwischen günstig und teuer ist nicht so groß, dass sich Netzladen inkl. der ca 10% Verlust rechnet um für später ne volle Batterie zu haben. 
Da nutze ich lieber den PV Strom um nur die Batterie zu laden und versorge das Haus in der Zeit aus dem Netz bis der Strom. Die zwei, drei teuren Stunden heute kann die Batterie dann gut überbrücken.

Bei Fenecon heißt das scheinbar Verzögerte Entladung: https://www.pv-magazine.de/2024/10/30/fenecon-app-ermoeglicht-intelligente-nutzung-dynamischer-stromtarife/