evcc-io/evcc #17020: Support current source type for EEBUS (Porsche Mobile Connect Charger)

spatialdude, 2024-11-01:
Hi

I'm using a Porsche Mobile Connect Charger with evcc, having switched from using it with the [Porsche HEM](https://press.porsche.com/prod/presse_pag/PressResources.nsf/Content?ReadForm&languageversionid=1292912) due to the lack of support for a home battery by the Porsche HEM. 

The PMCC works brilliantly with EVCC! 🥇🥇

A difference I did notice is that although EVCC is correctly managing the PMCC, the charger still thinks the current source is from the grid rather than solar as shown in the following two screen shots. The first is from the charger's dashboard web page and second is what is shown in evcc for the same session and the last is from the charge history in the My Porsche app.

[image]

[image]

[image]

While if the charger is connected to the Porsche HEM, both the PMCC dashboard and My Porsche app correctly indicate the session's current source as solar or grid.

This difference doesn't affect charging, evcc correctly habdles the charging sessions as per the desired configuration.

Is it possible for evcc to report the current source type to a EEBUS charger?

It would be nice a nice finishing touch if the charge session correctly reported the energy sources used.  

For reference this is my configuration for the charger -

```yaml
  - type: template
    template: pmcc 
    ski: e4836c94b3a8b035b5cf191564acebb8a497e203  
    ip: 192.168.0.123  
    name: pmcc
```

DerAndereAndi, 2024-11-01:
This is not possible because of the way evcc works compared to how the Porsche HEMS works.

closed 2024-11-01 by andig

spatialdude, 2024-11-03:
Ok. Thank-you for the response.

andig, 2025-02-12:
>  in evcc.yaml?

Yes. Please use discussions for questions.