evcc-io/evcc #17028: Juicebox / EeMotorWerks / EnelX

BeneWilh, 2024-11-01:
Enelx is shuting down its servers at least in the US an in EU they stopped providing an API to talk to their Wallboxes. The only way in EU is via the app.

There is a project [https://github.com/JuiceRescue](url) which is giving the chance to read the values from the Juicebox to mqtt.

Can somebody help to create a template to read the values in EVCC... and later write to mqtt settings? Writing settings need still to be implemented in juicepassproxy, so this is something for the future.

I am reading the values in node-red, adjusting values and topic and created a costum charger in evcc... many steps to configure.

this is how it looks like in MQTT Explorer

[image]

to

[image]

andig, 2024-11-01:
> Writing settings need still to be implemented in juicepassproxy, so this is something for the future.

There's no point in implementing a custom charger if you can't write?

BeneWilh, 2024-11-01:
At first glance it would be nice to have the template reading the values directly (hmd tree) without the need of node-red in the middel... 

Of cource evcc make mucho more sense if you can steer as well.. i guess at least my limited knowledge forces me to wait until steering is implemented in juicepassproxy.