evcc-io/evcc #17037: UI: Einzelne Ladevorgägnge nur Monat, nicht Gesamt oder Jahr

lmfgpeace, 2024-11-01:
### Describe the bug

Der Titel beschreibt es schon ganz gut.

Beim klicken auf „Einzelne Ladevorgänge anzeigen“ unter dem Reiter „Jahr“ oder „Gesamt“, springt die Übersicht zurück auf die Monatsansicht, und es werden lediglich die Lade Vorgänge des jeweiligen aktuellen Monats angezeigt. Eine Gesamt beziehungsweise Jahresübersicht lässt sich somit nicht erstellen.

### Steps to reproduce

1. „Ladevorgänge“ öffnen
2. Auf „Jahr“ oder „Gesamt“ klicken
3. „Einzelne Ladevorgänge anzeigen“ anklicken
4. Springt dann zurück auf „Monat“

### Configuration details

```yaml
n.e.
```

### Log details

```text
n.e.
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.0131.4

TobiasHuber1980, 2024-11-02:
du kannst dir die komplette csv-datei herunterladen, da hast du alle ladevorgänge....

naltatis, 2024-11-02:
Ja, auf Jahr/Gesamt-Ansicht haben wir aktuell bewusst keine Einzelvorgänge. Klar könnten wir das machen. Dann mit Paging und so. Aber das ist dann eigentlich sehr ähnlich zu dem was wir ohnehin schon in der Monatsansicht haben 😄

closed 2024-11-02 by naltatis