evcc-io/evcc #17042: Problems with Elli Wallbox: operation is not supported on function loadControlLimitConstraintsListData

jpe-databricks, 2024-11-02:
### Describe the bug

Apologies if this has been answered elsewhere - I wasn't able to find a fix:
I own a VW Id Charger Pro, connected via Ethernet, current software version. EVCC is registered as the energy manager for the Wallbox. The overall setup works fine. 
However, when trying to load my car, the IOS VW App reports me repeatedly a connection error and the charging doesn't start.  The log file shows different error message:
- " operation is not supported on function loadControlLimitConstraintsListData"
- "websocket read error:  websocket: close 1006 (abnormal closure): unexpected EOF"
- "Power:           0W
Current L1..L3:  0A 0A 0A
.Soc:            not available
Charge status:   A
Enabled:         false
Mix/Max Current: 0.0/0.0A
Identifier:      <none>
[eebus ] DEBUG 2024/11/02 13:49:21 mdns: stop announcement
"
Rebooting doesn't fix the issue.

Any recommendations which information to provide/how to proceed?

I appreciate the help and support.

### Steps to reproduce

Connect car to charger

### Configuration details

```yaml
/etc/evcc.yaml                                                                         
  usage: battery  
  host: 192.168.178.27  
  port: 5033
  user: XXX
  password: XXX
  key: XXX
  capacity: 8.6
  name: jpee3dcbattery
chargers:
- type: template
  template: ellipro 
  ski: XXX  
  ip: 192.168.178.55  
  name: vwwallbox 

vehicles:
- type: template
  template: vw 
  title: VW ID3  
  user: XXX 
  password: XXX  
  capacity: 58  
  timeout: 10s
  minCurrent: 6  
  maxCurrent: 16      
  phases: 3 # Die maximale Anzahl der Phasen welche genutzt werden können
  cache: 15m # Zeitintervall nach dem Daten erneut vom Fahrzeug abgefragt werden
  name: JPEID3

loadpoints:
- title: Garage
  charger: vwwallbox
  mode: pv
  vehicle: JPEID3

site:
  title: JPE Home
  residualPower: 200
  meters:
    grid: e3dcgrid
    pv:
      - jpefronius
      - jpee3dcpv 
    battery:
      - jpee3dcbattery

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          XXX
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
         XXX
          -----END CERTIFICATE-----
```

### Log details

```text
evcc -l debug charger
[main  ] INFO 2024/11/02 14:02:47 evcc 0.131.4
[main  ] INFO 2024/11/02 14:02:47 using config file: /etc/evcc.yaml
[db    ] INFO 2024/11/02 14:02:47 using sqlite database: /home/jan/.evcc/evcc.db
[eebus ] INFO 2024/11/02 14:02:47 Local SKI: 90dd4e6c313da0545742bd733eb9d34320d94dae
[eebus ] DEBUG 2024/11/02 14:02:47 starting websocket server on :4712
[eebus ] ERROR 2024/11/02 14:02:47 websocket server error: listen tcp :4712: bind: address already in use
[eebus ] DEBUG 2024/11/02 14:02:47 mdns: announce
[eebus ] DEBUG 2024/11/02 14:02:47 mdns: using avahi
[eebus ] DEBUG 2024/11/02 14:02:54 mdns: new - ski: 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 name: Elli-Wallbox-2205A1ZL6D brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2205A1ZL6D register: false host: wallbox-2205A1ZL6D.local port: 4712 addresses: [192.168.178.55]
[eebus ] DEBUG 2024/11/02 14:02:54 trying to connect to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 at wallbox-2205A1ZL6D.local
[eebus ] DEBUG 2024/11/02 14:02:54 initiating connection to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 at wallbox-2205A1ZL6D.local:4712/ship/
[eebus ] DEBUG 2024/11/02 14:02:54 connection to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 failed: dial tcp: lookup wallbox-2205A1ZL6D.local on 192.168.178.1:53: no such host
[eebus ] DEBUG 2024/11/02 14:02:54 trying to connect to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 at 192.168.178.55
[eebus ] DEBUG 2024/11/02 14:02:54 initiating connection to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 at 192.168.178.55:4712/ship/
[eebus ] DEBUG 2024/11/02 14:02:54 ski 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 connected
[eebus ] DEBUG 2024/11/02 14:02:54 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/11/02 14:02:54 Recv: d:_i:47859_Elli-Wallbox-2205A1ZL6D:NodeManagement to NodeManagement read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/11/02 14:02:54 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D reply 2 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/11/02 14:02:54 Recv: d:_i:47859_Elli-Wallbox-2205A1ZL6D:NodeManagement to NodeManagement reply 3 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/11/02 14:02:54 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D call 3 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/11/02 14:02:54 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D read 4 NodeManagementUseCaseData
[eebus ] DEBUG 2024/11/02 14:02:54 cs-lpp: 0 DeviceDiagnosis Server found
[eebus ] DEBUG 2024/11/02 14:02:54 cs-lpc: 0 DeviceDiagnosis Server found
[eebus ] DEBUG 2024/11/02 14:02:54 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D read 5 DeviceClassificationManufacturerData
[eebus ] DEBUG 2024/11/02 14:02:54 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D read 6 DeviceDiagnosisStateData
[eebus ] DEBUG 2024/11/02 14:02:54 ski 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 event cem-evsecc-EvseConnected
[eebus ] DEBUG 2024/11/02 14:02:54 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 websocket read error:  websocket: close 4201: Double connection
[eebus ] DEBUG 2024/11/02 14:02:54 ski 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 disconnected
[eebus ] DEBUG 2024/11/02 14:02:54 mdns: announce
[eebus ] DEBUG 2024/11/02 14:02:54 mdns: using avahi
[eebus ] DEBUG 2024/11/02 14:02:54 delaying connection to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 by 3.389s to minimize double connection probability
Power:           0W
Current L1..L3:  0A 0A 0A
.Soc:            not available
Charge status:   A
Enabled:         false
Mix/Max Current: 0.0/0.0A
Identifier:      <none>
[eebus ] DEBUG 2024/11/02 14:02:57 mdns: stop announcement
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.4

jpe-databricks, 2024-11-02:
_evcc charger_ has the following output

```
[main  ] INFO 2024/11/02 14:05:31 evcc 0.131.4
[main  ] INFO 2024/11/02 14:05:31 using config file: /etc/evcc.yaml
[db    ] INFO 2024/11/02 14:05:31 using sqlite database: /home/jan/.evcc/evcc.db
[eebus ] INFO 2024/11/02 14:05:31 Local SKI: 90dd4e6c313da0545742bd733eb9d34320d94dae
[eebus ] DEBUG 2024/11/02 14:05:31 starting websocket server on :4712
[eebus ] ERROR 2024/11/02 14:05:31 websocket server error: listen tcp :4712: bind: address already in use
[eebus ] DEBUG 2024/11/02 14:05:31 mdns: announce
[eebus ] DEBUG 2024/11/02 14:05:31 mdns: using avahi
[eebus ] DEBUG 2024/11/02 14:05:31 mdns: new - ski: 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 name: Elli-Wallbox-2205A1ZL6D brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2205A1ZL6D register: false host: wallbox-2205A1ZL6D.local port: 4712 addresses: []
[eebus ] DEBUG 2024/11/02 14:05:31 trying to connect to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 at wallbox-2205A1ZL6D.local
[eebus ] DEBUG 2024/11/02 14:05:31 initiating connection to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 at wallbox-2205A1ZL6D.local:4712/ship/
[eebus ] DEBUG 2024/11/02 14:05:31 mdns: update - ski: 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 name: Elli-Wallbox-2205A1ZL6D brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2205A1ZL6D register: false host: wallbox-2205A1ZL6D.local port: 4712 addresses: [192.168.178.55]
[eebus ] DEBUG 2024/11/02 14:05:31 trying to connect to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 at wallbox-2205A1ZL6D.local
[eebus ] DEBUG 2024/11/02 14:05:31 initiating connection to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 at wallbox-2205A1ZL6D.local:4712/ship/
[eebus ] DEBUG 2024/11/02 14:05:31 connection to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 failed: dial tcp: lookup wallbox-2205A1ZL6D.local on 192.168.178.1:53: no such host
[eebus ] DEBUG 2024/11/02 14:05:31 trying to connect to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 at 192.168.178.55
[eebus ] DEBUG 2024/11/02 14:05:31 initiating connection to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 at 192.168.178.55:4712/ship/
[eebus ] DEBUG 2024/11/02 14:05:31 connection to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 failed: dial tcp: lookup wallbox-2205A1ZL6D.local on 192.168.178.1:53: no such host
[eebus ] DEBUG 2024/11/02 14:05:31 trying to connect to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 at 192.168.178.55
[eebus ] DEBUG 2024/11/02 14:05:31 initiating connection to 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 at 192.168.178.55:4712/ship/
[eebus ] DEBUG 2024/11/02 14:05:31 ski 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 connected
[eebus ] DEBUG 2024/11/02 14:05:31 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/11/02 14:05:31 Recv: d:_i:47859_Elli-Wallbox-2205A1ZL6D:NodeManagement to NodeManagement read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/11/02 14:05:31 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D reply 2 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/11/02 14:05:31 Recv: d:_i:47859_Elli-Wallbox-2205A1ZL6D:NodeManagement to NodeManagement reply 3 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/11/02 14:05:31 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D call 3 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/11/02 14:05:31 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D read 4 NodeManagementUseCaseData
[eebus ] DEBUG 2024/11/02 14:05:31 cs-lpc: 0 DeviceDiagnosis Server found
[eebus ] DEBUG 2024/11/02 14:05:31 cs-lpp: 0 DeviceDiagnosis Server found
[eebus ] DEBUG 2024/11/02 14:05:31 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D read 5 DeviceClassificationManufacturerData
[eebus ] DEBUG 2024/11/02 14:05:31 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D read 6 DeviceDiagnosisStateData
[eebus ] DEBUG 2024/11/02 14:05:31 ski 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 event cem-evsecc-EvseConnected
[eebus ] DEBUG 2024/11/02 14:05:31 closing existing double connection
[eebus ] DEBUG 2024/11/02 14:05:31 ski 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 disconnected
[eebus ] DEBUG 2024/11/02 14:05:32 ski 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 connected
[eebus ] DEBUG 2024/11/02 14:05:32 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/11/02 14:05:32 Recv: d:_i:47859_Elli-Wallbox-2205A1ZL6D:NodeManagement to NodeManagement read 10 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/11/02 14:05:32 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D reply 2 10 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/11/02 14:05:32 Recv: d:_i:47859_Elli-Wallbox-2205A1ZL6D:NodeManagement to NodeManagement reply 12 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/11/02 14:05:32 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D call 3 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/11/02 14:05:32 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D read 4 NodeManagementUseCaseData
[eebus ] DEBUG 2024/11/02 14:05:32 cs-lpc: 0 DeviceDiagnosis Server found
[eebus ] DEBUG 2024/11/02 14:05:32 cs-lpp: 0 DeviceDiagnosis Server found
[eebus ] DEBUG 2024/11/02 14:05:32 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D read 5 DeviceClassificationManufacturerData
[eebus ] DEBUG 2024/11/02 14:05:32 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D read 6 DeviceDiagnosisStateData
[eebus ] DEBUG 2024/11/02 14:05:32 ski 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 event cem-evsecc-EvseConnected
[eebus ] DEBUG 2024/11/02 14:05:32 Recv: d:_i:47859_Elli-Wallbox-2205A1ZL6D:NodeManagement to NodeManagement call 14 NodeManagementSubscriptionRequestCall
[... cut ...]
[eebus ] DEBUG 2024/11/02 14:05:34 Recv: d:_i:47859_Elli-Wallbox-2205A1ZL6D:Measurement to Measurement reply 75 22 MeasurementDescriptionListData
[eebus ] DEBUG 2024/11/02 14:05:34 Send: d:_i:47859_Elli-Wallbox-2205A1ZL6D read 34 MeasurementListData
Power:           0W
Current L1..L3:  0A 0A 0A
.Soc:            not available
Charge status:   A
Enabled:         false
Mix/Max Current: 0.0/0.0A
Identifier:      <none>
[eebus ] DEBUG 2024/11/02 14:05:34 mdns: stop announcement
[eebus ] DEBUG 2024/11/02 14:05:34 ski 225815d9318cbc5a8983ddd734dbbc51dbc29ec0 disconnected
```

andig, 2024-11-02:
@DerAndereAndi ich sehe die Meldung, kann damit aber nichts anfangen. Ist das ein Thema im Stack oder ein Quirk der Box?

jpe-databricks, 2024-11-03:
Danke für die schnelle Antwort - bin Informatiker, aber habe keine wirkliche Erfahrung mit Wallboxen.
Wie kann ich rausfinden ob es im Stack oder ein Quirk ist (ein schnelle Google-Suche hat mir kein Ergebnis gebracht) ?

DerAndereAndi, 2024-11-03:
Kurze Antwort: Es gibt hier nichts zu sehen. Kein Fehler im Stack oder evcc.

Punkt 1: Wurde bereits mehrfach angesprochen: https://github.com/evcc-io/evcc/issues?q=operation+is+not+supported+on+function+loadControlLimitConstraintsListData
Die Ausgabe ist `Debug` und nicht `Error`, hier ist kein Fehler vorhanden. Eine optionale Funktion wird von der Wallbox nicht unterstützt.

Punkt 2: Ist ein dokumentierter Fehler in der Wallbox und wurde schon sehr oft besprochen: https://github.com/evcc-io/evcc/issues?q=websocket+read+error%3A+websocket%3A+close+1006+%28abnormal+closure%29%3A+unexpected+EOF
Und Problem 5 unter https://github.com/evcc-io/evcc/discussions/15367

Punkt 3: Beide Systeme haben sich gleichzeitig verbunden und die Wallbox hat beide Verbindungen geschlossen. Lösung:  Warten bis evcc oder die Wallbox eine neue Verbindung aufbaut oder evcc neu starten.

andig, 2024-11-03:
Danke @DerAndereAndi. Den `loadControlLimitConstraintsListData` hatte ich nicht präsent-

closed 2024-11-03 by andig

jpe-databricks, 2024-11-04:
Danke für den schnellen Support