evcc-io/evcc #17051: Sessions - Suggestion for change of title: Solar to Energy

RTTTC, 2024-11-03:
As can be seen in the print screen "Solar" makes sense - when "Total" is clicked.
But makes no sense to have a title "Solar" for Charging Point or Vehicle, (as it shows combined solar and Grid energy)
I think it would be better to change from "Solar" to "Energy" 
@naltatis What do you think ?

[image]

[image]

naltatis, 2024-11-03:
You're talking about the chart type selector (solar, price, co2), right?

I would not change the label based on another selected option (total, loadpoint, vehicle). In general, this creates confusion and possible layout shifts (languages).

But this would not be my main reason. The solar view still shows solar share data when you've selected a loadpoint. Especially if you've multiple entries (loadpoints, vehicles) you get the second row of charts that show solar % and amount per device.

For your special case (solar + 1 loadpoint) I'd rather think about graying out the option (which creates new challenges), then changing the solar option name.

closed 2024-11-03 by naltatis

RTTTC, 2024-11-03:
I actually have two wallboxes and two EV's, and it visually does not change anything per se, when I switch to previous month which has much more sessions and data. 
Yes, there is a second row shows more "solar data" - it would still fall under "energy" and be correct.

On the contrary - I was happy to get a printscreen with just a couple of charging days and one vehicle and one loadpoint.
Because this clearly shows that a chart name "Solar" is incorrect, because graphically the **top chart** does not show solar %, it shows _total energy_. (charging point, vehicle) 
That's why it instantly looked confusing or incorrect to see chart name "solar" and right below to see energy bars which are clearly grid energy.

I still think it would be better simply to rename the button to "**Energy**" (not to make it change its name dynamically)
This name would make it universal.
Any feedback from @andig @premultiply @VolkerK62 ?