evcc-io/evcc #17075: Charge limit for charging house battery from grid

Joshua-lab-dev, 2024-11-04:
I love the charge house battery from grid function and I'm utterly thankful for this great development. This is really the next step in revolutionizing energy usage at home.
Here is my wish for one of the next iterations: I'd appreciate an upper charging limit for this functionality to leave some room for solar energy that is produced unexpectedly, and to also protect the house battery. Would this be possible? What do you think? :)

ljouon, 2024-11-04:
I second this request. 😃 
It points in the same direction of the [idea mentioned here](https://github.com/evcc-io/evcc/issues/16673#issuecomment-2446236543) to handle the house battery with a similar feature set as loading a vehicle.

andig, 2024-11-04:
There are currently no plans to work on this. You can still use the API to disable grid charging once battery is sufficiently full.

naltatis, 2024-11-04:
There is a larger thread where we collect ideas for the next iteration of this feature:
https://github.com/evcc-io/evcc/discussions/16364

closed 2024-11-04 by naltatis