evcc-io/evcc #17080: Wrong MQTT Config via WebUI ends in dead server

poggenpower, 2024-11-04:
### Describe the bug

If you configured the wrong credentials/certificates for MQTT, evcc is not able to start.
As there is no access to the UI, there is no way to fix it, as accessiong the sqlite db

### Steps to reproduce

1. Create a MQTT integration with e.g. wrong password in the WebUI
2. Restart the Server to make this config active
3. evcc is logging an authentication issue with MQTT and runs into 15 Min restart loop

The only way to make it working again, was to remove the `mqtt` settings from the settings table in the sqlite db. As the container doesn't have a sqlite client, it requires some efforts to do so, if the DB is stored in a docker volume or k8s PV. 

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

main  ] INFO 2024/11/04 17:14:03 evcc 0.131.4

naltatis, 2024-11-04:
Hi @poggenpower,

thanks for reporting this. But I'm unable to reproduce. When I follow your instruction (configure MQTT via Config UI with wrong credentials, restart) I still have access to the Web UI after restart. evcc is in startup-error-mode and tries to recover every 15min, but this is expected behavior right now. Editing config (remove, reconfigure MQTT) via UI is still possible in this mode.

[image]

poggenpower, 2024-11-04:
Ahh, my fault. I use `evcc health` as readinessProbe and livenessProbe, as this command is failing it blocks the service coming up and make the UI reachable from external. (Running evcc in k8s)

closed 2024-11-04 by poggenpower

poggenpower, 2024-11-04:
For all who stumble into this situation with Kubernetes. Use a `port forward` to fix this issue and go on ;-) No need to touch the health check.