evcc-io/evcc #17093: Selection for fast charge with cheap prices only in 0,10€ steps

Stefanos-web, 2024-11-05:
### Describe the bug

i can select only the fast charge option only in 0,10€ steps?

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: https
  host: evcc
  port: 7070

log: debug

levels:
  cache: error

# unique installation id
plant: 4907b3213ca77234761b8cc82067dc0a1f0b7576ce9ec7f2832b8be274cce52f

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoiU3RlZmFub3Mtd2ViIiwiZXhwIjozMDEwMjA0ODAwLCJpYXQiOjE3MTQyMDQ4MDAsInNyYyI6ImdoIn0.jLghV5W72hVoMpLn0twYQ2xlXNO0bHxq6yMetf4gQx0

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.153  
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 10s  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.153  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 10s  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.153  
  port: 502  
  usage: battery  
  modbus: tcpip  
  timeout: 10s  
  name: battery3
- type: template
  template: sungrow-inverter 
  id: 1  
  host: 192.168.178.150  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv4

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.178.162  
  name: wallbox5
- type: template
  name: wallbox4
  template: go-e-v3
  host: 192.168.178.52

vehicles:
- type: template
  template: bmw 
  title: BMW i3  
  user: s.riedelsheimer@gmail.com  
  password: Ajcsr28081409#  
  vin: WBY8P210307D92098  
  capacity: 43  
  name: ev5
- name: eqe300
  type: template
  template: mercedes
  title: EQE 300
  user: s.riedelsheimer@gmail.com
  region: EMEA
  accessToken: eyJhbGciOiJkaXIiLCJlbmMiOiJBMjU2Q0JDLUhTNTEyIiwiY3R5IjoiSldUIiwia2lkIjoiQ0lBTVNZTSIsInBpLmF0bSI6Imh4M2UifQ..UghZqzMQ6ELiyp1EPG0zOA.B5VmtFOjqQeUXFmGtWJiNxJqRMHVt_4SYmMFJjsJRGHqV1nFm9NxSV6aLpYrkXYCg4_yfyMcS5ahKcHnCrLQRycQ5RrGtO6d7Antu9n1za0QX67u4EWeyvITCB0uHBqzlPTDIJt3QjqetIGWJIhPx0zj7jcaPzqu1O1d2-m4XJDt-PZ0QVYTr8DIQUCkJZI4YQ3mPQXaI-ILx_vgeaTQQJWXQ9OADZEL_vKlp3HXxeahTzDzxMcFeFjH1u_-1E6nBpAqsTW0NiwuibK8F6C0HjxzMn_ha7m5dN9dcF77fnMCxDL16mH3g-YtEeyi3XMHLnKucMmA_TgTywLRU_mskZLZEdZo7OczHIaBm9cKn9xVIoFd9ukCQNb9afoNyI2n2cUMISpCBrWwMvL1ZYaFcHe6xXb3jheD5aPZCF_nBoeYq5K1qa_tQ8ZWcVV3hHdWGxNi7T3Q_ltTaDrPbG1ccEn_EyQouJivEcVEGK9DLBb59xEQx--fcy_ozYYy9FE9QURa9JrxFzkLfJw7KWtm0ZJeuxcEJTTGRdnUrQqRMkHhQVJeLE9IeuA56WPwKnwIgXrAyjd5QAYyE2D-rNc9B8MlgYhGcffw-1BP4aEaVKJj6x_u7cPk1OjMngkk2CYJXPNRe6TsKDKuaHAmzAXR_gEXvIJIQqV7TLj8PGrcHu_7GHY3dEhrHooii6MfmuQaFy0_51EcHmBmdpaKL9lKnTLx4wOA-nMtrxLwxN2ViJbuwl7cTI-SPnudS91qoAIoaYPvzu-ggDwx-j5NziNzW86YV3_Jb4rNWnxhwwOIEHQwl5AcRmYe181uVq9qS_S1anvQHcpgDnxb5GJqZxikVaRR1KxdQ_QCPOD1ios58tGZz2lP0GoOg5Y16vR82F892uNb9vs3Bjvs2Q8iKAKPY6CEf4jSoyah9MxQiYg4KYpzOj9qMrt_RurHB6RGWY35zzLRt2trCTl8tPHk7y1Lga67ALpo4tcFUruj-3XQ3fSIxVdkAhDYO1asSRmdthDLMqDYV-8R3EXD5iART_Ybw3_dGTBeXWycQXZy-1iKkc4.X0OjgOtELVA2LrHMaZ-9O92E-Sh8r6DmKKmT2GFhRkY
  refreshToken: 8ScKGKe4aBLh3146qwYcqf5MEFRFiRwD0dh.U9P1NI
  vin: W1KEG1BB1PF036900
  capacity: 89
 
tariffs:
  currency: EUR
  grid:
    type: tibber
    token: "S8gj5XJTXjv40Hk1MO03HRxtrhygL1LX1UMbfqk1iRo"
  co2:
    type: grünstromindex
    zip: 86405
    
circuits:
  - name: main
    title: Hauptstromkreis
    maxCurrent: 45
    maxPower: 31000
    meter: grid1
  - name: garage
    title: Garage
    maxCurrent: 32
    parent: main

loadpoints:
- title: Garage Links
  charger: wallbox5
  mode: pv
  circuit: garage
- title: Garage Rechts
  charger: wallbox4
  mode: pv
  circuit: garage

site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv2
    - pv4
    battery:
    - battery3
  residualPower: 0 # additional household usage margin
  
influx:
  url: http://192.168.178.57:8086
  database: evcc # InfluxDB v2.x uses term `bucket` but for compatibility still named `database` here
  token: E9MzWVpZSE4Ht1ZxAWBCmdQ971EOYg6b05lz5qhJqHX9FsZD5YcViGR9tQP15c-ivaZwh2XU7A0yNnWiy9Ps9A==
  org: Home
```

### Log details

```text
no error
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.131.4

jvonrueden, 2024-11-05:
Heyho,
not helpful for your comment, but you have to delete your personal Informations!

- plant
- sponsortoken
- user
- password
- vin
- user
- accestoken
- refreshtoken
- Tiber token

Neisi, 2024-11-05:
+1 ...why changed it? Before it was much better

VolkerK62, 2024-11-05:
Maybe it depends on the spread of the price forecast

naltatis, 2024-11-05:
Yes, as @VolkerK62 said, the steps are dependent on the spread of prices to work with all currencies and price situations.

naltatis, 2024-11-05:
We could change the way tapping on a specific timeslot works. Currently, it selects the text higher step. We could instead apply the exact value of the slot as the limit.

jvonrueden, 2024-11-05:
but is it necessary to open this function to the high prices? 
I (and I think the most users) use the function to start charging, when the price ist low. I don't care if it is possible to charge at 60 ct/kWh. 
So if you represent €, lower prices as - 10 ct and higher prices as 50 ct per kWh won't be needed for this function. 
For me, all above 40 ct per kWh is not needed and if I need to charge at this prices, I can start it manually. 

But it would be helpful to select 30,5 ct, 31,0 ct, 31,5 ct , .... and not only 30 ct, 31 ct, 32 ct , ....

naltatis, 2024-11-05:
@jvonrueden acceptable price ranges vary from country to country and over time. We cant make assumptions or hard code ranges per region.

jvonrueden, 2024-11-05:
Yeah, understandable, I like and prefer your solution #17099 ! :)

Al3xand3r1987, 2024-11-06:
You should be able to enter the prices manually, and then it doesn’t matter what is displayed. This way, you understand the scaling factor, but in phases where high and low prices differ so significantly, it doesn't make sense for us "smart chargers."

naltatis, 2024-11-06:
Quick reminder: Via API you can always set arbitrary values.

Example
´´´
curl http://evcc.local:7070/api/batterygridchargelimit/0.35 -X POST
```

Basti77, 2024-11-06:
Today, electricity prices at Tibber are skyrocketing, at €1.117 per kWh.

Last night, I wanted to start charging the Battery, but I couldn't select the cheapest price because only 10-cent increments were available. (Presumably, the number of options is limited.)

naltatis, 2024-11-06:
We'll improve the selection (see above) but i dont see manual input right now.

closed 2024-11-06 by andig