evcc-io/evcc #17094: evcc crashed when no data incoming from tibber

falkkrueger, 2024-11-05:
### Describe the bug

evcc crashed when Tibber Pulse, the device Collection data from Home Grid and Sendung to tibber, loses Connection to tibber. Experienced by myself and here: https://github.com/evcc-io/evcc/discussions/10206

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
to be delivered when needed
```

### Log details

```text
Problem: main  ] FATAL 2024/11/04 13:17:15 will attempt restart in: 15m0s
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/11/04 13:18:11 evcc 0.131.4
[main  ] INFO 2024/11/04 13:18:11 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/11/04 13:18:11 connecting evcc-947393854 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/11/04 13:18:11 tcp://localhost:1883 connected
[main  ] INFO 2024/11/04 13:18:11 listening at :7070
[main  ] FATAL 2024/11/04 13:18:22 cannot create meter 'tibber': cannot create meter type 'template': cannot create meter type 'tibber-pulse': Message: Post "https://api.tibber.com/v1-beta/gql": context deadline exceeded, Locations: [], Extensions: map[code:request_error], Path: []
[main  ] FATAL 2024/11/04 13:18:22 will attempt restart in: 15m0s
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/11/04 13:21:13 evcc 0.131.4
[main  ] INFO 2024/11/04 13:21:13 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/11/04 13:21:13 connecting evcc-985939043 at tcp://localhost:1883
[main  ] INFO 2024/11/04 13:21:13 listening at :7070
[mqtt  ] DEBUG 2024/11/04 13:21:13 tcp://localhost:1883 connected
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/11/04 13:24:15 evcc 0.131.4
[main  ] INFO 2024/11/04 13:24:15 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/11/04 13:24:15 connecting evcc-397906012 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/11/04 13:24:15 tcp://localhost:1883 connected
[main  ] INFO 2024/11/04 13:24:15 listening at :7070
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/11/04 13:27:17 evcc 0.131.4
[main  ] INFO 2024/11/04 13:27:17 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/11/04 13:27:18 connecting evcc-1991284905 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/11/04 13:27:18 tcp://localhost:1883 connected
[main  ] INFO 2024/11/04 13:27:18 listening at :7070
[main  ] FATAL 2024/11/04 13:27:30 cannot create meter 'tibber': cannot create meter type 'template': cannot create meter type 'tibber-pulse': Message: Post "https://api.tibber.com/v1-beta/gql": context deadline exceeded, Locations: [], Extensions: map[code:request_error], Path: []
[main  ] FATAL 2024/11/04 13:27:30 will attempt restart in: 15m0s
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/11/04 13:30:20 evcc 0.131.4
[main  ] INFO 2024/11/04 13:30:20 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/11/04 13:30:20 connecting evcc-1417228583 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/11/04 13:30:20 tcp://localhost:1883 connected
[main  ] INFO 2024/11/04 13:30:20 listening at :7070
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/11/04 13:33:22 evcc 0.131.4
[main  ] INFO 2024/11/04 13:33:22 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/11/04 13:33:22 connecting evcc-2028869654 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/11/04 13:33:22 tcp://localhost:1883 connected
[main  ] INFO 2024/11/04 13:33:22 listening at :7070
[pulse ] ERROR 2024/11/04 13:36:23 unexpected end of JSON input
[main  ] FATAL 2024/11/04 13:36:23 cannot create meter 'tibber': cannot create meter type 'template': cannot create meter type 'tibber-pulse': Message: unable to start stream api:98eb1cb388434ad68591fbdb3347fb71 for device 98eb1cb3-8843-4ad6-8591-fbdb3347fb71; userId 70b31f29-e689-402f-bc4b-3b797f3b0800, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
[main  ] FATAL 2024/11/04 13:36:23 will attempt restart in: 15m0s
[... 34 lines cut ...]
[mqtt  ] INFO 2024/11/04 13:45:31 connecting evcc-290531763 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/11/04 13:45:31 tcp://localhost:1883 connected
[main  ] INFO 2024/11/04 13:45:31 listening at :7070
[pulse ] ERROR 2024/11/04 13:45:31 unexpected end of JSON input
[main  ] FATAL 2024/11/04 13:45:31 cannot create meter 'tibber': cannot create meter type 'template': cannot create meter type 'tibber-pulse': Message: unable to start stream api:98eb1cb388434ad68591fbdb3347fb71 for device 98eb1cb3-8843-4ad6-8591-fbdb3347fb71; userId 70b31f29-e689-402f-bc4b-3b797f3b0800, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
[main  ] FATAL 2024/11/04 13:45:31 will attempt restart in: 15m0s
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/11/04 13:48:33 evcc 0.131.4
[main  ] INFO 2024/11/04 13:48:33 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/11/04 13:48:33 connecting evcc-897561613 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/11/04 13:48:33 tcp://localhost:1883 connected
[main  ] INFO 2024/11/04 13:48:33 listening at :7070
[pulse ] ERROR 2024/11/04 13:48:33 unexpected end of JSON input
[main  ] FATAL 2024/11/04 13:48:33 cannot create meter 'tibber': cannot create meter type 'template': cannot create meter type 'tibber-pulse': Message: unable to start stream api:98eb1cb388434ad68591fbdb3347fb71 for device 98eb1cb3-8843-4ad6-8591-fbdb3347fb71; userId 70b31f29-e689-402f-bc4b-3b797f3b0800, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
[main  ] FATAL 2024/11/04 13:48:33 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

StefanSchoof, 2024-11-05:
Same as https://github.com/evcc-io/evcc/issues/12927?

falkkrueger, 2024-11-05:
Hallo Stefan,ich denke schon!!Ich hatte gesucht, das aber nicht gefunden. Sorry!Viele GrüßeFalkVon meinem iPhone gesendet

StefanSchoof, 2024-11-05:
Die aktuelle Meter Anbindung geht über die Cloud und ohne ein Grid Meter funktioniert evcc nicht.

Solange nicht ein lokales Tibber Meter (siehe https://github.com/evcc-io/evcc/discussions/16514) gibt, gibt es wenig was evcc machen kann.

andig, 2024-11-05:
Danke @StefanSchoof für die Erklärung- so ist es (derzeit…)

closed 2024-11-05 by andig

falkkrueger, 2024-11-06:
Vielen Dank für die Erklärung, das war schonmal sehr hilfreich!aber dass das zum Absturz führt, muss doch eigentlich nicht sein. Excel stürzt ja auch nicht ab, wenn ich versuche, etwas durch 0 zu teilen. Ich kann dann ja nicht mal mehr meinen Laderoboter steuern, der macht einfach weiter….Eine Fehlermeldung im Log und bei den Verbrauchs- Daten ein „#nv“ wären für den User hilfreicher als der Absturz. Man könnte dann noch überlegen, ob man das Überschussladen abbricht, oder diese Option per Schalter dem User überlassen.Gibt es ähnliche Probleme eigentlich auch bei anderen dynamischen Stromtarifen?By the way - vielen Dank, dass es Euch gibt, ich finde evcc GROSSARTIG!Viele GrüsseFalkVon meinem iPad gesendet

andig, 2024-11-06:
Wie gesagt. Es stürzt nicht ab.