evcc-io/evcc #17102: UI: Ladeplaner-Info: Startzeit fehlt, Zielzeit-Anzeige doppelt

An-Wi, 2024-11-06:
### Describe the bug

Seit kurzem fehlt für in der Zukunkft liegende Ladeplanungen die errechnete Startzeit.
Dafür wird die Zielzeit nun doppelt angezeigt:
[image]

_Hinweis: Musste leider den Haken "I have verified that the issue is reproducible with the latest nightly build" setzen, obwohl mir das mit meinem Windows-System nicht möglich ist._

### Steps to reproduce

1. Ladung in Zukunft planen
2. EV verbinden

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Windows

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.4

andig, 2024-11-06:
> Seit kurzem fehlt für in der Zukunkft liegende Ladeplanungen die errechnete Startzeit.

Gabs die schonmal? Da es mehrere Slots/Startzeiten geben kann ist die egtl. ohne Bedeutung?

naltatis, 2024-11-06:
@andig ja, wir zeigen eigentlich Startzeit des nächsten slots an. Ich schau da rein.

Bzgl dopplung: aktuell wird unten das eingestellte Ziel und oben (bei aktiven Plan) das prognostiziere Ende dargestellt. Das kann gleich sein, muss aber nicht.

An-Wi, 2024-11-06:
> Bzgl dopplung: aktuell wird unten das eingestellte Ziel und oben (bei aktiven Plan) das prognostiziere Ende dargestellt. Das kann gleich sein, muss aber nicht.

Wenn der Plan läuft, ist das auch ok und gut so.
Solange der Plan aber noch in der Zukunft lag, war die Info, wann der Plan voraussichtlich startet, sehr hilfreich.

naltatis, 2024-11-06:
> Solange der Plan aber noch in der Zukunft lag, war die Info, wann der Plan voraussichtlich startet, sehr hilfreich.

Ja, genau so sollte es auch sein. Kannst du mal Log und die Antwort von `/api/state` hier posten? Bin mir noch nicht ganz sicher wie das Zustande kommt

An-Wi, 2024-11-07:
Habe zufällig das EV heute da, angeschlossen und es lädt gerade auch.
Wenn ich jetzt einen Plan aktiviere, dann steht wieder die Startzeit des Plans (in der Zukunft) da - so wie es sein sollte.

Jetzt mal noch das Laden gestoppt, auf "PV" gestellt. Zielplan wie vorgestern - und der "Bug" ist nicht reproduzierbar!
[image]

<pre>
[lp-1  ] DEBUG 2024/11/07 13:32:47 charge power: 0W
[lp-1  ] DEBUG 2024/11/07 13:32:47 charge currents: [0 0 0]A
[script] DEBUG 2024/11/07 13:32:47 /fupcService/pvCurrentPower.bat: 496
[site  ] DEBUG 2024/11/07 13:32:47 pv power: 496W
[site  ] DEBUG 2024/11/07 13:32:48 grid power: 212W
[site  ] DEBUG 2024/11/07 13:32:48 grid currents: [0.841 0.694 2.8]A
[site  ] DEBUG 2024/11/07 13:32:48 site power: 112W
[lp-1  ] DEBUG 2024/11/07 13:32:48 charge total import: 7516.727kWh
[lp-1  ] DEBUG 2024/11/07 13:32:48 charger status: B
[lp-1  ] DEBUG 2024/11/07 13:32:48 next soc poll remaining time: 59m20s
[lp-1  ] DEBUG 2024/11/07 13:32:48 plan: charge 3h29m13s between 2024-11-08 04:10:47 +0100 CET until 2024-11-08 07:40:00 +0100 CET (power: 8280W, avg cost: 0.296)
[lp-1  ] DEBUG 2024/11/07 13:32:48 pv charge current: 0A = 0A + -0.162A (112W @ 3p)
[db    ] TRACE 2024/11/07 13:33:02 INSERT INTO `settings` (`key`,`value`) VALUES ("savings.started","2023-10-05T10:05:17+02:00"),("savings.gridCharged","423.075212018717"),("savings.gridCost","133.8678684231701"),("savings.gridSavedCost","7.140685711261891"),("savings.selfConsumptionCharged","40.56666730025957"),("savings.selfConsumptionCost","5.787900641499003"),("vehicle.0.targetSoc","95"),("vehicle.0.targetEnergy","80"),("vehicle.0.minSoc","40"),("vehicle.0.targetTime","0001-01-01T00:00:00Z"),("vehicle.1.targetSoc","75"),("vehicle.1.targetEnergy","0"),("vehicle.1.minSoc","0"),("vehicle.1.targetTime","0001-01-01T00:00:00Z"),("telemetry","true"),("vehicle.ev1.planTime","2024-11-08T06:40:00Z"),("vehicle.ev1.planSoc","85"),("lp1.mode","pv"),("lp1.limitSoc","0"),("vehicle.ev1.minSoc","40"),("vehicle.ev1.limitSoc","85"),("lp1.planEnergy","0"),("lp1.planTime","0001-01-01T00:00:00Z"),("smartCostLimit","0"),("lp1.minCurrent","6"),("lp1.maxCurrent","12"),("lp1.phasesConfigured","3"),("lp1.smartCostLimit",""),("adminPassword","$2a$10$.CSjTf2f4sRYiiSvsY2y7Ol9QTd4F4/wyyfrynfJHOH0hjZfMZ2eG"),("jwtSecretKey","bcbadc169f75b205ee8158258cc745fa8c28ede856590146d4c73b2104b6f5e8"),("lp1.limitEnergy","0") ON CONFLICT (`key`) DO UPDATE SET `value`=`excluded`.`value` 31 <nil>
[site  ] DEBUG 2024/11/07 13:33:07 ----
[lp-1  ] DEBUG 2024/11/07 13:33:07 charge power: 0W
[lp-1  ] DEBUG 2024/11/07 13:33:07 charge currents: [0 0 0]A
[script] DEBUG 2024/11/07 13:33:07 /fupcService/pvCurrentPower.bat: 496
[site  ] DEBUG 2024/11/07 13:33:07 pv power: 496W
[site  ] DEBUG 2024/11/07 13:33:08 grid power: 103W
[site  ] DEBUG 2024/11/07 13:33:08 grid currents: [0.838 0.693 0.763]A
[site  ] DEBUG 2024/11/07 13:33:08 site power: 3W
[lp-1  ] DEBUG 2024/11/07 13:33:08 charge total import: 7516.727kWh
[lp-1  ] DEBUG 2024/11/07 13:33:08 charger status: B
[lp-1  ] DEBUG 2024/11/07 13:33:08 next soc poll remaining time: 59m0s
[lp-1  ] DEBUG 2024/11/07 13:33:08 plan: charge 3h29m13s between 2024-11-08 04:10:47 +0100 CET until 2024-11-08 07:40:00 +0100 CET (power: 8280W, avg cost: 0.296)
[lp-1  ] DEBUG 2024/11/07 13:33:08 pv charge current: 0A = 0A + -0.0042A (3W @ 3p)
[site  ] DEBUG 2024/11/07 13:33:27 ----
[lp-1  ] DEBUG 2024/11/07 13:33:27 charge power: 0W
[lp-1  ] DEBUG 2024/11/07 13:33:27 charge currents: [0 0 0]A
[script] DEBUG 2024/11/07 13:33:27 /fupcService/pvCurrentPower.bat: 505
[site  ] DEBUG 2024/11/07 13:33:27 pv power: 505W
[site  ] DEBUG 2024/11/07 13:33:28 grid power: 306W
[site  ] DEBUG 2024/11/07 13:33:28 grid currents: [0.86 0.719 4.03]A
[site  ] DEBUG 2024/11/07 13:33:28 site power: 206W
[lp-1  ] DEBUG 2024/11/07 13:33:28 charge total import: 7516.727kWh
[lp-1  ] DEBUG 2024/11/07 13:33:28 charger status: B
[lp-1  ] DEBUG 2024/11/07 13:33:28 next soc poll remaining time: 58m39s
[lp-1  ] DEBUG 2024/11/07 13:33:28 plan: charge 3h29m13s between 2024-11-08 04:10:47 +0100 CET until 2024-11-08 07:40:00 +0100 CET (power: 8280W, avg cost: 0.296)
[lp-1  ] DEBUG 2024/11/07 13:33:28 !! pvScalePhases DOWN activePhases: 3, available power: -206W, scalable: false
[lp-1  ] DEBUG 2024/11/07 13:33:28 !! pvScalePhases UP maxPhases: 3, available power: -206W, scalable: false
[lp-1  ] DEBUG 2024/11/07 13:33:28 !! pvScalePhases EXIT
[lp-1  ] DEBUG 2024/11/07 13:33:28 pv charge current: 0A = 0A + -0.299A (206W @ 3p)
[site  ] DEBUG 2024/11/07 13:33:37 set ev1 plan soc: 80 @ 2024-11-08 07:40:00 +0100 CET
[site  ] DEBUG 2024/11/07 13:33:47 ----
[lp-1  ] DEBUG 2024/11/07 13:33:47 charge power: 0W
[lp-1  ] DEBUG 2024/11/07 13:33:47 charge currents: [0 0 0]A
[script] DEBUG 2024/11/07 13:33:47 /fupcService/pvCurrentPower.bat: 510
[site  ] DEBUG 2024/11/07 13:33:47 pv power: 510W
[site  ] DEBUG 2024/11/07 13:33:48 grid power: 368W
[site  ] DEBUG 2024/11/07 13:33:48 grid currents: [0.835 0.71 5.01]A
[site  ] DEBUG 2024/11/07 13:33:48 site power: 268W
[lp-1  ] DEBUG 2024/11/07 13:33:48 charge total import: 7516.727kWh
[lp-1  ] DEBUG 2024/11/07 13:33:48 charger status: B
[lp-1  ] DEBUG 2024/11/07 13:33:48 next soc poll remaining time: 58m19s
[lp-1  ] DEBUG 2024/11/07 13:33:48 plan: charge 2h54m12s between 2024-11-08 04:45:48 +0100 CET until 2024-11-08 07:40:00 +0100 CET (power: 8280W, avg cost: 0.296)
[... cut ...]
[lp-1  ] DEBUG 2024/11/07 13:34:12 pv charge current: min 6A > 0A (45W @ 1p, battery: false)
[lp-1  ] DEBUG 2024/11/07 13:34:12 max charge current: 6A
[lp-1  ] DEBUG 2024/11/07 13:34:12 charger enable
[lp-1  ] DEBUG 2024/11/07 13:34:12 wake-up timer: start
[site  ] DEBUG 2024/11/07 13:34:16 set ev1 plan soc: 0 @ 0001-01-01 01:00:00 +0100 CET
[site  ] DEBUG 2024/11/07 13:34:27 ---- 
</pre>

Der einzige Unterschied zu vorgestern:
Vor dem Planer hatte ich noch Mindestladen auf 40% aktiviert, was auch ausgeführt wurde...

naltatis, 2024-11-08:
> C:\Windows\system32>curl http://localhost:7070/api/state

Der State ist vmtl. von nach dem Ladeplan. Zumindest sind alle ladeplanrelevanten Daten zurückgesetzt.

>  und der "Bug" ist nicht reproduzierbar!

Schade. Die Suche nach der Ursache ist idR. komplizierter, wenn mans nicht reproduzieren kann.

 Da es hier eher ein nicht-kritisches Anzeigethema ist, würd ich vorschlagen wir schließen den Issue erst einmal. Wenn du (oder jemand anders) hier ein Muster sieht machen wir den wieder auf.

closed 2024-11-10 by naltatis