evcc-io/evcc #17105: Add support for B+G e-tech WS100/DS100

Vogelbecker, 2024-11-06:
Hi,

I would like support in setting up additional electricity meters.
In Germany you can get two RS485-capable devices from B+G e-tech quite cheaply: 
1-phase: WS100 ( https://data.xn--stromzhler-v5a.eu/manuals/bg_ws100serie_de.pdf )
3-phase: DS100 ( https://data.xn--stromzhler-v5a.eu/manuals/bg_ds100serie_de.pdf )

I have both models in use (connected to an RS485-to-Ethernet converter).
The corresponding registers can be found in the manual linked above.
Where are the registers added to the template? Then I can test this myself if necessary.

andig, 2024-11-06:
Happy to take a PR, see https://github.com/evcc-io/evcc/tree/master/templates/definition/meter for examples. Could also be added to https://github.com/volkszaehler/mbmd/tree/master/meters/rs485 instead.

closed 2024-11-22 by andig

premultiply, 2025-08-09:
Fixed by https://github.com/evcc-io/evcc/commit/f64bd2c97b9342aa30bc6a5fbd0418b5aa1dd9e2

Bieral, 2025-08-10:
**Many thanks**, it perfectly  works

LED-Pixel, 2025-08-29:
Hi,

while using the B+GE-TECH DS100 meter as meter for an ID-Charger Pro I discovered another issue. The wrong scaling for the currents is used resulting in currents 100 times too large.

```
[lp-1 ] DEBUG 2025/08/29 18:20:35 charge power: 3W 
[lp-1 ] DEBUG 2025/08/29 18:20:35 charge currents: [4.6 0 0]A
[lp-1 ] DEBUG 2025/08/29 18:20:35 charge voltages: [232 231 231]V
[lp-1 ] DEBUG 2025/08/29 18:20:35 charger status: A
```

In line 43, 50 and 57 in the ``currents`` section the scaling factor should be ``0.001`` as the values are in ``mA``, similar to the ``voltages`` section.

Can you please check.