evcc-io/evcc #17115: Kein Netzaden mit SMA template "sma-hybrid" und WR SMA Sunny Tripower 8.0 Smart Energy mit BYD HVS

birkenhost, 2024-11-06:
### Describe the bug

Verwende das template "sma-hybrid". 
Bei erreichen der Preisschwelle wird nur die Batterie in Standby versetzt. Sollte sie gerade entladen werden wird sie "gesperrt". 

Achtung: wenn ich das "falsche" template "sma-sbs-15-25-modbus" verwende, dann wird die Batterie geladen wenn die Preisschwelle unterschritten wird. Leider ist dann die maximale Ladeleistung nur 2,7kW

Der Code der SMA templates macht einen sehr unterschiedlichen Eindruck vom Entwicklungsstand. Aber ihr habt ja auch geschrieben, dass das Feature aktuell experimentell ist.

Vielen Dank für eure Mühe!

### Steps to reproduce

1. Konfiguration wie angegeben
2. manuelles einstellen der zur Unterschreitung der Preisschwelle im UI
3. Batterie geht in Standby, mehr nicht (sollte sie entladen worden sein, geht sie hier auch in Standby) 

### Configuration details

```yaml
meters:
  - name: SMA_PV
    type: template
    template: sma-hybrid
    usage: pv
    
    # Modbus TCP
    modbus: tcpip
    id: 3
    host: 192.168.178.137 # Hostname
    port: 502 # Port
    watchdog: 30s # optional
    #maxacpower: 7 # Maximale AC Leistung des Hybrid-Wechselrichters (optional)
    
  - name: BYD_battery
    type: template
    template: sma-hybrid
    usage: battery
    capacity: 10
    
    # Modbus TCP
    modbus: tcpip
    id: 3
    host: 192.168.178.137 # Hostname
    port: 502 # Port
    
  - name: Shelly_grid
    type: template
    template: shelly-3em
    usage: grid
    host: 192.168.178.90
```

### Log details

```text
?
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.4

birkenhost, 2024-11-09:
Danke für den Hinweis, werde ich beim nächsten Bug-Report tun. Hier kann ich es gern noch einmal ergänzen. 

btw. Firmware ist in allen Komponenten aktuell, 
SHM 2 Firmware-Version des Hauptprozessors | 2.15.6.R | 25.09.2024 23:26:54

Werde die Traces für die beiden Templates noch einmal erstellen.

Tombra1889, 2024-11-10:
Habe das selbe Problem. Ein Netzladen via evcc mit dem sma-hybrid-Template funktioniert nicht.

andig, 2024-11-10:
"ich auch" nutzt nichts. Lösung siehe https://github.com/evcc-io/evcc/issues/17115#issuecomment-2463156187.

Tombra1889, 2024-11-10:
Sowohl auf dem Wechselrichter als auch auf dem SHM befindet sich die aktuelle Firmware. Leider erscheint keinerlei Eintrag im Log (gefiltert auf SMA). Ich hatte den Trace-Modus an und habe dann lediglich Einträge von SHM und einem anderen SMA Wechselrichter gesehen. Vom Hybrid-Wechselrichter gibt es keinen Eintrag.

birkenhost, 2024-11-17:
ich habe mit dem template "sma-hybrid" noch einmal ein Log hinzugefügt, hoffe das hilft. Kann gern weitere Daten liefern

premultiply, 2024-11-17:
> [site  ] DEBUG 2024/11/17 12:26:32 set battery mode: normal

Tut was es soll. Nichts Gegenteiliges erkennbar.

birkenhost, 2024-11-17:
Ich habe hier das unterschreiten und überschreiten der Preisschwelle manuell eingestellt. Wenn der aktuelle Preis unter der Schwelle ist,  kommt "set battery mode: charge". Es findet jedoch kein Netzladen statt. UI sieht folgender maßen aus (UI Screenshot nicht Zeitgleich mit Log erstellt, erst nachträglich zur Veranschaulichung des Verhaltens) :
[image]
Die Batterie wird dann weder geladen noch entladen. Wenn sie in "set battery mode: normal" geht, wird sie wieder entladen. (Untergrenze bei 10% eingestellt)

birkenhost, 2024-11-21:
würdet ihr den Bug Report wieder öffnen oder müsste ich ihn neu einstellen?

andig, 2024-11-21:
> Achtung: wenn ich das "falsche" template "sma-sbs-15-25-modbus" verwende, dann wird die Batterie geladen wenn die Preisschwelle unterschritten wird. Leider ist dann die maximale Ladeleistung nur 2,7kW

@premultiply haben wir hier einfach ein Problem mit dem Scoping der Templates? Ich mache wieder auf.

> Achtung: wenn ich das "falsche" template "sma-sbs-15-25-modbus" verwende, dann wird die Batterie geladen wenn die Preisschwelle unterschritten wird. Leider ist dann die maximale Ladeleistung nur 2,7kW

Ich vermute das sind die 2,5kW aus dem Template.

@premultiply parametrieren?

PR #17393 (2024-11-25): SMA: fix battery control
Fixes #17115
Fixes #17374
Fixes #17201
closed 2024-11-25 by andig