evcc-io/evcc #17118: Polestar API broken?

flo9127, 2024-11-06:
### Describe the bug

Keine Verbindung zur Polestar-API mehr möglich.

" ERROR 2024/11/06 20:37:14 vehicle soc: Message: Post "https://pc-api.polestar.com/eu-north-1/mystar-v2": resumePath not found, Locations: [], Extensions: map[code:request_error], Path: []"

### Steps to reproduce

1. Polestar 2 an Wallbox anklemmen
2. Fehlermeldung bewundern

### Configuration details

```yaml
#vehicles:
#  type: template
#  template: polestar
#  name: ps2b
#  title: Polestar 2 - Beast
#  user: xxx@xxx.de # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
#  password: yyyyyy # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
#  vin: yyyyyyyyyyyyy # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
#  capacity: 61 # Akkukapazität in kWh (optional)
```

### Log details

```text
[lp-1 ] DEBUG 2024/11/06 20:37:14 pv charge current: min 6A > 0A (5664W @ 3p, battery: false)
[site ] DEBUG 2024/11/06 20:37:33 ----
[lp-1 ] DEBUG 2024/11/06 20:37:33 charge power: 4086W
[lp-1 ] DEBUG 2024/11/06 20:37:33 charge currents: [6.03 5.97 5.9]A
[site ] DEBUG 2024/11/06 20:37:33 pv power: 0W
[db:3 ] TRACE 2024/11/06 20:37:33 modbus: send 6f b7 00 00 00 06 64 04 03 4a 00 01
[db:3 ] TRACE 2024/11/06 20:37:33 modbus: recv 6f b7 00 00 00 05 64 04 02 00 00
[db:3 ] TRACE 2024/11/06 20:37:33 modbus: send 6f b8 00 00 00 06 64 04 03 4b 00 01
[db:3 ] TRACE 2024/11/06 20:37:33 modbus: recv 6f b8 00 00 00 05 64 04 02 00 16
[site ] DEBUG 2024/11/06 20:37:33 battery soc: 22%
[site ] DEBUG 2024/11/06 20:37:33 battery power: 0W
[site ] DEBUG 2024/11/06 20:37:33 grid power: 5421W
[site ] DEBUG 2024/11/06 20:37:33 grid powers: [1826 1929 1666]W
[site ] DEBUG 2024/11/06 20:37:33 grid currents: [8.43 8.52 7.36]A
[site ] DEBUG 2024/11/06 20:37:33 site power: 5521W
[lp-1 ] DEBUG 2024/11/06 20:37:34 charge voltages: [231 230 231]V
[lp-1 ] DEBUG 2024/11/06 20:37:34 detected connected phases: 3p
[lp-1 ] DEBUG 2024/11/06 20:37:34 detected active phases: 3p
[lp-1 ] DEBUG 2024/11/06 20:37:34 charge total import: 917.200kWh
[lp-1 ] DEBUG 2024/11/06 20:37:34 charger status: C
[lp-1 ] ERROR 2024/11/06 20:37:34 vehicle soc: Message: Post "https://pc-api.polestar.com/eu-north-1/mystar-v2": resumePath not found, Locations: [], Extensions: map[code:request_error], Path: []
[lp-1 ] DEBUG 2024/11/06 20:37:34 pv charge current: min 6A > 0A (5521W @ 3p, battery: false)
```

### What type of operating system are you running?

Windows

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.4

demichve, 2024-11-07:
Seems to be a change in Polestar API,
see
https://github.com/pypolestar/polestar_api/issues/197

Demophobie, 2024-11-07:
> Seems to be a change in Polestar API, see [pypolestar/polestar_api#197](https://github.com/pypolestar/polestar_api/issues/197)

Scheint ja nur die Client ID geändert worden zu sein? Musste beim Einloggen online nun auch irgendwelche Sachen akzeptieren.

jahoooo, 2024-11-08:
Das ist hier schon gelöst worden:
https://github.com/pypolestar/polestar_api/pull/198
Teil von v1.8.0 des polestar_api und funktioniert bei mir.

Kann das jemand in evcc übernehmen?

andig, 2024-11-09:
Fixed in https://github.com/evcc-io/evcc/commit/355c0e3ffeb92f130291c19e7b2ab18aa3f45b4f

closed 2024-11-09 by andig

rostbeule, 2024-11-10:
No, this does not fix the bug. According to the commit, only the client ID was changed.

The OAuth2 login process is still failing because the code parameter is not returned after the first request. This occurs because additional user interactions are required (acceptance of terms and conditions). To resolve this issue, an additional check needs to be introduced: if the code parameter is missing but a uid parameter is present, a POST request should be sent to submit the user's consent. After this confirmation, the code parameter can be successfully extracted.

This was also fixed in the pypolestar project, and I solved it similarly in my C# project. It works, but it was not done here.

Try something like this:

```go
// Check if the authorization code is missing
if code == "" {
    // Attempt to extract the "uid" parameter
    v.Client.CheckRedirect, param = request.InterceptRedirect("uid", true)
    defer func() { v.Client.CheckRedirect = nil }()

    if _, err = v.Get(uri); err != nil {
        return nil, err
    }

    var uid string
    if uid, err = param(); err == nil && uid != "" {
        util.Log.Debugf("Code missing; sending confirmation for uid=%s and retrying", uid)

        // Send a POST request with user consent
        data := url.Values{
            "pf.submit": []string{"true"},
            "subject":   []string{uid},
        }

        if _, err = v.Post(fmt.Sprintf("%s/as/%s/resume/as/authorization.ping", OAuthURI, resume), request.FormContent, strings.NewReader(data.Encode())); err != nil {
            return nil, fmt.Errorf("Error sending confirmation: %w", err)
        }

        v.Client.CheckRedirect, param = request.InterceptRedirect("code", true)
        defer func() { v.Client.CheckRedirect = nil }()

        if _, err = v.Get(uri); err != nil {
            return nil, fmt.Errorf("Error retrieving code after confirmation: %w", err)
        }

        code, err = param()
        if code == "" || err != nil {
            return nil, fmt.Errorf("Could not extract authorization code after confirmation: %w", err)
        }
    }
}

if code == "" {
    return nil, fmt.Errorf("Authorization code not received")
}

```

loebse, 2024-11-11:
@andig, I can confirm that the issue is not fixed and the user consent is required in addition as mentioned by @rostbeule.

Do you want us to open a new issue or continue here?

andig, 2024-11-11:
Imho that's fine. It should be up to the user to accept new ToS.

loebse, 2024-11-11:
Unfortunately it is not 😞 

Even if you accept the T&Cs on https://www.polestar.com/de/login/profile/, you still get the error.

I agree @andig that you can argue that the integration should not auto accept it. However, this seems to be the only option at the moment. 

The current version of Polestar in EVCC does not work independent from any manual accepting of T&Cs that you can do.

MASHpony, 2024-11-17:
Any news 😅
Ich hab die neuste Version aber immer noch keine soc

loebse, 2024-11-17:
Welche Version hast du genau? Was steht denn im Log?

MASHpony, 2024-11-17:
> Welche Version hast du genau? Was steht denn im Log?

Ah das ärgert mich jetzt 
Hatte gestern ein Update gemacht auf die letzte .5 wahrscheinlich jetzt hat sich das genau überschnitten mit der .6 
Geht jedenfalls wieder 

Danke :)

MASHpony, 2024-11-20:
Ich habe eben wieder Updates installiert. Nun kommt folgender Fehler:

`[main ] ERROR 2024/11/20 22:48:33 creating vehicle ev2 failed: cannot create vehicle type 'template': cannot create vehicle type 'polestar': cannot get vehicles: Message: 404 Not Found, Locations: [], Extensions: map[code:request_error], Path: []`

rostbeule, 2024-11-20:
Kann ich nicht bestätigen. Läuft erwartungsgemäß.

Demophobie, 2024-11-21:
Also ich habe das gleiche Problem nach einem Neustart von EVCC. Vorher lief es noch:

```
[main  ] ERROR 2024/11/21 08:50:55 creating vehicle polestar failed: cannot create vehicle type 'template': cannot create vehicle type 'polestar': cannot get vehicles: Message: 404 Not Found, Locations: [], Extensions: map[code:request_error], Path: []
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'polestar': cannot get vehicles: Message: 404 Not Found, Locations: [], Extensions: map[code:request_error], Path: []
Capacity: 75.0kWh
Features: [Offline Retryable]
```

MASHpony, 2024-11-21:
Bei meinem Mini SE geht es auch nicht mehr 
Ich denke es ist kein Polestar Problem

andig, 2024-11-21:
Ist es- passiert ja beim Polestar initialisieren. Das BMW nicht mehr geht ist bekannt.

nickels, 2024-12-09:
I can confirm this is still an issue.

`creating vehicle polestar failed: cannot create vehicle type 'template': cannot create vehicle type 'polestar': login failed: resumePath not found`

Perhaps a workaround could be to login via the home assistant add-in to accept the T&Cs? I've just changed user account - perhaps I need to wait some time anyway..

wlanrouter, 2024-12-20:
Same here. v0.131.11 
Is it working somewhere?

flo9127, 2024-12-20:
> Same here. v0.131.11 Is it working somewhere?

Yes, here with this version.
0.131.12 not tested yet.

rostbeule, 2024-12-20:
Still on v0.131.10 and it’s working like a charm. Didn‘t update to v0.131.11 nor v0.131.12 so far (for reasons)

jahoooo, 2024-12-20:
I'm also on v0.131.10 and no problems, so I will stay there.
But I see that in .11 @andig did some polestar code simplifications. No Polestar changes mentioned for .12, so I would expect, the bug still exists in there.

nickels, 2024-12-20:
I'm running  v0.131.11 and it's working fine. There was an issue with my polestart account!

jahoooo, 2024-12-20:
ok, then I will update to v.0.131.12 and test if it works later on when I can connect my car... probably around lunch.

jahoooo, 2024-12-20:
Everything working well for me with v.0.131.12!