evcc-io/evcc #17120: PV mode charging - bufferStartSoc not considered anymore after upgrade to 0.131.4

tilibibi, 2024-11-06:
### Describe the bug

Since I upgraded to 0.131.4 (from 0.130.x - not sure) my setting in the battery UI does not work anymore properly. Before it was working perfectly fine as intended. 

Plugging my car(s), charging is started despite my battery SOC is below 'bufferStartSoc' defined in the UI. Also 'bufferSoc' is not considered anymore, charging continues, despite battery SOC is getting below defined value. 
After plugging, charging is starting and a 'pv disable' count down of 3 mins is starting (this I never observed before). After the 3 mins, sometimes the charging is stopping, sometimes not. 
I do not see any entry in the DEBUG log regarding 'bufferStartSoc' or 'bufferSoc' ... should I?

### Steps to reproduce

1. several trials with two different cars
2. not yet downgraded to previous version

### Configuration details

```yaml
interval: 10s # control cycle interval (default 10s)

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.178.68  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.178.68  
  port: 8080  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.178.68  
  port: 8080
  token: xxxxxx  
  name: battery3

chargers:
- type: template
  template: heidelberg 
  id: 1  
  device: /dev/ttyUSB0  
  baudrate: 19200  
  comset: 8E1  
  modbus: rs485serial  
  name: wallbox4
- type: template
  template: heidelberg 
  id: 2  
  device: /dev/ttyUSB0  
  baudrate: 19200  
  comset: 8E1  
  modbus: rs485serial  
  name: wallbox5

loadpoints:
- title: Garage door
  charger: wallbox4
  mode: pv
- title: Garage back
  charger: wallbox5
  mode: pv

site:
  title: My EVCC Home
  meters:
    grid: grid1
    pv:
[... 472 lines cut ...]
[lp-1  ] DEBUG 2024/11/06 18:37:32 charge total import: 622.527kWh
[lp-1  ] DEBUG 2024/11/06 18:37:32 charger status: B
[lp-1  ] DEBUG 2024/11/06 18:37:32 charge final energy: 0.199kWh
[lp-1  ] INFO 2024/11/06 18:37:32 stop charging <-
[lp-1  ] DEBUG 2024/11/06 18:37:32 pv timer reset
[lp-1  ] DEBUG 2024/11/06 18:37:32 pv timer inactive
[lp-1  ] DEBUG 2024/11/06 18:37:33 charge total import: 622.527kWh
[lp-1  ] DEBUG 2024/11/06 18:37:33 soc estimated: 52.84% (vehicle: 52.00%)
[lp-1  ] DEBUG 2024/11/06 18:37:33 vehicle soc: 53%
[lp-1  ] DEBUG 2024/11/06 18:37:33 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/11/06 18:37:33 vehicle range: 29km
[lp-1  ] DEBUG 2024/11/06 18:37:33 pv charge current: 0A = 0A + -0.594A (410W @ 3p)
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.4

mfuchs1984, 2024-11-06:
Although you have `welcomecharge: false` in the car configs, it seems to start a welcome charge
`[lp-1  ] DEBUG 2024/11/06 18:33:53 welcome charge: BMW_X1`

Already tried removing the property completely from the config?

tilibibi, 2024-11-06:
Not yet, but I can surely give it a trial tomorrow. However I'm also quite sure, I didn't touch the config with which it was working before.

mfuchs1984, 2024-11-07:
Think this should help. Afaik, Evcc only checks for existence of welcomecharge in the vehicle config, not for a Boolean value

`{{- if .welcomecharge }}
  features: ["welcomecharge"]`

closed 2024-11-07 by andig