evcc-io/evcc #17127: Grid-Charging: Re-Set active battery setting instead of static setting

maatinh, 2024-11-07:
Right now the grid-charging function is statically setting the battery-mode to "Auto" (for my goodwe inverter) when the grid-charging is finished. In case the active setting was different then "Auto" when grid-charging was started this leads to increase grid costs.

Example: Grid-Price is "low", I use "Conserve" Mode to put all PV-Power into battery and use Grid-Power for house consumption. evcc is configured to charge battery when Grid-Price is "very low". When done, evcc will set battery to "Auto" which causes battery to discharge even grid-price is still "very low" or relatively "low". That could cause that I do not have sufficient batter capacity for the later hours when grid-price is high/very-high. This weeks grid-costs are a perfect example.

Would be nice if evcc could check battery-mode and reset it back when grid-charging is done.

andig, 2024-11-07:
It will set the „right“ mode in the next cycle, i.e. After 30s.

maatinh, 2024-11-07:
This is not what I observed. Goodwe-hybrid.yaml only knows two settings for battery: 0 for normal operations and 2 for discharge-hold or charge.

I had manually set a different mode (6) from external. Then evcc started grid-charging and then set the value to 0 and not back to the original value of 6 as it was when evcc started grid-charging. Not after 30s and not after 10mins when I noticed that.

andig, 2024-11-07:
It does bot know the original mode. Either template needs be updated to chose a default mode (i.e. normal) or you can‘t use evcc battery control for this purpose.

andig, 2024-11-07:
Da die Confif fehlt geschlossen.

closed 2024-11-07 by andig

maatinh, 2024-11-07:
Just wanted to address it… I have a solution, i dont care.

andig, 2024-11-07:
You‘re welcome to use the issue template. It‘s there on purpose 👍🏻