evcc-io/evcc #17136: Solar mode using grid energy

rlcintra, 2024-11-08:
### Describe the bug

I am testing evcc and during my first charge using the Solar mode, I noticed it was using not only my solar excess but also consuming energy from the grid. At the time my panels were generating ~9.8Kw and evcc was charging the vehicle with the maximum capacity of my my charger, 11.6kW (see screenshot below).

[image]

Other times it was ignoring the energy consumed in my house and still using more energy than produced by my solar:

[image]

At the end of the charging, roughly 10% of the energy used was from the grid instead of the my solar, despite the whole time evcc was running on "Solar" mode.

[image]

I am not sure if this is a configuration problem, if this is expected, or if this could be a bug.

### Steps to reproduce

1. Charge car using Solar mode and check if grid energy was also used during charging.

### Configuration details

```yaml
# trial token, valid until 2024-11-14
sponsortoken: ...

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ...

interval: 30s # control cycle interval

chargers:
- type: template
  template: twc3
  host: 192.168.1.136
  name: wallbox1

loadpoints:
- title: Garage
  charger: wallbox1
  mode: pv

site:
  title: My home
  meters:
    grid: ha.grid
    pv:
      - ha.pv

meters:
  - name: ha.grid
    type: custom
    power:
      source: http
      uri: http://192.168.1.5:8123/api/states/sensor.sa_import_export_power
      method: GET
      headers:
        - Authorization: Bearer ...
      insecure: true
      jq: .state|tonumber
      timeout: 2s
  - name: ha.pv
    type: custom
    power:
      source: http
      uri: http://192.168.1.5:8123/api/states/sensor.sa_generation_power
      method: GET
      headers:
        - Authorization: Bearer ...
      insecure: true
      jq: .state|tonumber
      timeout: 2s

vehicles:
  - name: mycar
    type: template
    template: tesla
#    title: # Will be displayed in the user interface (optional)
    accessToken: ...
    refreshToken: ...
#vin: W... # Required if you own multiple vehicles of the same brand (optional)
```

### Log details

```text
Sorry, I didn't saved the logs at the time of the charging.
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.4

VolkerK62, 2024-11-08:
see documenation for getting a log file
https://docs.evcc.io/en/docs/faq#how-do-i-create-a-log-file-for-error-analysis

razfaz23, 2024-11-08:
I have the same issue with my evcc setup. I am running on  latest Home Assistant OS, and evcc addon. PV, Grid, Battery are setup and providing correct readings. The Tesla Wall Connector 3 is implemented and connected. I have a Tesla Model 3, for which the virtual key is set up, the API keys are generated and added to evvc.yaml. 

I have checked the configuration several times and re-setup each element from scratch, including revoking virtual key and API key.  EVCC receives all PV, grid, battery, and car telemetries, but I cannot mapulate the loadpoints i.e. charge plans and max current/charge rates are ignored, and loadpoint charge states do not have any effect to the actual charging.

Cheers.

**EVCC:** 
Current version: 0.131.4 

**Hass.IO/Home Assistant:**
Core: 2024.11.0
Supervisor: 2024.11.2
Operating System: 13.2
Frontend: 20241106.0

**EVCC Config:**
`network:
  schema: http
  port: 7070
interval: 30s # control cycle interval

log: debug
levels:
  cache: error

plant: <redacted>
site:
  title: Zuhause
  meters:
    grid: grid
    pv: pv
    battery: battery

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.375 
  feedin:
    type: fixed
    price: 0.08 
  co2:
    type: grünstromindex
    zip: <redacted>

mqtt:
  broker: 192.168.1.20:1883
  topic: evcc
  clientid: evcc
  user: evcc
  password: '<redacted>'

# meter definitions
meters:
- name: grid # Net Grid Power (import - export)
  type: custom
  power: # Current Load (W) 
    source: mqtt
    topic: homeassistant/energy/grott/pv/netgridpower
    timeout: 310s
    scale: 0.1 
  energy: # Energy Total (kWh)
    source: mqtt
    topic: homeassistant/energy/grott/pv/elocalload_tot
    timeout: 310s
    scale: 0.1  
  currents:
    - source: mqtt
      topic: homeassistant/energy/grott/pv/pvgridcurrent
      timeout: 310s
    - source: mqtt
      topic: homeassistant/energy/grott/pv/pvgridcurrent2
      timeout: 310s
    - source: mqtt
      topic: homeassistant/energy/grott/pv/pvgridcurrent3
      timeout: 310s

- name: pv
  type: custom
  power: 
    source: mqtt
    topic: homeassistant/energy/grott/pv/pvpowerin
    timeout: 310s
    scale: 0.1  # Convert from payload value to W
  energy: 
    source: mqtt
    topic: homeassistant/energy/grott/pv/epvtotal
    timeout: 310s
    scale: 0.1  # Convert from payload value to kWh

- name: battery #Net Batttery Power (discharge - charge)
  type: custom
  power:
    source: mqtt
    topic: homeassistant/energy/grott/pv/netbatpower
    timeout: 310s  # Optional
    scale: 0.1  # Convert from payload value to kWh
  energy:
    source: mqtt
    topic: homeassistant/energy/grott/pv/eharge1_tod
    timeout: 310s  # Optional
    scale: 0.1  # Convert from payload value to kWh
  capacity: 10.2 # Capacity (kWh)
  soc: #Battery SOC (%)
    source: mqtt
    topic: homeassistant/energy/grott/pv/soc
    timeout: 310s  # Optional

#- name: charge-meter
  #type: charger
  #power: 
    #source: mqtt
    #topic: homeassistant/energy/grott/pv/p1charge1e
    #timeout: 30s  # Optional
  #energy: 
    #source: mqtt
    #topic: homeassistant/energy/grott/pv/eharge1_tod
    #timeout: 30s  # Optional
    #scale: 1  # Convert from payload value to kWh

chargers:
- type: template
  template: twc3 
  host: 192.168.1.251  
  name: charger_garage

- type: template
  template: twc3 
  host: 192.168.1.252  
  name: charger_carport

loadpoints:
  - title: Garage
    charger: charger_garage
    mode: pv
    vehicle: vehicle_1

  - title: Carport
    charger: charger_carport
    mode: off
    #meter: charge-meter
    vehicle: vehicle_2

vehicles:
  - name: vehicle_1
    icon: car
    type: template
    template: tesla
    title: Tesla Model 3
    accessToken:<redacted>
    refreshToken: <redacted>
    vin: <redacted>
    capacity: 73.5 

  - name: vehicle_2
    icon: car
    type: template
    template: offline
    title: Smart ForTwo EQ

  - name: vehicle_3
    type: template
    template: offline
    title: Cube Kathmandu Hybrid
    icon: bike
    capacity: 0.5

  - name: vehicle_4
    type: template
    template: offline
    title: Cube Reaction Hybrid
    icon: bike
    capacity: 0.75`

andig, 2024-11-08:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-11-08 by andig

razfaz23, 2024-11-08:
Hi Andi,

thanks for your quick response. Since I  can export logs via the EVCC ui, which of these would ou require, debug or trace? If it is trace logs, I can redact sensitive information before sharing.

Thanks in advance.

andig, 2024-11-08:
Debug is sufficient. It _must_ show the issue, please check the pv mode calculations in the log for the relevant time stamps.

razfaz23, 2024-11-08:
I guess I have found the "issue"... Since I was still in the setup and testing phase of the evcc installation, I had not set up a sponsorship now. Adding this fixed the "issue". While I am waiting for my own sponsorship token, I have found the trial token in the documentation. 

Thanks for the effort though, and this awesome project. ;)

RTTTC, 2024-11-08:
So there was an error message in the log.

Could we show such messages in UI also?
This would reduce support requests for sure.

razfaz23, 2024-11-08:
I found an error indicating the missing sponsorship token in the trace logs, but as andig said, it is in the trace logs as well. The error message was produced when I switched the charging state to "off" while the car was being charged. After adding the trail token, it is now working as intended. I will replace the trail token as soon as I have my own.

You can find the trail token here: https://docs.evcc.io/en/docs/sponsorship

Log excerpt from my debug logs:
```
[site ] DEBUG 2024/11/08 15:22:41 pv power: 300W
[site ] DEBUG 2024/11/08 15:22:41 battery soc: 33%
[site ] DEBUG 2024/11/08 15:22:41 battery power: -18W
[site ] DEBUG 2024/11/08 15:22:41 grid power: 0W
[site ] DEBUG 2024/11/08 15:22:41 grid currents: [7 7 7]A
[site ] DEBUG 2024/11/08 15:22:41 battery has priority at soc 33% (< 75%)
[site ] DEBUG 2024/11/08 15:22:41 site power: 100W
[lp-1 ] DEBUG 2024/11/08 15:22:41 charge voltages: [231 232 232]V
[lp-1 ] DEBUG 2024/11/08 15:22:41 detected connected phases: 3p
[lp-1 ] DEBUG 2024/11/08 15:22:41 detected active phases: 3p
[lp-1 ] DEBUG 2024/11/08 15:22:41 charger status: C
[lp-1 ] DEBUG 2024/11/08 15:22:41 soc estimated: 39.27% (vehicle: 39.00%)
[lp-1 ] DEBUG 2024/11/08 15:22:41 vehicle soc: 39%
[lp-1 ] DEBUG 2024/11/08 15:22:41 vehicle soc limit: 60%
[lp-1 ] DEBUG 2024/11/08 15:22:41 vehicle range: 180km
[lp-1 ] ERROR 2024/11/08 15:22:41 charger disable: sponsorship required, see https://github.com/evcc-io/evcc#sponsorship
[site ] DEBUG 2024/11/08 15:23:07 ----
[lp-1 ] DEBUG 2024/11/08 15:23:07 charge power: 11218W
[lp-1 ] DEBUG 2024/11/08 15:23:07 charge currents: [16.1 16.2 16.1]A
```

[image]

andig, 2024-11-08:
> charger disable: sponsorship required, see https://github.com/evcc-io/evcc#sponsorship

This looks a bit confusing but comes from the fact, that the Tesla cloud proxy (for the vehicle!) requires sponsorship.