evcc-io/evcc #17151: use average values instead of current values for threasholds

tipuraneo, 2024-11-09:
Evcc compares current values of meters against thresholds in combination with fixed delays to start or stop charging session.

This has two side effects:
1. The algorithm does not honer the power drawn from or delivered to the net. If a massive load (e. g. heat pump) leads to massive net draw, total net consumption is relatively high. Same occours vice verca if sun suddenly comes up behind clouds. 
2. If the current value swings around the threshold, the delay timer restarts every time the current value exceeds the threshold or falls below it. This leads to unnecassary delays in enabling or disabling charging sessions.

Instead of using the current value I suggest to use an average value of a sliding time window.

This maximalizes the use of pv power, minimizes net draw and keeps the amount of unwanted power switches low.

andig, 2024-11-09:
This is not super simple as the averaging needs to be understandable to the user. Not a priority but happy to take a PR.

tipuraneo, 2024-11-09:
Maybe it is possible to provide the two modes via config: (legacy) delay mode and average mode. While in delay mode remaining time is shown in the ui as usual, in average mode difference between average and threshold is shown. Negative values mean below, positive above threshold. For introducing average mode two more variables would be necassary:
- hysteresis-mode (delay|average)
- sliding time window

That should be easy to understand.

It would be also possible to make a trend computation based on the average value and use it to calculate the remaining time till the threshold will probably be hit. But in certain cases this will lead to very long (invinite) times and to times which do not in-/decrease linear (like a copy progressbar).

closed 2024-11-21 by andig