evcc-io/evcc #17168: Ellipro with EM24 as an external meter: With 3-phase charging, only 1/3 of the kWh is displayed. With 1-phase charging, nothing at all.

OJack0815, 2024-11-10:
### Describe the bug

I have retrofitted the Ellipro Gen1 with an external CG-EM24 ethernet meter. I have configured the EM24 as a meter in evcc.yaml.
Unfortunately, EVCC only displays 1/3 of the kWh charged during a 3-phase charging process. Both on the dashboard and in the charging history. During a 1-phase charging process, no kWh are displayed on the dashboard and the charging process does not appear in the history at all.
The correct current charging power in kW is always displayed on the dashboard. The control for PV surplus also works very well.
What is the error that only the kWh from phase L2 or L3 are apparently integrated?

I had previously integrated the Ellipro without an external meter. The control was a bit slow at times.
But the recording of the charging energy worked correctly both on the dashboard and in the history.

### Steps to reproduce

1. Install Ellipro and EM24
2. Konfigure evcc.yaml
3. start charging and compare kWh at evcc with kWh at victron GX.

### Configuration details

```yaml
meters:
- type: template
  template: cg-em24
  usage: charge
  modbus: tcpip
  id: 1
  host: 192.168.172.205
  port: 502
  name: elli_pro1

chargers: 
- type: template
  template: ellipro
  ski: .....
  ip: 192.168.172.206
  name: wallbox4

loadpoints:
- title: Garage
  charger: wallbox4
  meter: elli_pro1
  mode: pv
```

### Log details

```text
Unfortunately I'm not able to make a log because I'm not very good at linux.
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.131.4

andig, 2024-11-10:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-11-10 by andig

OJack0815, 2024-11-10:
Thanks for your answer.
Now I found the function to make a logfile via webui. Can this help to make a diagnostic?

OJack0815, 2024-11-10:
@andig 
I updated to latest v0.131.5 but issue is still there.
I logged out the EM24 from the victron cerbo gx but no changes.
Only when I commented out it at evcc.yaml (only at loadpoints the line #meter: elli_pro1) the counting of the kWh gets correct again.
At the victron cerbo gx and in the victron vrm portal the kWh are counted correct in any cases.

OJack0815, 2024-11-10:
All charging today was done with 11kW. But only the one with commented out EM24 is displayed with average power of about 11kW.
[image]

andig, 2024-11-10:
Is your meter even 3p connected?

OJack0815, 2024-11-10:
The meter is connected 3p+n all time.
If I want to charge 1p I disconnect L2 and L3 between the meter and the wallbox and change the setting in the charger menu.
[image]

andig, 2024-11-10:
Try run

    evcc meter

while charging. Is that value correct?

andig, 2024-11-10:
Pls don‘t post screenshots.

OJack0815, 2024-11-10:
Ok, just a last one because it shows a readout from CarloGavazzi Configuration Software.
There you can see that it looks like evcc reads the energy from L2 or L3. It's 44kWh as read at "evcc meter".
[image]

andig, 2024-11-13:
Should be fixed in tomorrow's nightly.

OJack0815, 2024-11-14:
Just updated but seems not that good. 
Dashboard sasy 243kW while not charging.
Called "evcc meter" while not charging gives me:

elli_pro1
---------
Power:          235930W
Energy:         471859.2kWh
Current L1..L3: 2.82e+03A 0A 0A
Voltage L1..L3: 1.49e+07V 1.52e+07V 1.53e+07V
Power L1..L3:   255590W 0W 0W

OJack0815, 2024-11-14:
[evcc-20241114-164128-debug.log](https://github.com/user-attachments/files/17751188/evcc-20241114-164128-debug.log)

andig, 2024-11-16:
@premultiply ich habs reverted. Fehler sollte (auch) in mbmd gefixed werden.

PR #17173 (2024-11-13): EM24: add pv usage & fix energy
Fixes #17088

Also may fix #17168