evcc-io/evcc #17174: Grid cost not loaded using PUN template

MobilettoSoft, 2024-11-10:
### Describe the bug

Using the PUN template causes the cost of grid electricity is not showed; the issue is not affected by using or not using charges, tax or formula .

(the site https://www.mercatoelettrico.org/it/ has been recently updated in its layout: I dont' know if this could be the root cause of this issue) 

### Steps to reproduce

1. Use PUN template in evcc.yaml

### Configuration details

```yaml
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: template
    template: pun
    charges: 0.130267926
    tax: 0.1
    formula: math.Max((1.1 * price + charges) * (1 + tax), 0.0)
```

### Log details

```text
The logs in evcc UI don't show anything related to grid tariff

In the homeassistant add-on logs, when using the PUN tariff this message is reported shortly after add-on startup:

Error unmarshalling XML:  expected element type <NewDataSet> but have <html>

When using other tariff templates, the message disappears
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.5

andig, 2024-11-10:
Please show

    evcc tariff —log trace,db:error

MobilettoSoft, 2024-11-10:
grid:
Error unmarshalling XML:  expected element type <NewDataSet> but have <html>
Error unmarshalling XML:  expected element type <NewDataSet> but have <html>
From    To    Price/Cost

andig, 2024-11-10:
Missing requested log.

MobilettoSoft, 2024-11-10:
Well, nice attitude to someone hinting you a possible bug and even the possible reason (the way you're pulling data from www.mercatoelettrico.org doesn't work anymore because they completely overhauled their website).

However, the CLI command you suggested (`evcc tariff -log trace,db:error`) didn't work, so I used this:  

`evcc tariff -l trace,db:error`

And I got this:

```
[main  ] INFO 2024/11/10 23:00:53 evcc 0.131.5
[main  ] INFO 2024/11/10 23:00:53 using config file: C:\Users\abern\Downloads\evcc\evcc.yaml
grid:
Error unmarshalling XML:  expected element type <NewDataSet> but have <html>
Error unmarshalling XML:  expected element type <NewDataSet> but have <html>
From    To    Price/Cost
```

I didn't think the first two lines were essential, so excuse me.

I hope this can be useful for other users possibly encountering this issue. I reverted to the method I was already using (importing the electricity cost via REST API from homeassistant and its nice PUN integration).

andig, 2024-11-11:
I see. Missing log is a bug in the tariff. Will fix that and then we can double check.

andig, 2024-11-12:
@motze92 do you want to take a look? I've added the logging but test is broken since server returns a different response.

MobilettoSoft, 2024-11-14:
I've got no experience developing in go, so I can't help with code, but maybe I can help with some info. Take a look at these discussions from the [Homeassistant plugin repo](https://github.com/virtualdj/pun_sensor).

Starting from october 1 until december 12, there is a temporary workaround to access the old mercatoelettrico.org site:
https://github.com/virtualdj/pun_sensor/issues/59

The new site uses a different API as explained here (look at the last post):
https://github.com/virtualdj/pun_sensor/issues/42

MobilettoSoft, 2024-11-14:
It seems that only changing the endpoint from www.mercatoelettrico.org to storico.mercatoelettrico.org temporarily fixes the problem. If you want I can make some tests and then open a PR.

At least we have a solution until december 31.

andig, 2024-11-16:
Fixed for now thanks to @MobilettoSoft but will stop working end of year. /cc @motze92

PR #17900 (2024-12-30): Updated PUN tariff to use the new GME website
I hope I managed to fix the PUN tariff retrieval using the new GME site thanks to the great work of @virtualdj with https://github.com/virtualdj/pun_sensor 
This should (I hope) definitely close #17174 

Added code to better log the various steps needed to retrieve data.
closed 2024-11-16 by andig