evcc-io/evcc #17184: Solar mode using grid energy

rlcintra, 2024-11-11:
### Describe the bug

This is basically the same bug reported at the closed issue [#17136 · evcc-io/evcc](https://github.com/evcc-io/evcc/issues/17136), but this time I included the logs as requested.

I made two charges with evcc using the Solar mode, I noticed it was using not only my solar excess but also consuming energy from the grid. At the time, my panels were generating ~9Kw and evcc was charging the vehicle with the maximum capacity of my my charger, 11.6kW (see screenshot below).

[image]

[image]

I stopped the charging and tried a second time. On the second attempt I had similar issues as shown below:

[image]
[image]

After both charging session, you can see the amount of solar energy used:

[image]

I am not sure if this is a configuration problem, if this is the behaviour expected, or if this could be a bug. Let me know if you need more information.

### Steps to reproduce

1. Charge car using Solar mode and check if grid energy was also used during charging.

### Configuration details

```yaml
# trial token, valid until 2024-11-14
sponsortoken: ...

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ...

interval: 30s # control cycle interval

chargers:
- type: template
  template: twc3
  host: 192.168.1.136
  name: wallbox1

loadpoints:
- title: Garage
  charger: wallbox1
  mode: pv

site:
  title: My home
  meters:
    grid: ha.grid
    pv:
      - ha.pv

meters:
  - name: ha.grid
    type: custom
    power:
      source: http
      uri: http://192.168.1.5:8123/api/states/sensor.sa_import_export_power
[... 274 lines cut ...]
evcc  | [site  ] DEBUG 2024/11/11 01:06:16 pv power: 9614W
evcc  | [site  ] DEBUG 2024/11/11 01:06:16 site power: -5801W
evcc  | [lp-1  ] DEBUG 2024/11/11 01:06:16 charge voltages: [3.9 0 0]V
evcc  | [lp-1  ] DEBUG 2024/11/11 01:06:16 charger status: A
evcc  | [site  ] DEBUG 2024/11/11 01:06:46 ----
evcc  | [lp-1  ] DEBUG 2024/11/11 01:06:46 charge power: 0W
evcc  | [lp-1  ] DEBUG 2024/11/11 01:06:46 charge currents: [0.3 0.3 0.2]A
evcc  | [site  ] DEBUG 2024/11/11 01:06:46 pv power: 9700W
evcc  | [site  ] DEBUG 2024/11/11 01:06:46 grid power: -5905W
evcc  | [site  ] DEBUG 2024/11/11 01:06:46 site power: -5905W
evcc  | [lp-1  ] DEBUG 2024/11/11 01:06:46 charge voltages: [3.9 0 0]V
evcc  | [lp-1  ] DEBUG 2024/11/11 01:06:46
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.5

VolkerK62, 2024-11-11:
it is two times the same log (01:00:44 - 01:06:46)  and there is no charging active

rlcintra, 2024-11-11:
@VolkerK62 , apologies, I copied the wrong log. I added the proper logs now.

mfuchs1984, 2024-11-11:
What is the vehicle "RPC22"? The config only shows a "mycar" of type Tesla.

rlcintra, 2024-11-11:
> What is the vehicle "RPC22"? The config only shows a "mycar" of type Tesla.

This is the name of my vehicle configured in my Tesla app. I assume evcc is picking up the name from there.

VolkerK62, 2024-11-11:
your grid meter is not fast enough. It took more than 1 min to react, after charging starts.

```
evcc  | [lp-1  ] DEBUG 2024/11/11 02:57:43 charge power: 0W
evcc  | [site  ] DEBUG 2024/11/11 02:57:43 grid power: -9489W
evcc  | [site  ] DEBUG 2024/11/11 02:57:43 pv power: 9638W
evcc  | [lp-1  ] DEBUG 2024/11/11 02:57:43 ----
evcc  | [lp-1  ] DEBUG 2024/11/11 02:58:13 charge power: 3748W
evcc  | [site  ] DEBUG 2024/11/11 02:58:13 grid power: -9489W
evcc  | [site  ] DEBUG 2024/11/11 02:58:13 pv power: 9638W
evcc  | [site  ] DEBUG 2024/11/11 02:58:43 ----
evcc  | [lp-1  ] DEBUG 2024/11/11 02:58:43 charge power: 11649W
evcc  | [site  ] DEBUG 2024/11/11 02:58:43 grid power: -9459W
evcc  | [site  ] DEBUG 2024/11/11 02:58:43 pv power: 9620W
evcc  | [site  ] DEBUG 2024/11/11 02:59:13 ----
evcc  | [lp-1  ] DEBUG 2024/11/11 02:59:13 charge power: 11670W
evcc  | [site  ] DEBUG 2024/11/11 02:59:13 grid power: -9459W
evcc  | [site  ] DEBUG 2024/11/11 02:59:13 pv power: 9620W
evcc  | [site  ] DEBUG 2024/11/11 02:59:43 ----
evcc  | [lp-1  ] DEBUG 2024/11/11 02:59:43 charge power: 11653W
evcc  | [site  ] DEBUG 2024/11/11 02:59:43 grid power: -1613W
evcc  | [site  ] DEBUG 2024/11/11 02:59:43 pv power: 9664W
evcc  | [site  ] DEBUG 2024/11/11 03:00:13 ----
evcc  | [lp-1  ] DEBUG 2024/11/11 03:00:13 charge power: 11613W
evcc  | [site  ] DEBUG 2024/11/11 03:00:13 grid power: -1613W
evcc  | [site  ] DEBUG 2024/11/11 03:00:13 pv power: 9664W
evcc  | [site  ] DEBUG 2024/11/11 03:00:43 ----
evcc  | [lp-1  ] DEBUG 2024/11/11 03:00:43 charge power: 11525W
evcc  | [site  ] DEBUG 2024/11/11 03:00:43 grid power: -1613W
evcc  | [site  ] DEBUG 2024/11/11 03:00:43 pv power: 9664W
evcc  | [site  ] DEBUG 2024/11/11 03:01:13 ----
evcc  | [lp-1  ] DEBUG 2024/11/11 03:01:13 charge power: 11611W
evcc  | [site  ] DEBUG 2024/11/11 03:01:13 pv power: 9664W
evcc  | [site  ] DEBUG 2024/11/11 03:01:13 grid power: -1613W
evcc  | [site  ] DEBUG 2024/11/11 03:01:43 ----
evcc  | [lp-1  ] DEBUG 2024/11/11 03:01:43 charge power: 11616W
evcc  | [site  ] DEBUG 2024/11/11 03:01:43 grid power: 5193W
evcc  | [site  ] DEBUG 2024/11/11 03:01:43 pv power: 6611W
evcc  | [site  ] DEBUG 2024/11/11 03:02:13 ----
```

rlcintra, 2024-11-11:
Ok, I will look into the issue with the meter. Thanks, @VolkerK62 .

closed 2024-11-11 by rlcintra