evcc-io/evcc #17186: OAuth2 Login Fails Due to Missing code Parameter in Polestar API - Requires Additional Consent Handling

rostbeule, 2024-11-11:
### Describe the bug

The OAuth2 login process with the Polestar API is failing because the code parameter is not returned after the initial authentication request. This issue occurs due to a required user interaction step, specifically the acceptance of terms and conditions, which prevents the automatic retrieval of the code parameter.

### Steps to reproduce

### Steps to Reproduce
1. Start the OAuth2 login process with the Polestar API.
2. After entering credentials, note that the `code` parameter is not returned, as user consent for terms and conditions is still pending.
3. Only a `uid` parameter is available at this stage.

### Expected Behavior
The OAuth2 login process should automatically handle any additional consent requirements from the Polestar API, enabling the retrieval of the `code` parameter without extra user interaction.

### Proposed Solution
1. Implement a check to detect when the `code` parameter is missing but a `uid` parameter is present.
2. In such cases, send a POST request to submit the user’s consent for terms and conditions.
3. After successful consent submission, the process should retrieve the `code` parameter as expected.

### Additional Context
This adjustment will allow the OAuth2 login to handle user consent programmatically within the Polestar API, streamlining the process by removing any manual acceptance step.

### Acceptance Criteria
- [ ] The OAuth2 login process with the Polestar API successfully retrieves the `code` parameter after automatically submitting user consent.
- [ ] No additional user interactions are required post initial login.

### Configuration details

```yaml
vehicles:
  - name: my_car
    type: template
    template: polestar
    title: "Polestar 2"
    user: me@something.com
    password: ********
    vin: LPSVS************
    capacity: 78
```

### Log details

```text
[main] ERROR 2024/11/11 08:20:58 - Vehicle creation failed for 'my_car':
  - Cannot create vehicle type 'template':
      - Cannot create vehicle type 'polestar':
          - Login failed: code not found
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

rostbeule, 2024-11-11:
see also: https://github.com/evcc-io/evcc/issues/17118

kptkip, 2024-11-11:
Consent in the WEB-UI of Polestar already given.
How to do this via evcc?!

loebse, 2024-11-11:
This is exactly the problem @kptkip: it is not possible, the website consent has no impact on the API access. If the EVCC Polestar implementation stays as is, it cannot not work. But you also never know what Polestar is changing tomorrow 😉 

The HA Integration team implemented the consent via API already, this is why it is working again.

andig, 2024-11-11:
> the website consent has no impact on the API access

Since the API is reverse-engineered: who's the regular consumer? The Polestar app? Does that allow giving consent?

loebse, 2024-11-11:
No, the app does not allow to give consent. I am **assuming** it is using another API.

The API we are using here is used by the https://www.polestar.com/ website. Here you can see the SOC etc. as well and you are forced to give consent before you can see the data:

[image]

The HA integration is making the consent via an API call to make the integration work again.

This is all valid for today. Polestar is known for making sudden changes without any communication. Which is understandable as this is not a public / documented API.

kptkip, 2024-11-11:
> The HA integration is making the consent via an API call to make the integration work again.

Why not using the same approach like the HA integration?

rostbeule, 2024-11-11:
Okay, long story short: 
I took a look at what actually needs to be accepted. 
It's just cookies, and nothing is stopping us from rejecting them.

So you can also send:
```csharp
var confirmData = new Dictionary<string, string>
{
    { "pf.submit", "false" },
    { "subject", uid }
};
```
this should do the trick for now.

rostbeule, 2024-11-11:
@andig, I understand your concern. I hope my proposed solution is acceptable. I've adjusted the issue text accordingly. Unfortunately, I am unable to provide further assistance at this time and would appreciate it if either you or someone else could implement it. Personally, I lack the necessary Go skills, and it would take me quite a while to implement it myself.

jahoooo, 2024-11-12:
evcc v0.131.5 as home assistant add-on doesn't solve the issue for me neither. Guess the suggested approach from rostbeule is needed.
I get the same error
[main ] ERROR 2024/11/12 01:31:34 creating vehicle Polestar_2 failed: cannot create vehicle type 'template': cannot create vehicle type 'polestar': login failed: code not found 

Pypolestar in home assistant does display the SOC for me, though. I hoped if it the consent was already given by pypolestar inside home assistant, it would work without consent then for evcc. But looks like this is handled independently and evcc has to do this credentials hack itself.
Oh my god Polestar... What have you done...

andig, 2024-11-12:
> The API we are using here is used by the https://www.polestar.com/ website

Apparently there is a difference regarding consent. Anyway, happy to take a PR.

rostbeule, 2024-11-14:
@andig here you go https://github.com/evcc-io/evcc/pull/17252

andig, 2024-11-16:
Leider gibts jetzt 2 PRs mit unterschiedlichen Kommentaren, einer ist nicht hier her verlinkt. Solange nicht klar ist, welcher funktioniert wird es leider nicht weiter gehen.

andig, 2024-11-16:
Danke- bitte schau Dir die restlichen Kommentare noch an

loebse, 2024-11-16:
Danke Dir, ich habe mir Deine Kommentare angeschaut und bearbeitet.

closed 2024-11-16 by andig