evcc-io/evcc #17192: evcc steuert Batterie nicht korrekt an - Victron Energy

FE-Phoalie, 2024-11-11:
### Describe the bug

Batterie-System in evcc.yaml konfiguriert und auch auslesbar, allerdings wird das Laden und entladen der Batterie nicht freigegeben wenn das Victron Energy System auf externer Steuerung steht. 

Im Modus "Optimiert" steuert das Victron das Batteriesystem eigenständig und die Lade- und Entladeströme werden von evcc korrekt ausgelesen und angezeigt. 
[image]

Schaltet man nun den Modus zurück auf "externe Steuerung" befindet sich die Batterie dauerhaft im Leerlauf und das entsprechende Modbus-Register wird nicht mehr beschrieben (Slave ID: 227, Modbus-Register 37): 
[image]

VRM Instanz: 
[image]

### Steps to reproduce

1. Victron Energy auf "externe Steuerung" einstellen
2. min. SOC erhöhen bzw. Hausbatterie Konfiguration abändern. 

### Configuration details

```yaml
meters:
# Netzeinspeise Zähler - Janitza UM801
- type:           template
  template:       janitza
  id:             1
  host:           
  port:           502
  usage:          grid
  modbus:         tcpip
  name:           main_grid

- name:       my_battery
  type:       template
  template:   victron-energy
  usage:      battery
  host:         # IP-Adresse oder Hostname
  port:       502 # Port (optional)
  capacity:   20 # Akkukapazität in kWh (optional)
  minsoc:     25 # Ladung mit maximaler Geschwindigkeit bis zu dem angegeben Ladestand unabhängig PV-Erzeugung, wenn der Lademodus nicht auf 'Aus' steht (optional)
  maxsoc:     90 # optional

site: 
- title: 
  meters: 
   grid: main_grid
   battery: my_battery
 residualPower: 100
```

### Log details

```text
evcc -l trace
[main  ] INFO 2024/11/11 14:41:38 evcc 0.131.5
[main  ] INFO 2024/11/11 14:41:38 using config file: /etc/evcc.yaml
[db    ] INFO 2024/11/11 14:41:38 using sqlite database: /home/Administrator/.evcc/evcc.db
[db    ] TRACE 2024/11/11 14:41:38 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/11/11 14:41:38 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/11/11 14:41:38 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/11/11 14:41:38 SELECT * FROM `settings` 6 <nil>
[db    ] TRACE 2024/11/11 14:41:38 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/11/11 14:41:38 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/11/11 14:41:38 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/11/11 14:41:38 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/11/11 14:41:38 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/11/11 14:41:38 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/11/11 14:41:38 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[main  ] INFO 2024/11/11 14:41:38 listening at :7070
[db    ] TRACE 2024/11/11 14:41:38 SELECT * FROM `configs` WHERE `configs`.`class` = 2 1 <nil>
[... 124 lines cut ...]
[bender] TRACE 2024/11/11 14:41:39 modbus: send 00 05 00 00 00 06 ff 03 03 e8 00 01
[bender] TRACE 2024/11/11 14:41:39 modbus: recv 00 05 00 00 00 05 ff 03 02 00 00
[lp-6  ] DEBUG 2024/11/11 14:41:39 phase timer inactive
[lp-6  ] DEBUG 2024/11/11 14:41:39 pv timer inactive
[bender] TRACE 2024/11/11 14:41:39 modbus: send 00 05 00 00 00 06 ff 03 03 e8 00 01
[bender] TRACE 2024/11/11 14:41:39 modbus: recv 00 05 00 00 00 05 ff 03 02 00 00
[site  ] WARN 2024/11/11 14:41:39 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/11/11 14:41:39 ----
[server] FATAL 2024/11/11 14:41:39 unlinkat /tmp/evcc: operation not permitted
[lp-6  ] DEBUG 2024/11/11 14:41:39 charge power: 0W
[lp-3  ] DEBUG 2024/11/11 14:41:39 charge power: 0W
[lp-4  ] DEBUG 2024/11/11 14:41:39 charge power: 0W
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.5

andig, 2024-11-12:
Welcher evcc Modus ist hier vermeintlich falsch? Was ist daran falsch?

andig, 2024-11-12:
> Schaltet man nun den Modus zurück auf "externe Steuerung"

Nach Blick ins Template: evcc ändert tatsächlich überhaupt nichts am Modus. Vermutung: externe Fehlerquelle.

OJack0815, 2024-11-12:
Ich habe ebenfalls ein fehlerhaftes Verhalten mit der Batteriesteuerung von Victron beobachtet. 
Wenn Schnellladen mit der Option die Hausbatterie dabei nicht zu entladen gestartet wird während sich die Hausbatterie im idle am minSoC befindet, wird der aktuelle SoC als min SoC gesetzt und die Batterie bleibt idle. Wenn dann aber der Ladevorgang beendet wird, wird ein minSoC gesetzt, der 10-20% geringer ist als der vorher vom User im Victronsystem gewählte minSoC. Das hat zur Folge, dass die Hausbatterie dann weiter entladen wird als gewünscht.

Ich bin mir nicht sicher ob das mit dem hier beschriebenen Problem zusammenhängt oder ein eigenes Problem ist.

andig, 2024-11-13:
Gar kein Peoblem. Den minsoc kannst du in evcc konfigurieren.

FE-Phoalie, 2024-11-13:
Beim genannten Modus handelt es sich um die Einstellung im Victron System, nicht im evcc selbst. 
In der Victron Remote-Konsole kann die Betriebsart der Batterie ausgewählt werden und im Falle der Betriebsart "externe Steuerung" werden die genannten Modbus-Register, welche die Leistung zum Laden- / Entladen der Batterie enthalten, nicht beschrieben. 
Regelt das Victron die Batterie selbst, abhängig vom Netzzähler, so werden diese Register korrekt beschrieben und die Batterie korrekt angesteuert. Daher unsere Vermutung, dass evcc diese Modbus Register nicht beschreibt. 

Die Modbus-Register 37 bis 41 sind im Template auch nicht als Register-Adresse hinterlegt.