evcc-io/evcc #17194: Battery discharge lock not applied with planned charging

petepall, 2024-11-11:
### Describe the bug

According to the info related to battery setup, the battery locking should be applied during fast charging as well as planned charging.

[image]

When selecting fast charging the locking is applied and after a cycle the battery utilisation goes down to zero and the power is pulled from the grid.
[image]

Using a planning the system is not applying the battery locking and continues to charge from the battery.
[image]

Graphical UI configuration:
[image]

### Steps to reproduce

**Failure steps:**
1. Enable min + PV
2. Setup a charging plan
3. Check the locking is applied or not

**Success steps:**
1. Enable fast
2. Check the locking is applied or not.

### Configuration details

```yaml
The setup of PV, battery and car are done via the graphical UI.
Screenshot is shared above.

## minimal configuration example
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070
  
site:
  title: Home # display name for UI
  meters:
    grid: SMA_Home_Manager
    # pv:
    #   - SMA_SB4.0
    #   - SMA_TRI6.0
    #battery:
    #  - SMA_SBS5.0
  residualPower: 100

sponsortoken: xxxx

# define your loadpoints according your needs
# see https://docs.evcc.io/en/docs/reference/configuration/loadpoints
loadpoints:
  - title: Garage # display name for UI
    charger: Alfen_Eve # charger
#    vehicle: Tesla # default vehicle
    mode: MinPV # charge mode (off, now, minpv, pv)

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
[... 654 lines cut ...]
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: recv 64 59 00 00 00 06 03 10 ac 07 00 02
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: send 64 5a 00 00 00 0b 03 10 ac 07 00 02 04 00 00 00 00
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: recv 64 5a 00 00 00 06 03 10 ac 07 00 02
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: send 64 5b 00 00 00 0b 03 10 ac 07 00 02 04 00 00 00 00
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: recv 64 5b 00 00 00 06 03 10 ac 07 00 02
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: send 64 5c 00 00 00 0b 03 10 ac 07 00 02 04 00 00 00 00
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: recv 64 5c 00 00 00 06 03 10 ac 07 00 02
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: send 64 5d 00 00 00 0b 03 10 ac 07 00 02 04 00 00 00 00
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: recv 64 5d 00 00 00 06 03 10 ac 07 00 02
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: send 64 5e 00 00 00 0b 03 10 ac 07 00 02 04 00 00 00 00
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: recv 64 5e 00 00 00 06 03 10 ac 07 00 02
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: send 64 5f 00 00 00 0b 03 10 ac 07 00 02 04 00 00 00 00
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: recv 64 5f 00 00 00 06 03 10 ac 07 00 02
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: send 64 60 00 00 00 0b 03 10 ac 07 00 02 04 00 00 00 00
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: recv 64 60 00 00 00 06 03 10 ac 07 00 02
[db:5 ] TRACE 2024/11/11 17:08:16 modbus: send 64 61 00 00 00 0b 03 10 ac 07 00 02 04 00 00 00 00
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

VolkerK62, 2024-11-11:
Maybe the problem is the mode "min+solar". 
Did you try it with mode "pv"?

petepall, 2024-11-11:
> Maybe the problem is the mode "min+solar". Did you try it with mode "pv"?

I did not test with PV yet as currently we don’t have enough sun to get solar charging to kick in.
Will try this when possible.

skipperkm, 2024-11-26:
Hallo ,

ich hatte heute das gleiche Problem, allerdings im Modus "schnell laden".

Konfiguration:
HomeAssistant AddOn v 0.131.7
[image]

# open evcc at http://evcc.local:7070 network: schema: http host: evcc.local # .local suffix announces the hostname on MDNS port: 7070

log: debug
levels:
  cache: error

# trial token, valid until 2024-06-18
sponsortoken: XXXX
# unique installation id
plant: 07d734e1e062599ec53bebef720886a8956ea59190f77b318cd070af7fb8b306

interval: 30s # control cycle interval

meters:
  - name: grid1
#    type: template
#    template: tasmota-sml
#    usage: grid
#    host: 172.16.98.71  # IP-Adresse oder Hostname
#    user: # Standard-User ist admin (optional)
#    password: # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) (optional)
    type: custom
    power: # Leistung (W)
      source: mqtt
      topic: iob/mt681/power-w
#      timeout: 30s
#      scale: -1
    energy: # Energie (kWh) - Umrechnung von Wh -> kWh
      source: mqtt
      topic: iob/mt681/energy-import-wh
#      timeout: 30s
      scale: 0.001
    currents:
      - source: mqtt
        topic: iob/mt681/current-l1-a
#        timeout: 30s
      - source: mqtt
        topic: iob/mt681/current-l2-a
#        timeout: 30s
      - source: mqtt
        topic: iob/mt681/current-l1-a
#        timeout: 30s

  - name: pv1
    type: template
    template: solaredge-hybrid
    id: 1
    host: 172.16.98.70
    port: 1502
    usage: pv
    modbus: tcpip
    watchdog: 60s

  - name: battery1
    type: template
    template: solaredge-hybrid
    id: 1
    host: 172.16.98.70
    port: 1502
    usage: battery
    modbus: tcpip
    watchdog: 60s
[... 1298 lines cut ...]
[site  ] DEBUG 2024/11/26 12:30:06 battery soc: 20%
[site  ] DEBUG 2024/11/26 12:30:06 battery power: 0W
[site  ] DEBUG 2024/11/26 12:30:06 pv power: 869W
[site  ] DEBUG 2024/11/26 12:30:06 site power: 10411W
[lp-1  ] DEBUG 2024/11/26 12:30:06 detected active phases: 3p
[lp-1  ] DEBUG 2024/11/26 12:30:07 charge total import: 937.280kWh
[lp-1  ] DEBUG 2024/11/26 12:30:07 charger status: C
[lp-1  ] DEBUG 2024/11/26 12:30:07 soc estimated: 65.09% (vehicle: 61.30%)
[lp-1  ] DEBUG 2024/11/26 12:30:07 vehicle soc: 65%
[lp-1  ] DEBUG 2024/11/26 12:30:07 vehicle range: 232km
[site  ] DEBUG 2024/11/26 12:30:36 ----
[lp-1  ] DEBUG 2024/11/26 12:30:36 charge power: 10456W
[lp-1  ] DEBUG 2024/11/26 12:30:36 charge currents: [14.9 15.4 15]A
[site  ] DEBUG 2024/11/26 12:30:36 grid power: 10537W
[site  ] DEBUG 2024/11/26 12:30:36 grid currents: [16.5 14.1 16.5]A
[site  ] DEBUG 2024/11/26 12:30:37 battery soc: 20%
[site  ] DEBUG 2024/11/26 12:30:37 battery power: 0W
[site  ] DEBUG 2024/11/26 12:30:37 pv power: 910W
[site  ] DEBUG 2024/11/26 12:30:37 site power: 10787W
[lp-1  ] DEBUG 2024/11/26 12:30:37 detected active phases: 3p
[lp-1  ] DEBUG 2024/11/26 12:30:37 charge total import: 937.360kWh
[lp-1  ] DEBUG 2024/11/26 12:30:37 charger status: C
[lp-1  ] DEBUG 2024/11/26 12:30:37 soc estimated: 65.20% (vehicle: 61.30%)
[lp-1  ] DEBUG 2024/11/26 12:30:37 vehicle soc: 65%
[lp-1  ] DEBUG 2024/11/26 12:30:37 vehicle range: 232km

harrykausl, 2024-11-27:
Ich hatte gestern auch das Problem, dass im Schnelllademodus trotz Batteriesperre (Fronius Hybrid WR) Strom aus der Batterie floss. Nach Aus- und wieder Einschalten der Sperre funktionierte es wieder richtig.

skipperkm, 2024-11-30:
Das hat leider auch nicht geholfen, habe es mehrfach probiert.

andig, 2024-12-01:
@skipperkm ich hab Deinen Post ausgeblendet da es hier erstmal um Planner geht. Ob Batteriesperre bei Dir _überhaupt_ funktioniert testest Du mit

    evcc meter --battery-mode hold

Falls nein bitte separates Issue.

andig, 2024-12-01:
@petepall could you please provide another log with more context, `debug` is sufficient. We use

```go
// IsFastChargingActive indicates if fast charging with maximum power is active
func (lp *Loadpoint) IsFastChargingActive() bool {
	lp.RLock()
	defer lp.RUnlock()

	return lp.mode == api.ModeNow || lp.planActive || lp.minSocNotReached()
}
```

for checking if battery mode should be set. This looks correct as it respects the planner. When you reproduce the problem- plan active but battery mode not set to hold- please also add output of /api/state here.

Closing until updated lock available.

closed 2024-12-01 by andig

skipperkm, 2024-12-01:
Hi,

Danke für die Info - leider kann ich das nicht testen - ich benutze evcc als HA-AddOn und hab somit keine Konsole.

Mit freundlichen Grüßen
Skiperkm

> Am 01.12.2024 um 12:43 schrieb andig ***@***.***>:
> 
> evcc meter --battery-mode hold