evcc-io/evcc #17196: BMW: start charge with 403 (Forbidden)

PVahldiek, 2024-11-11:
### Describe the bug

Hi, 

we are using evcc in the following setup:

- PV and Battery: sma-home-manager and sma-hybrid
- chargers: abl emh1 with modbus rs485tcpip
- vehicles: BMW i3 

When using the command `evcc vehicle -a -l trace` we get the following error:

```
[bmw   ] TRACE 2024/11/11 19:20:06 POST https://cocoapi.bmwgroup.com/eadrax-crccs/v1/vehicles/***/start-charging
[bmw   ] TRACE 2024/11/11 19:20:06 {"type":"https://tools.ietf.org/html/rfc9110#section-15.5.4","title":"Forbidden","status":403,"traceId":"xxx"}
[main  ] ERROR 2024/11/11 19:20:06 start charge: unexpected status: 403 (Forbidden)
```

This is only the last messages of the trace, the previous steps e.g GET https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles?apptimezone=120&appDateTime=1731349206091 works fine. This seems correlated with some bmw side api error and it seems that something is blocking from their side?

Do you have any idea why this happens?
Also we don't see any change in the cars UI state, when plugging in the charger (see screenshot below). The log details (which is logged if we try to plugin the charger) is listed below, evcc is trying to wake up the car, but it can't. I'm not 100% sure but this might be related to the issue above. 

[image]

Thank you in advance

### Steps to reproduce

1. Plug in charger into bmw and check vehicle status and logfile

### Configuration details

```yaml
interval: 30s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sma-hybrid
  id: 3
  host: xxx
  port: 502
  usage: pv
  modbus: tcpip
  name: pv1
- type: template
  template: sma-hybrid
  id: 3
  host: xxx
  port: 502
  usage: battery
  modbus: tcpip
  name: battery2
- type: template
  template: sma-home-manager
  usage: grid
  host: xxx
  name: grid3

chargers:
- type: template
  template: abl
  id: 1
  host: xxx
  port: 502
  modbus: rs485tcpip
  timeout: 10s
  name: wallbox5

vehicles:
- type: template
  template: bmw
  title: BMW i3
  user: xxx
  password: xxx
  vin: xxx
  capacity: xxx
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  vehicle: ev4
  mode: minpv

site:
  title: Mein Zuhause
  meters:
    grid: grid3
    pv:
    - pv1
    battery:
    - battery2

hems:
  type: sma
  AllowControl: false
```

### Log details

```text
[lp-1 ] DEBUG 2024/11/11 19:07:46 wake-up vehicle, attempts left: 0
[bmw ] TRACE 2024/11/11 19:07:46 POST https://cocoapi.bmwgroup.com/eadrax-vrccs/v3/presentation/remote-commands/***/door-lock
[bmw ] TRACE 2024/11/11 19:07:47 {"eventId":"xxx","creationTime":"2024-11-11T18:07:47.007444Z"}
[abl ] TRACE 2024/11/11 19:08:16 modbus: send ":0103002E0005C9\r\n"
[abl ] TRACE 2024/11/11 19:08:16 modbus: recv ">01030A2EB190640001000100011C\r\n"
[abl ] TRACE 2024/11/11 19:08:16 modbus: send ":0103002E0005C9\r\n"
[abl ] TRACE 2024/11/11 19:08:16 modbus: recv ">01030A2EB190640001000100011C\r\n"
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.5

andig, 2024-11-12:
Völlig unabhängig vom API: der i3 muss bei Verbindung laden, sonst startet er auch später die Ladung nicht. Dafür ist "Welcomecharge" vorgesehen und kann über das UI am Fahrzeug konfiguriert werden. Der Bedarf über das BMW API zu starten sollte sich damit erledigen.

andig, 2024-11-13:
...und natürlich https://github.com/evcc-io/evcc/discussions/17213

andig, 2024-11-14:
> Offenbar sind wir nicht die einzigen mit dem Problem.

Die BMW Fehler waren gestört. Das machen sie nicht selektiv für Dich...

> When using the command evcc vehicle -a -l trace we get the following error:

Geht bei meinem i3 ohne Fehler.

> Kann das auch mit der modbus Kommunikation zusammenhängen?

Nein

> Den Tipp mit dem welcomecharge habe ich ausprobiert.

Damit sollte das Auto bei Ladefreigabe auch aufwachen. Mein i3 tut das.

> Dementsprechend schaltet er auch nie auf C.

Das sollte nicht sein. Aber am Ende können wir es auch nicht ändern. Wenn Dein Auto bei Ladefreigabe der WB nicht aufwacht musst Du dich bitte an BMW wenden. BMW ist herzlich willkommen uns eine dokumentierte Schnittstelle anzubieten.

> er versucht es 5 mal und scheitert dann.

Das Log zeigt kein scheitern.

Long story short: hier gibts leider nix was wir tun können :(

closed 2024-11-14 by andig