evcc-io/evcc #17199: add zones to variable tariffs

HolgerMiara, 2024-11-12:
**Is your feature request related to a problem? Please describe.**
The dynamic tariff feature (in my case energinet) plus charges correctly shows the spot market price. But the grid operator adds variable transmission charges on top (see picture below).

Reason is that the grid operator wants us to shift consumption (like charging of our cars) to low load hours. But because the transmission charges (Netzübertragungsgebühren) are missing, the smart "Charging Plan" feature in evcc is unreliable and cannot be used here in Denmark with variable electricity prices.

**Describe the solution you'd like**
I would love to add variable "zones" to my variable tariffs, similar to the zones which are available for the `fixed` tariff type. Example:

``` yaml
tariffs:
  currency: DKK # three letter ISO-4217 currency code (default EUR)
  grid:
    type: energinet # Energinet using the *spot market* price in DKK
    region: dk2 # or dk1
    charges: 0.01 # additional charge in DKK per kWh
    zones:
      - months: Oct-Mar # winter tariff / low load
        days: Mon-Sun
        hours: 0-5    
        price: 0.1145   # DKK/kWh
      - months: Oct-Mar # winter tariff / high load
        days: Mon-Sun
        hours: 6-16   
        price: 0.3434   
      - months: Oct-Mar # winter tariff / peak load
        days: Mon-Sun
        hours: 17-20    
        price: 1.0302 
      - months: Oct-Mar # winter tariff / high load
        days: Mon-Sun
        hours: 21-23    
        price: 0.3434   
 ```

Feel free to ignore the months, they would be nice to have, though... 

Calculation of the total cost is then `total := (price + charges + zones) * (1 + tax) ` where price is the spot market price at Nordpool from the api, charge is the income of my electricity supplier and zones is the income of the grid operator. 

**Describe alternatives you've considered**
The topic has been up a couple of times (rf. https://github.com/evcc-io/evcc/issues/16977 under **zones** and https://github.com/evcc-io/evcc/issues/12371), but I haven't found any solution yet.

**Additional context**
Below a graph that shows today's prices. Values for the dark brown bars are delivered by the api, the light part is missing. Rf. https://sparenergi.dk/privat/spar-energi-i-hverdagen/brug-stroemmen-naar-den-er-billigst
 
[image]

andig, 2024-11-12:
Duplicate to https://github.com/evcc-io/evcc/issues/12371?

haxlex, 2024-11-12:
Hey! I am also using such a tariff on electricity transmission here in Finland. So this would be very much appreciated. For reference my electricity transmission is as following: 

Winter months: 
November - March (1.11-31.3)

Winter high peak times = high transmission prices:
7-22
Winter low peak times = low transmission prices:
22-7

This is valid thru Monday to Saturday. Sunday is an exception day, the transmission price is the low transmission price for the entire day.

April - October (1.4-31.10) are priced at low transmission price.

andig, 2024-11-12:
We do generally not have zone dates, only time of day. You‘ll have to change your config twice per year which should be easy enough using the UI.

haxlex, 2024-11-12:
That would work for me at least. 👍

andig, 2024-11-12:
Closing in favor of https://github.com/evcc-io/evcc/issues/12371

closed 2024-11-12 by andig