evcc-io/evcc #17201: Kein Netzaden mit SMA template "sma-sbs-modbus" / STP8.0 / SBS3.7-10 / BYD

eruberion, 2024-11-12:
### Describe the bug

Es findet kein Netzladen mit dem Template "sma-sbs-modbus".
Bei Nutzung des "sma-sbs-15-25" funktioniert das laden. 

hier hat im Forum ein Mitglied geschrieben woran es liegen könnte (bin aber kein Programmierer)
=> https://github.com/evcc-io/evcc/discussions/16929#discussioncomment-11215017

Versionen
SHM: 2.15.6.R
STP: 4.00.70.R
SBS: 3.12.33.R

Ich hoffe das Log reicht ansonsten reiche ich was nach. Bin nicht sicher ob es reicht habe im Tracelog (über die GUI) nach "battery3" und "site" gefiltert.

### Steps to reproduce

1. Konfiguration
2. Preisschwelle angepasst
3. Änderung werden angenommen (GUI pass sich an)
4. kein Netzladen
5. Änderung wieder zurückgenommen (GUI pass sich an)
...

### Configuration details

```yaml
meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.1.41  
  name: grid1
  
- type: template
  template: sma-inverter-modbus
  usage: pv
  modbus: tcpip
  host: 192.168.1.40
  port: 502 
  id: 3
  name: pv2
  
- type: template
  usage: battery
  host: 192.168.1.44
  name: battery3
  #template: sma-sbs-15-25-modbus
  template: sma-sbs-modbus
  modbus: tcpip
  id: 3
  port: 502
```

### Log details

```text
[site ] DEBUG 2024/11/12 08:09:59 battery soc: 0%
[site ] DEBUG 2024/11/12 08:09:59 battery power: 0W
[site ] DEBUG 2024/11/12 08:09:59 site power: 796W
[site ] DEBUG 2024/11/12 08:10:01 set grid charge limit: 0.6
[site ] DEBUG 2024/11/12 08:10:29 ----
[battery3] TRACE 2024/11/12 08:10:29 modbus: send 00 41 00 00 00 0b 03 10 ac 07 00 02 04 00 00 00 00
[battery3] TRACE 2024/11/12 08:10:29 modbus: recv 00 41 00 00 00 06 03 10 ac 07 00 02
[site ] DEBUG 2024/11/12 08:10:29 set battery mode: charge
[site ] DEBUG 2024/11/12 08:10:29 grid power: 695W
[battery3] TRACE 2024/11/12 08:10:29 modbus: send 00 42 00 00 00 06 03 04 78 37 00 02
[site ] DEBUG 2024/11/12 08:10:29 grid powers: [144 296 256]W
[site ] DEBUG 2024/11/12 08:10:29 grid currents: [1.04 2.94 1.64]A
[battery3] TRACE 2024/11/12 08:10:29 modbus: recv 00 42 00 00 00 07 03 04 04 00 00 00 00
[... 90 lines cut ...]
[site ] DEBUG 2024/11/12 08:13:29 grid currents: [1.3 2.92 1.71]A
[battery3] TRACE 2024/11/12 08:13:29 modbus: recv 00 57 00 00 00 07 03 04 04 00 00 00 00
[battery3] TRACE 2024/11/12 08:13:29 modbus: send 00 58 00 00 00 06 03 03 77 31 00 04
[battery3] TRACE 2024/11/12 08:13:29 modbus: recv 00 58 00 00 00 0b 03 03 08 00 00 00 00 00 78 2c 66
[battery3] TRACE 2024/11/12 08:13:29 modbus: send 00 59 00 00 00 06 03 03 78 7d 00 02
[site ] DEBUG 2024/11/12 08:13:29 pv power: 115W
[battery3] TRACE 2024/11/12 08:13:29 modbus: recv 00 59 00 00 00 07 03 03 04 00 00 00 00
[site ] DEBUG 2024/11/12 08:13:29 battery soc: 0%
[site ] DEBUG 2024/11/12 08:13:29 battery power: 0W
[site ] DEBUG 2024/11/12 08:13:29 site power: 868W
[site ] DEBUG 2024/11/12 08:13:59 ----
[site ] DEBUG 2024/11/12 08:13:59 grid power: 761W
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.5

wittmannchris, 2024-11-17:
Es scheint auch momentan mit den SBS, egal welcher, noch ein Problem zu geben wenn sie im Stand By sind. 

Sie wachen quasi nicht auf…
Ich setze aktuell einmalig die 40151 auf. 802 und danach zyklisch die anderen 6. dann dauert es ca. 60sek bis er aufwacht….

MikeRow1972, 2024-11-17:
> Es scheint auch momentan mit den SBS, egal welcher, noch ein Problem zu geben wenn sie im Stand By sind. 
> 
> 
> 
> Sie wachen quasi nicht auf…
> 
> Ich setze aktuell einmalig die 40151 auf. 802 und danach zyklisch die anderen 6. dann dauert es ca. 60sek bis er aufwacht….

Kann ich auch so reproduzieren

DerJoker-de, 2024-11-18:
Ich habe das gleiche Problem seit dem Update auf ivcc 0.130.5 allerdings mit dem sma-sbs-15-25 Template. Mit dem evcc 0.130.4 hat es funktioniert.

Vielleicht gab es ja eine Änderung im Modbustelegramm in der 0.130.5? 

Kann man eigentlich downgraden auf die 0.130.4 ?

andig, 2024-11-18:
Siehe https://github.com/evcc-io/evcc/issues/15881#issuecomment-2483015924. Ist das hier das gleiche Thema?

DerJoker-de, 2024-11-19:
@andig das könnte sein. Aber das wurde ja schon gefixt von daher sollte es ja gehen. Ich habe gestern noch mal auf die v0.131.4 zurück gedowngraded weil es damit ja bei mir vor drei Wochen noch ging. Was soll ich sagen, daran lag es nicht.

Die Firmware des HM 2.0 (2.15.6R) und des SBS (3.11.11R) hat sich auch nicht geändert.

PR #17393 (2024-11-25): SMA: fix battery control
Fixes #17115
Fixes #17374
Fixes #17201
closed 2024-11-25 by andig