evcc-io/evcc #17204: Sungrow Batterie entlädt sich nach geplanter Ladung.

btota, 2024-11-12:
### Describe the bug

Sungrow Batterie entlädt sich nach geplanter Ladung gesteuert über festgelegten Preis.
Zu beobachten ist, dass " Set forced charge discharge power in W " auf MAX (=10500) und die Batterie mit voller Leistung/Spannung (8kW) geladen wird. Nach Ladevorgang wird Batterie mit ~ 1,1-1,3kW ins Netz bzw. ins Haus entladen. Wenn PV dann UV bekommt dann verteilen sich die 1,1-1,3kW auf PV+Batterie und es wird weiter entladen.

--> Meine Vermutung ist, dass die Zwangsentladung aktiviert wird durch die längere 8kW Ladung.
--> Wenn ich in HA meinen Automaten mit  " Set forced charge discharge power in W " auf MAX (=3000) stelle, dann passiert das nicht.
--> Ich habe über evcc keinen Einfluss/Einstellmöglichkeit für die Batterieladepower. 

### Steps to reproduce

1. evcc -->UI Batterie --> Netzladen --> Preisgrenze setzen
2. Log ist auf trace. Leider zeigt er bei der Simulation um 12 Uhr ca. 5,4kW (vtl. weil)
3. Bild aus HA zeigt das angesprochene Verhalten u. Entladung
[image]
[image]
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ***

interval: 30s # control cycle interval

sponsortoken: ***

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.43  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.43  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.43  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3

chargers:
- type: template
[... 350 lines cut ...]
[site ] DEBUG 2024/11/12 12:10:56 site power: 5148W
[sungrow] TRACE 2024/11/12 12:10:56 modbus: send 53 8c 00 00 00 06 03 04 53 3d 00 02
[sungrow] TRACE 2024/11/12 12:10:56 modbus: recv 53 8c 00 00 00 07 03 04 04 0c e2 00 00
[sungrow] TRACE 2024/11/12 12:10:56 modbus: send 53 8d 00 00 00 06 03 04 53 33 00 02
[sungrow] TRACE 2024/11/12 12:10:56 modbus: recv 53 8d 00 00 00 07 03 04 04 f4 74 00 0f
[lp-1 ] DEBUG 2024/11/12 12:10:56 charge total import: 1045.620kWh
[sungrow] TRACE 2024/11/12 12:10:56 modbus: send 53 8e 00 00 00 06 03 04 53 44 00 01
[sungrow] TRACE 2024/11/12 12:10:56 modbus: recv 53 8e 00 00 00 05 03 04 02 00 06
[lp-1 ] DEBUG 2024/11/12 12:10:56 charger status: B
[sungrow] TRACE 2024/11/12 12:10:56 modbus: send 53 8f 00 00 00 06 03 04 53 41 00 01
[sungrow] TRACE 2024/11/12 12:10:56 modbus: recv 53 8f 00 00 00 05 03 04 02 00 00
[lp-1 ] DEBUG 2024/11/12 12:10:56 limitSoc reached: 86.0% > 85%
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.5

andig, 2024-11-12:
> dass die Zwangsentladung aktiviert wird durch die längere 8kW Ladung.

Evcc setzt nirgendwo forced discharge. Vmtl. Entsteht der fehler durch andere Automatisierungen.

btota, 2024-11-12:
> Evcc setzt nirgendwo forced discharge. Vmtl. Entsteht der fehler durch andere Automatisierungen. 

Das stimmt. evcc setzt **nach** geplanter Ladung alles wieder auf Eigenverbrauch:
[image]
**Während** der geplanten Ladung sieht das so aus:
[image]
[image]

 der gesetzte "Battery forced charge discharge power" bleibt auf 10000, was nebensächlich ist.
[image]

> Aus den Screenshot is eine Entladung zu erkennen- aber die kann ja durchaus durch eine Last hervorgerufen sein.

Klar, der Hausverbrauch ist weiterhin da, aber ~ 300, das Verhalten der Batterieentladung ist aber ~1.3kW konstant.

andig, 2024-11-12:
Ich verstehe nicht was das mit evcc zu tun haben soll wenn der WR auf Eigenverbrauch steht? Klingt mir nach einer Frage an den Sungrow Support.

btota, 2024-11-12:
Ich muss auch mal sagen, eine TOLLE Arbeit von EUCH. Sehr schöne Möglichkeiten, die sich eröffnen. DANK!

Ich finde an dem Verhalten auch fast alles richtig, bis auf die Tatsache, dass die Battery mit voller Power geladen wird u. dann der WR die Batterie vermutlich in den Zwangsentladung-Modus schickt?!.

Kann ich evcc irgendwie noch beim geplanten "Netzladen" der Batterie einen "Ladelimit" mitgeben? Das würde schon einiges bessermachen.

andig, 2024-11-12:
> Ich finde an dem Verhalten auch fast alles richtig, bis auf die Tatsache, dass die Battery mit voller Power geladen wird u. dann der WR die Batterie vermutlich in den Zwangsentladung-Modus schickt?!.

Wir raten. Bitte erstmal die Ursache raus finden.

> Kann ich evcc irgendwie noch beim geplanten "Netzladen" der Batterie einen "Ladelimit" mitgeben? Das würde schon einiges bessermachen.

Gerne feature request- für Batteriethemen gibts schon einen Sammelthread. Im Moment nicht.

naltatis, 2024-11-12:
@btota um sich den Problem zu nähern bzw. es zu isolieren würde ich dir einmal empfehlen deine Home Assistant Integration auszuschalten sodass "nur" evcc mit dem Wechselrichter spricht.

timpetersson81, 2024-11-13:
@andig EVCC could set the allowed discharge current to 100w instead of forced stop. This way we can charge the house battery doing night time allso.

andig, 2024-11-13:
This does not make any sense.

timpetersson81, 2024-11-13:
@andig perhaps I lost some in translation, but I have a related issue. I have asked EVCC not to use the battery, and the set forced mode to stop on the battery. I allso have sungrow. But this allso stops my charge of the home battery during night time when there is little sun expected the next day. Another way to stop the discharging of the battery is to set the allowed discharge current to 100w. This will allowing charging of the battery. I use this doing daytime, when I won't to save battery for peak hours.

andig, 2024-11-13:
> I have asked EVCC not to use the battery, and the set forced mode to stop on the battery. I allso have sungrow. But this allso stops my charge of the home battery during night time when there is little sun expected the next day.

OT. Please don't hijack this as an "all battery" issue.

closed 2024-11-22 by andig