evcc-io/evcc #17216: Reimplementation of Climate check for Tesla

kesaloh13, 2024-11-12:
**Is your feature request related to a problem? Please describe.**
Climate check is not working for tesla, because the according feature was removed due to battery drain

**Describe the solution you'd like**
Maybe because of the new Tesla API, this is working now. It would be very helpful to have this feature for Tesla

**Describe alternatives you've considered**
Use Battery for pre-conditioning or create an automation in home assistant

**Additional context**
This was discussed already here: https://github.com/evcc-io/evcc/discussions/11640

kesaloh13, 2024-11-12:
It would be cool if you can reimplement this feature. I could help with testing if needed. My setup: Tesla Model 3 Highland, go-e Charger Gemini flex 11KW.

andig, 2024-11-13:
> Maybe because of the new Tesla API, this is working now.

Ita still the same api.

closed 2024-11-13 by andig