evcc-io/evcc #17225: Car was not charged as planned

timpetersson81, 2024-11-13:
### Describe the bug

Car was plugin, charging was set to sun, and planned for the next day. I was not charged. In the log I get

[lp-1  ] ERROR 2024/11/13 08:18:36 planner: outdated

It allso seems that the charger does not have any electricity prices

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  port: 7070

log: debug

interval: 15s

meters:
  - name: my_grid
    type: template
    template: sungrow-hybrid
    usage: grid
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.1.63 # Hostname
    port: 502 # Port
    timeout: 10s # optional

  - name: my_pv
    type: template
    template: sungrow-hybrid
    usage: pv

    modbus: tcpip
    id: 1
    host: 192.168.1.63 # Hostname
    port: 502 # Port
    timeout: 10s # optional

  - name: my_battery
    type: template
    template: sungrow-hybrid
    usage: battery

    modbus: tcpip
    id: 1
    host: 192.168.1.63 # Hostname
    port: 502 # Port
    timeout: 10s # optional

chargers:
  - name: my_charger
    type: template
    template: pulsarplus-fw5

vehicles:
  - name: Xpeng
    type: template
    template: offline
    title: # Will be displayed in the user interface (optional)
    capacity: 100

site:
  title: Zuhause
  meters:
    grid: my_grid
    pv: my_pv
    battery: my_battery

loadpoints:
  - title: Carport
    charger: my_charger
    mode: pv

tariffs:
  currency: DKK
  grid:
    type: template
    template: energinet
    region: dk2 # optional
#   zones:
#      - days: Mo-So
#        hours: 00-06
#        charges: 0.1431
#      - days: Mo-So
#        hours: 06-17
#        charges: 0.4293 # DKK/kWh
#      - days: Mo-So
#        hours: 17-21
#        charges: 1.28 # DKK/kWh
#      - days: Mo-So
#        hours: 21-00
#        charges: 0.4293 # DKK/kWh

sponsortoken:  eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidGltcGV0ZXJzc29uODEiLCJleHAiOjE4MjQzNjQ4MDAsImlhdCI6MTcyOTc1NjgwMCwic3JjIjoiZ2gifQ.K0GIZT6c3jCGxdJjTs-FZ96VYmuPKQ5OwrsoPHsgI9Y
```

### Log details

```text
[lp-1  ] DEBUG 2024/11/13 08:18:21 charge power: 0W
[lp-1  ] DEBUG 2024/11/13 08:18:21 charge currents: [0 0 0]A
[site  ] WARN 2024/11/13 08:18:21 planner: outdated
[site  ] DEBUG 2024/11/13 08:18:21 grid power: -2W
[site  ] DEBUG 2024/11/13 08:18:21 pv power: 0W
[site  ] DEBUG 2024/11/13 08:18:21 battery soc: 80%
[site  ] DEBUG 2024/11/13 08:18:21 battery power: 1973W
[site  ] DEBUG 2024/11/13 08:18:21 grid currents: [5.42 -2.7 -2.69]A
[site  ] DEBUG 2024/11/13 08:18:21 site power: 2071W
[lp-1  ] DEBUG 2024/11/13 08:18:21 charge voltages: [227 247 227]V
[lp-1  ] DEBUG 2024/11/13 08:18:21 detected connected phases: 3p
[lp-1  ] DEBUG 2024/11/13 08:18:21 charge total import: 885.149kWh
[lp-1  ] DEBUG 2024/11/13 08:18:21 charger status: B
[lp-1  ] ERROR 2024/11/13 08:18:21 planner: outdated
[lp-1  ] DEBUG 2024/11/13 08:18:21 pv charge current: 0A = 0A + -4.5A (2071W @ 2p)
[site  ] DEBUG 2024/11/13 08:18:36 ----
[lp-1  ] DEBUG 2024/11/13 08:18:36 charge power: 0W
[lp-1  ] DEBUG 2024/11/13 08:18:36 charge currents: [0 0 0]A
[site  ] WARN 2024/11/13 08:18:36 planner: outdated
[site  ] DEBUG 2024/11/13 08:18:36 grid power: -0W
[site  ] DEBUG 2024/11/13 08:18:36 pv power: 0W
[site  ] DEBUG 2024/11/13 08:18:36 grid currents: [5.41 -2.7 -2.7]A
[site  ] DEBUG 2024/11/13 08:18:36 battery soc: 80%
[site  ] DEBUG 2024/11/13 08:18:36 battery power: 1971W
[site  ] DEBUG 2024/11/13 08:18:36 site power: 2071W
[lp-1  ] DEBUG 2024/11/13 08:18:36 charge voltages: [227 247 227]V
[lp-1  ] DEBUG 2024/11/13 08:18:36 detected connected phases: 3p
[lp-1  ] DEBUG 2024/11/13 08:18:36 charge total import: 885.149kWh
[lp-1  ] DEBUG 2024/11/13 08:18:36 charger status: B
[lp-1  ] ERROR 2024/11/13 08:18:36 planner: outdated
[lp-1  ] DEBUG 2024/11/13 08:18:36 pv charge current: 0A = 0A + -4.5A (2071W @ 2p)
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.5

timpetersson81, 2024-11-13:
[image]

andig, 2024-11-13:
> planner: outdated

The planner has not reveiced any rates.

> I was not charged. 

It should charge at latest time possible if not rates are available. The logfile is too short to see this and your screenshot shows tomorrow only (so it isn't even expected to have done that yet).

Converting to discussion.