evcc-io/evcc #17227: solax-inverter-cloud template: tokenid needs quotes, no error handling

mrichtarsky, 2024-11-13:
### Describe the bug

pv value is always 0

Reason:
- Token without quotes is interpreted as a number and encoded as float
Solution:
- Put number in single quotes to force string

ToDo:
- Mention this in the docs (I will push a PR)
- Error handling (see comment)

### Steps to reproduce

```yaml
meters:
  - name: Pv2
    type: template
    template: solax-inverter-cloud
    usage: pv
    tokenid: 65465465465 # some example number
    serial: xxx
```

```
$ evcc meter

[main  ] INFO 2024/11/13 07:25:10 evcc 0.131.4
[main  ] INFO 2024/11/13 07:25:10 using config file: /etc/evcc.yaml
[db    ] INFO 2024/11/13 07:25:10 using sqlite database: /var/lib/evcc/evcc.db
Pv2
---
Power: 0W
```

### Configuration details

```yaml
See above
```

### Log details

```text
[Pv2   ] TRACE 2024/11/13 07:31:56 GET https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=6.5465465465e%2B22&sn=xxx
[Pv2   ] TRACE 2024/11/13 07:31:56 
GET /proxyApp/proxy/api/getRealtimeInfo.do?tokenId=6.5465465465e%2B22&sn=xxx HTTP/1.1
Host: www.solaxcloud.com
User-Agent: Go-http-client/1.1
Accept-Encoding: gzip

HTTP/1.1 200 
Transfer-Encoding: chunked
Connection: keep-alive
Content-Type: application/json;charset=ISO-8859-1
Date: Wed, 13 Nov 2024 07:31:56 GMT
Server: nginx

5c
{"exception":"token invalid!","code":103,"tokenId":"6.5465465465e+22","success":false}
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

mrichtarsky, 2024-11-13:
In addition to adjusting the docs, which I will do, I wonder about error handling in general. It is misleading to output a power value of 0, which could be a real value, instead an error should be raised. How can an error be raised in this case for this template? I checked some others but could not see a mechanism.

The API HTTP response is 200, perhaps an error would be raised if it wasn't?
Otherwise, a check should be done for `success` and an error raised if it is not `true`.

andig, 2024-11-13:
> It is misleading to output a power value of 0, which could be a real value, instead an error should be raised. How can an error be raised in this case for this template? I checked some others but could not see a mechanism.

That's a good point. One approach could be for the templates to extract data using JQ and- if data not available return something else ("error" string) that will trigger an upstream parsing error that is then returned.

> Token without quotes is interpreted as a number and encoded as float

    uri: https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId={{ urlEncode .tokenid }}&sn={{ urlEncode .serial }}

Wondering where that does wrong.

mrichtarsky, 2024-11-15:
> That's a good point. One approach could be for the templates to extract data using JQ and- if data not available return something else ("error" string) that will trigger an upstream parsing error that is then returned.

Would it be feasible to come up with a common way of doing that for all templates to use? (and document it) I could fix it in the Solax template then.

> ```
> uri: https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId={{ urlEncode .tokenid }}&sn={{ urlEncode .serial }}
> ```
> 
> Wondering where that does wrong. Anyway, using quotes to make this disambiguous in yaml is generally best practice and should not be required to be documented per template.

It's easy to forget though here. Usually it's not an issue since tokens often also contain characters.

mrichtarsky, 2024-11-15:
I have pushed #17281, also updating the instructions which seem to have changed.

closed 2024-11-26 by andig