evcc-io/evcc #17233: EVCC - ID.4 Reduces charging power in the middle of the charging process

scoma24, 2024-11-13:
### Describe the bug

In the middle of the charging process (fast charging) or after a few minutes, the charging power is reduced from 11 kWh to 6 kWh. Under Max power 16 A in the settings, the specification is no longer 11 kWh but 7 kWh, and the single-phase charging specification is also too low. Only plugging in and unplugging again leads to charging with 11 kWh for some time. Deactivating EVCC and normal charging with the Wallbox app and VW app also works with 11 kWh.  Charging according to plan (PV setting) does not work either. After a certain time, the charging power is reduced and the departure destination is not reached.
I use EVCC in Home Assistant, which runs on my Synology in a VM

### Steps to reproduce

1. Einstecken
2. EVCC Schnelles Laden aktivieren
3. warten. Ladevorgang wird nach ein paar Minuten oder auch mitten im Ladeprozess reduziert

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: host-ip:8123
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# trial token, valid until 2024-11-14
sponsortoken: token

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

meters:

- name: SMA_blau_pv
    type: template
    template: sma-inverter-modbus
    usage: pv

    # Modbus TCP
    modbus: tcpip
    id: 3
    host: ip # Hostname
    port: port # Port

  - name: SMA_rot_pv
    type: template
    template: sma-inverter-modbus
    usage: pv

    # Modbus TCP
    modbus: tcpip
    id: 3
    host: ip # Hostname
    port: port # Port

  - name: Sonnenbatterie_grid
    type: template
    template: sonnenbatterie
    usage: grid
    host: ip # IP-Adresse oder Hostname

[... 251 lines cut ...]
[site  ] DEBUG 2024/11/13 09:43:50 battery soc: 8%
[site  ] DEBUG 2024/11/13 09:43:50 battery power: -1W
[site  ] DEBUG 2024/11/13 09:43:50 battery has priority at soc 8% (< 50%)
[site  ] DEBUG 2024/11/13 09:43:50 site power: 7907W
[lp-1  ] DEBUG 2024/11/13 09:43:50 detected active phases: 2p
[lp-1  ] DEBUG 2024/11/13 09:43:51 charger status: C
[lp-1  ] DEBUG 2024/11/13 09:43:51 soc estimated: 73.09% (vehicle: 71.00%)
[lp-1  ] DEBUG 2024/11/13 09:43:51 vehicle soc: 73%
[lp-1  ] DEBUG 2024/11/13 09:43:51 vehicle soc limit: 90%
[lp-1  ] DEBUG 2024/11/13 09:43:51 vehicle range: 231km
[lp-1  ] DEBUG 2024/11/13 09:43:51 plan: charge 1h43m17s between 2024-11-13 10:16:43 +0100 CET until 2024-11-13 12:00:00 +0100 CET (power: 7360W, avg cost: 0.242)
[site  ] DEBUG 2024/11/13 09:44:20 ----
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.5

VolkerK62, 2024-11-13:
No idea why, but the car is reducing the power on phase 1
```
[lp-1  ] DEBUG 2024/11/13 07:34:20 charge power: 9984W
[lp-1  ] DEBUG 2024/11/13 07:34:20 charge currents: [15.4 15.5 15]A
[site  ] DEBUG 2024/11/13 07:34:50 ----
[lp-1  ] DEBUG 2024/11/13 07:34:50 charge power: 6656W
[lp-1  ] DEBUG 2024/11/13 07:34:50 charge currents: [0.784 15.3 15.1]A
```
this is not unter control of evcc

scoma24, 2024-11-13:
It's only a problem when I charge via evcc. Otherwise via wallbox app, understand app alone works perfectly. So I think it's due to evcc.

andig, 2024-11-13:
Elli software is a mess, see https://github.com/evcc-io/evcc/discussions/15367. Logfile clearly shows that evcc is NOT requesting the power drop. I can only suggest to get rid of the Elli...