evcc-io/evcc #17234: Shelly Pro 3 EM als Charger funktioniert nicht

stasihasi, 2024-11-13:
### Describe the bug

Shelly Pro 3 EM als charger führt zur Fehlermeldung: charger enabled: unexpected status: 404 (Not Found)

### Steps to reproduce

1. Shelly Pro 3EM als Charger angelegt ->
2. Shelly Pro 3EM durch Demo-Charger ersetzt -> kein Fehler

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

telemetry: false

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

meters:
  - name: grid
    type: template
    template: sma-home-manager
    usage: grid
    host: 192.168.178.10 # IP address or hostname
    
  - name: pv
    type: template
    template: sma-inverter-modbus
    usage: pv
    # Modbus TCP
    modbus: tcpip
    id: 3
    host: 192.168.178.8 # Hostname
    port: 502 # Port

  - name: battery
    type: template
    template: sma-sbs-modbus
    usage: battery
    # Modbus TCP
    modbus: tcpip
    id: 3
    host: 192.168.178.9 # Hostname
    port: 502 # Port
    capacity: 5.12 # Akkukapazität in kWh (optional)
    watchdog: 60s # optional
    
  - name: wp_meter
    type: template
    template: shelly-pro-3em
    #usage: charge
    host: 192.168.178.23 
    
chargers:
- name: Waermepumpe
  type: template
  template: shelly
  host: 192.168.178.23 # IP-Adresse oder Hostname
[... 140 lines cut ...]
[lp-1  ] ERROR 2024/11/13 12:36:35 charger enabled: unexpected status: 404 (Not Found)
[site  ] DEBUG 2024/11/13 12:36:39 ----
[lp-1  ] DEBUG 2024/11/13 12:36:39 charge power: 2876W
[lp-1  ] DEBUG 2024/11/13 12:36:39 charge currents: [12.4 0.218 0.03]A
[site  ] DEBUG 2024/11/13 12:36:39 grid power: 2170W
[site  ] DEBUG 2024/11/13 12:36:39 grid powers: [2563 -192 -201]W
[site  ] DEBUG 2024/11/13 12:36:39 grid currents: [11.2 -0.924 -1.05]A
[site  ] DEBUG 2024/11/13 12:36:39 pv power: 916W
[site  ] DEBUG 2024/11/13 12:36:39 battery soc: 20%
[site  ] DEBUG 2024/11/13 12:36:39 battery power: 0W
[site  ] DEBUG 2024/11/13 12:36:39 site power: 2270W
[lp-1  ] DEBUG 2024/11/13 12:36:39 charge voltages: [230 232 232]V
[lp-1  ] DEBUG 2024/11/13 12:36:39 detected connected phases: 3p
[lp-1  ] DEBUG 2024/11/13 12:36:39 charge total import: 95.346kWh
[lp-1  ] DEBUG 2024/11/13 12:36:39 charger status: B
[lp-1  ] ERROR 2024/11/13 12:36:39 charger enabled: unexpected status: 404 (Not Found)
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

VolkerK62, 2024-11-13:
Hat der Pro3EM überhaupt einen Schaltkontakt?

stasihasi, 2024-11-13:
Ich möchte mit dem Shelly den Verbrauch der Wärmepumpe (=WP) messen und die Funktion in EVCC nutzen namens: "Verhindere Entladung im Schnell-Modus und bei geplantem Laden", sprich der ermittelte Verbrauch der WP wird nicht aus einer geladenen Hausbatterie entnommen, Grund hierfür ist, dass die WP durch einen günstigen Heizstrom betrieben wird und der Hausstrom rund 15 cent teurer ist.
Die Wärmepumpe soll also nicht gesteuert oder geschaltet werden, sondern es soll nur verhindert werden, dass die Hausbatterie durch die WP entladen wird. Modus wäre dann "Schnell" und unter Einstellungen HausBatterie ist der Schalter "Verhindere Entladung im Schnell-Modus und bei geplantem Laden" gesetzt

stasihasi, 2024-11-13:
Dann vielen Dank für die ausführliche Erklärung und die schnelle Hilfe. Ggf. könnte die Info noch in die Doku, dass der Shelly Pro 3EM nicht als charger verwendet werden kann, wegen fehlendem Schaltkontakt.

VolkerK62, 2024-11-13:
nun ja, da gibt es viele. Ein Messgerät ist halt kein Schalter.

Aber, das Schöne an evcc ist, dass es einen Werkzeugkasten zur Verfügung stellt, mit dem man fast alles zusammenbauen kann.

Ich hab dir einen Charger gebaut, der genau deinen Wünschen entspricht. er beinhaltet auch die Leistungsmessung des Shelly. Ein Vehicle ist nicht notwendig.
Wenn die Leistung über 80W ist, wechselt der Status auf "heizen" und die Batterie wird gesperrt. 
```
chargers:
- name: Waermepumpe
  type: switchsocket
  enabled:
    source: script #http
    cmd: /bin/sh -c 'echo 1'
  enable:
    source: script #http
    cmd: /bin/sh -c 'echo 1'
  power:
    source: http
    uri: http://192.168.178.23/rpc/Shelly.GetStatus
    jq: ."em:0".total_act_power
  standbypower: 80
  features:
    - integrateddevice
    - heating
  icon: waterheater

loadpoints:
  - title: Wärmepumpe
    charger: Waermepumpe 
    mode: now 
```