evcc-io/evcc #17235: SMA Hybrid Begrenzung der Netzladung

eckerse, 2024-11-13:
**Is your feature request related to a problem? Please describe.**
Currently forced charging of batteries with SMA STP SE series does not work properly. When setting a charging power above ~50% of inverter power the charging power starts to oscillate nearly full scale between max and 1kW with 20s intervals (also reported e.g. in #16972). This is clearly a problem in SMA firmware and not really EVCC related. It works without oscillations for charging power below 5kW. There are tickets open at SMA, but not from me.
For pictures of the bahaviour see [photovoltaik forum](https://www.photovoltaikforum.com/thread/236843-sma-sunny-tripower-5-0-se-byd-hvm22-probleme-mit-schwankender-ladeleistung/?postID=4029845#post4029845)

**Describe the solution you'd like**
Make battery charging power a configuration parameter in [sma-hybrid template](https://github.com/evcc-io/evcc/blob/3398312e772c314e201691407da5cde64059ea64/templates/definition/meter/sma-hybrid.yaml#L170) instead of virtual infinity.

**Describe alternatives you've considered**
I manually copied the contents of template to evcc.yaml, changed the value for modbus address 40149 to -5000 and charging works correctly.

**Additional context**
This is (hopefully) a firmware problem from SMA side. If addressed from their side, this parameter is not really necessary. I don't know if there are usecases from other users to limit the charging power.

andig, 2024-11-13:
Happy to take a PR making max charging power a template parameter. We already have this for other inverters.

eckerse, 2024-11-13:
created PR #17241

Only inverter template with charging power was the template for sonnenbatterie, so took same paramater names from there.

PR #17241 (2024-11-14): add chargepower to sma-hybrid
Fix #17235
closed 2024-11-14 by andig